avafinger/bananapi-zero-ubuntu-base-minimal #1: Question: i2c, spi & ethernet?

peterberbec, 2018-07-04:
Just wondering if this image supports i2c, spi and ethernet?

Thanks for all the work.

avafinger, 2018-07-04:
I updated to kernel 4.17.4 recently and will check if that is working. For the current Image, ethernet should be working. i2c and spi must be activated. Decompile the DTB and change **status = "disabled";** to **status = "okay";**

peterberbec, 2018-07-05:
Got it! Would sharing the new file be helpful?

avafinger, 2018-07-06:
@peterberbec 
Can you please try the new kernel 4.17.4 with **i2c, spi and ethernet enabled** and see if it works as expected?

I have updated github but due to size constraint  I pushed an SD CARD 8GB Image to mega, please use 7zip to unzip and flash it to SD CARD using **etcher** or **win32diskimager**. 

Link: https://mega.nz/#!obIAVQiA!G0CCABkLunCcG8hEqMf7NfjTBK1jgMytt2f0VdtOl08

peterberbec, 2018-07-07:
I think I got it working by recompiling the dtb file. I currently have an i2c display working. I just have to figure spi out. I just don't have an spi-only device.

avafinger, 2018-07-07:
OK.
The new release has two DTB files which employes different DVFS:
bpi-m2-zero-v1.dtb_4.17.4-m2z (conservative DVFS)
bpi-m2-zero-v2.dtb_4.17.4-m2z (aggressive DVFS)

default: **bpi-m2-zero.dtb -> bpi-m2-zero-v1.dtb_4.17.4-m2z**

In the final, they work very similarly and is hard to tell which one is best, unless you push the CPU to its limits.

peterberbec, 2018-07-11:
Whatever was current when I posted the issue, `Linux bpi0 4.15.7-h2-4 #1 SMP Thu Mar 8 17:41:45 -03 2018 armv7l armv7l  armv7l GNU/Linux` has excellent power management. It is lightyears cooler than the banana kernels.

peterberbec, 2018-07-11:
Is the DTB where I enable gpio support?

avafinger, 2018-07-11:
I think for gpio you should use GPIO SYSFS userspace:
https://www.kernel.org/doc/Documentation/gpio/sysfs.txt

avafinger, 2018-07-11:
You can resort to http://linux-sunxi.org/GPIO

avafinger, 2018-07-19:
I had ssd1306 working with i2c-0 on the v2 build. Things you should check:

* Wiring the ssd1306 to SDA and CLK pins, check if you have SSD1306 that works with 3.3v or 5v. If it is a 3.3v one and you wired 5v, you lost the sdd1306

* Check the software side to talk to ssd1306. Check **dmesg** for any errors

* check if you have the i2c dev nodes:
**sudo ls /dev/i2c***

For the SPI LCD it is a way more complex, you need to make sure the pins you will be using are not shared or in use by another device, for example **uart**. You need to change the settings on the DTB.

To check if you have wired the SPI LCD correctly i would suggest reading:
https://github.com/avafinger/Ubuntu-18.04-LTS-Bionic-Beaver-on-H5#wiring-28-tft-display-to-the-boards

And use https://github.com/bitbank2/SPI_LCD to check if it is working, it rescued me on the above case.

peterberbec, 2018-07-25:
Thanks for the help! I'll keep playing with SPI, but I2C works fine for me. My dream is getting that PaPiRus Zero to work, but that'll take crazy rewrites of the fuse library.

closed 2018-07-25 by peterberbec