avafinger/bananapi-zero-ubuntu-base-minimal #2: 8GB is a bit too big

peterberbec, 2018-07-18:
[image]

I have an 8GB SanDisk card that works for everything. The image you uploaded to mega seems to be slightly too big. I swapped to a MicroCenter card and it seems to work.

Is it possible to shrink the `.img` by just a 100MB?

avafinger, 2018-07-18:
Interesting, let me see if i understand it correctly, your have two 8GB SD CARD, a SanDIsk card and a MicroCenter card, for the SanDisk the Image if too big and for he MicroCenter it fits and works, is that correct?

The **img** format is the "physical"' disk compressed and i can't change it unless i change the layout of the SD CARD (sector numbers and size). My SD CARD is a generic brand (seems to be Kingston rebranded).

Please issue in shell:
**sudo fdisk -l /dev/sdX** where X is your drive letter
[sudo] password for alex: 
```
Disk /dev/sdc: 7,5 GiB, 8078229504 bytes, 15777792 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xa5720053

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sdc1        49152   253951   204800  100M 83 Linux
/dev/sdc2       253952 15777791 15523840  7,4G 83 Linux
```

Do you use win32diskimager to burn your sd card?

Please, note that there are instructions to flash only **the contents of the card** so the size of the SD CARD **does not matter**, can be from 4GB to 32GB.

Can you use linux to format and flash the card? If you have a linux box type in shell:
**sudo ./format_sd_mainline.sh /dev/sdX** (if USB) or **sudo ./format_sd_mainline.sh /dev/mmcblk0** (if internal sd card reader)
and then:
```
sync
sudo ./flash_sdcard_m2z_v2.sh /dev/sdX
```
where X is your device letter.

peterberbec, 2018-07-18:
The Microcenter card is:

```
Disk /dev/mmcblk0: 7.4 GiB, 7969177600 bytes, 15564800 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x58b49123

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       49152   253951   204800  100M 83 Linux
/dev/mmcblk0p2      253952 15532031 15278080  7.3G 83 Linux
```

I'll check the SanDisk one when I can. I have been using Etcher on windows. My linux machine is headless, and makes burning cards annoying

avafinger, 2018-07-18:
Check this: https://github.com/resin-io/etcher/issues/629

avafinger, 2018-07-18:
type too fast, mean this: https://github.com/resin-io/etcher/issues/1614

peterberbec, 2018-07-19:
Interesting. Using win32diskimager gives:

[image]

I guess I got the 8GB - 0.0001GB microsd card!

Thanks for the help.

closed 2018-07-19 by peterberbec

avafinger, 2018-07-19:
I pushed the missing file: flash_sdcard_mz2_v2.sh in case you want to flash >  8GB sd cards...