avafinger/bananapi-zero-ubuntu-base-minimal #3: boot_kernel_4.17.y.tar.gz only contains kernel zImage_4.17.3-5

Anoyomouse, 2018-07-21:
Hi

I tried flashing an SD card for my banana pi, and the boot kernel only contains kernel 4.17.3, and not 4.17.4, which means it can't load any modules and the system boots, but with no modules (or WiFi)

Any change you could update your tar.gz to contain the correct kernel?

Thanks

avafinger, 2018-07-21:
Can you please try again and see if it's been fixed?

avafinger, 2018-07-21:
Thanks for the info, i am updating to kernel 4.17.8 in a few hrs...

Anoyomouse, 2018-07-21:
is it possible to bundle the update as an update (boot + modules directory) for easy upgrading of existing flashed images?

avafinger, 2018-07-21:
I just pushed linux-image-4.17.8-m2z_1.0-1.deb.
To update the kernel, issue in shell:

`sudo dpkg -i linux-image-4.17.8-m2z_1.0-1.deb `
and then:
`sync && sudo reboot`
it will update to **kernel 4.17.8** with kernel-headers.

I have changed for some experiments the freq. kernel checks DVFS, it is more responsive but has a 5ºC increase on Temp. I will find the best approach on next kernel update. There is a script to monitor Freq and Temp. If you can push the board to its limits and check if Freq and cpu core is reduced when Temp is very high.

`sudo ./h2-monitor.sh -w `

Anoyomouse, 2018-07-22:
All applied, my BPi went from 27C to 31C, cpu frequencies also seem to fluctuate a little more, but i thin that's because it's trying to idle more

My application is not so heavy on processing, so this shouldn't be too much of a problem for me

avafinger, 2018-07-22:
stable Kernel 4.17.9 has been released.

`sudo dpkg -i linux-image-4.17.9-m2z_1.0-2.deb`

and then:

`sync && sudo reboot`

Things are set back to normal Temp on this kernel update.

closed 2018-09-10 by avafinger