avafinger/bananapi-zero-ubuntu-base-minimal #7: modprobe: FATAL: Module tun not found in directory /lib/modules/4.17.4-m2z

thiagobucca, 2018-08-22:
Hello,

First of all thanks for this experimental image, so far it is the best image i found for this board. Im trying to use pivpn with my banana pi but just found that the included kernel doesnt have the tun module.

I get the following error:
`
ERROR: Cannot open TUN/TAP dev /dev/net/tun: No such file or directory (errno=2)`

and when i run:

`sudo modprobe tun `

I get that the module is not installed:
`
modprobe: FATAL: Module tun not found in directory /lib/modules/4.17.4-m2z`

Can you please include the module with the next update?

Thanks in advance.

avafinger, 2018-08-22:
Sure.

I have no experience with TUN/TAP.
Anything else to set for the TUN/TAP as modules?

I will try to build a kernel 4.18.3 with this enable as module, but it can take some time.

```
config TUN
	tristate "Universal TUN/TAP device driver support"
	depends on INET
	select CRC32
	---help---
	  TUN/TAP provides packet reception and transmission for user space
	  programs.  It can be viewed as a simple Point-to-Point or Ethernet
	  device, which instead of receiving packets from a physical media,
	  receives them from user space program and instead of sending packets
	  via physical media writes them to the user space program.

	  When a program opens /dev/net/tun, driver creates and registers
	  corresponding net device tunX or tapX.  After a program closed above
	  devices, driver will automatically delete tunXX or tapXX device and
	  all routes corresponding to it.

	  Please read <file:Documentation/networking/tuntap.txt> for more
	  information.

	  To compile this driver as a module, choose M here: the module
	  will be called tun.

	  If you don't know what to use this for, you don't need it.

config TAP
	tristate
	---help---
	  This option is selected by any driver implementing tap user space
	  interface for a virtual interface to re-use core tap functionality.

config TUN_VNET_CROSS_LE
	bool "Support for cross-endian vnet headers on little-endian kernels"
	default n
	---help---
	  This option allows TUN/TAP and MACVTAP device drivers in a
	  little-endian kernel to parse vnet headers that come from a
	  big-endian legacy virtio device.

	  Userspace programs can control the feature using the TUNSETVNETBE
	  and TUNGETVNETBE ioctls.

	  Unless you have a little-endian system hosting a big-endian virtual
	  machine with a legacy virtio NIC, you should say N.

config VETH
	tristate "Virtual ethernet pair device"
	---help---
	  This device is a local ethernet tunnel. Devices are created in pairs.
	  When one end receives the packet it appears on its pair and vice
	  versa.

config VIRTIO_NET
	tristate "Virtio network driver"
	depends on VIRTIO
	---help---
	  This is the virtual network driver for virtio.  It can be used with
	  QEMU based VMMs (like KVM or Xen).  Say Y or M.

config NLMON
	tristate "Virtual netlink monitoring device"
	---help---
	  This option enables a monitoring net device for netlink skbs. The
	  purpose of this is to analyze netlink messages with packet sockets.
	  Thus applications like tcpdump will be able to see local netlink
	  messages if they tap into the netlink device, record pcaps for further
	  diagnostics, etc. This is mostly intended for developers or support
	  to debug netlink issues. If unsure, say N.

config NET_VRF
	tristate "Virtual Routing and Forwarding (Lite)"
	depends on IP_MULTIPLE_TABLES
	depends on NET_L3_MASTER_DEV
	depends on IPV6 || IPV6=n
	depends on IPV6_MULTIPLE_TABLES || IPV6=n
	---help---
	  This option enables the support for mapping interfaces into VRF's. The
	  support enables VRF devices.

config VSOCKMON
    tristate "Virtual vsock monitoring device"
    depends on VHOST_VSOCK
    ---help---
     This option enables a monitoring net device for vsock sockets. It is
     mostly intended for developers or support to debug vsock issues. If
     unsure, say N.

endif # NET_CORE
```

thiagobucca, 2018-08-22:
No, i think thats all. No problem, take your time. I will be happy to test it when its done.

Thanks

avafinger, 2018-08-23:
Banana Pi M2 Zero Mainline Kernel 4.17.17 is out, not tested... See if works for you.

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/commit/0ea27baeb878b550903575266d9c8e182bfb4fc1

thiagobucca, 2018-08-23:
Now it works as expected. Thank you!

closed 2018-08-23 by thiagobucca