avafinger/bananapi-zero-ubuntu-base-minimal #8: g_ether

peterberbec, 2018-08-26:
According to [the forums](http://forum.banana-pi.org/t/ssh-connection-over-the-usb-otg-connection-how/2587/2), a patch is required to enable OTG gadget ethernet. The file is [here](http://igor2.repo.hu/tmp/opi_usb_gadget.txt). Is this enabled on any of the kernels?

avafinger, 2018-08-26:
This patch is for Kernel 3.4.112/113.

CONFIG_USB_OTG is not set for a reason, although I have it in DTB
```
&usb_otg {
	dr_mode = "**otg**";
	status = "okay";
};

```
When configured and enabled you will experience a high avg load > 1.0 and thus generate heat.
I can configure it on next release as experimental, although I don't use it and never tested it.

peterberbec, 2018-08-26:
Can I enable it by decompiling the dtb or does it take a kernel recompilation?

avafinger, 2018-08-26:
It needs a kernel recompilation, I was expecting kernel 4.18.6 for today and I set it for one shot and you tell the results unless you are in a hurry.

peterberbec, 2018-08-26:
Oh great! Let me know when the deb is up and I'll test it.

Thanks for all your work. Best image out there! You actually make this sbc useful.

avafinger, 2018-08-26:
I will build one with OTG support, takes a couple of hours...

peterberbec, 2018-08-26:
Thanks, I'm not home anyway.

In the readme you say the board is colder if you disable hdmi. How would I go about doing that?

avafinger, 2018-08-26:
Ok, i will do thist on 4.18.6...

avafinger, 2018-08-26:
Please, try this kernel image and report back your findings...
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/commit/b753342f1b6122c96e4497bd3a3f5e02354e0962

peterberbec, 2018-08-26:
`[   51.164710] udc-core: couldn't find an available UDC - added [g_ether] to list of pending drivers`

peterberbec, 2018-08-26:
Does [this](https://vjordan.info/log/fpga/nanopi-neo-usb-otg-with-armbian-mainline-4x-kernel.html) sound like it makes sense to this issue?

avafinger, 2018-08-26:
Seems so, I will build a new kernel image with that config set, please report back when you test.

peterberbec, 2018-08-26:
I have not tested otg on the board at all. I was concentrating on i2c, spi, wifi and ethernet. I just now got around to otg.

avafinger, 2018-08-26:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/commit/b91c11031eb9f7eeab91c94cd4a73d33be89f564

Built-in modules, you should have now OTG / gadget functionality. Hopefully.

peterberbec, 2018-08-27:
Got the same dmesg output. Should I try a banana-supplied image, figure out what makes it tick, and come back? I don't want you spending hours on this if there's some trick we need to know.

avafinger, 2018-08-27:
Maybe you should have some g_ether params like this: https://gist.github.com/dwilkins/2be58627bfdd675e3a53a011f55c5d07

Try that image and see if it is working and there are parameters to be passed to g_ether

Sounds like you can't have OTG and gadget at the same time, who knows...

peterberbec, 2018-08-27:
Possibly. The inner workings of these devices follow rules only Dali would understand...