avafinger/bananapi-zero-ubuntu-base-minimal #9: Supporting a ttyS onb GPIO

noradtux, 2018-09-12:
Hi, I currently use an uart on the GPIO port with the original kernel (ttyS3). Is it possible to get this working on this kernel?

avafinger, 2018-09-12:
@noradtux 

To enable uart3 you can do the following:

* do a backup of the DTB file you will change in case something goes wrong

* find the DTB file in use
```
  cd  /boot
  ll bpi-m2-zero.dtb
```  

you will see the linked file as something like this (the extension will be your current kernel version):

`lrwxrwxrwx 1 root root 29 Jul 22 12:36 bpi-m2-zero.dtb -> bpi-m2-zero-v1.dtb_4.17.9-m2z`

* Decompile the dtb file to a source file which you can edit
  
 ` sudo dtc -I dtb -O dts -o m2z.dts bpi-m2-zero-v1.dtb_4.17.9-m2z `

* edit the **DTS** (source file is the **m2z.dts**) and enable the uart3 like so:

```
		serial@1c28000 {
			compatible = "snps,dw-apb-uart";
			reg = <0x1c28000 0x400>;
			interrupts = <0x0 0x0 0x4>;
			reg-shift = <0x2>;
			reg-io-width = <0x4>;
			clocks = <0x3 0x3e>;
			resets = <0x3 0x31>;
			dmas = <0x14 0x6 0x14 0x6>;
			dma-names = "rx", "tx";
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <0x1c>;
		};

		serial@1c28400 {
			compatible = "snps,dw-apb-uart";
			reg = <0x1c28400 0x400>;
			interrupts = <0x0 0x1 0x4>;
			reg-shift = <0x2>;
			reg-io-width = <0x4>;
			clocks = <0x3 0x3f>;
			resets = <0x3 0x32>;
			dmas = <0x14 0x7 0x14 0x7>;
			dma-names = "rx", "tx";
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <0x1d 0x1e>;
		};

		serial@1c28800 {
			compatible = "snps,dw-apb-uart";
			reg = <0x1c28800 0x400>;
			interrupts = <0x0 0x2 0x4>;
			reg-shift = <0x2>;
			reg-io-width = <0x4>;
			clocks = <0x3 0x40>;
			resets = <0x3 0x33>;
			dmas = <0x14 0x8 0x14 0x8>;
			dma-names = "rx", "tx";
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <0x1f>;
		};

		serial@1c28c00 {
			compatible = "snps,dw-apb-uart";
			reg = <0x1c28c00 0x400>;
			interrupts = <0x0 0x3 0x4>;
			reg-shift = <0x2>;
			reg-io-width = <0x4>;
			clocks = <0x3 0x41>;
			resets = <0x3 0x34>;
			dmas = <0x14 0x9 0x14 0x9>;
			dma-names = "rx", "tx";
			status = "disabled"; <<== change this to: status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <0x20>;
		};

```
Save the file and then recompile it to the original file name, like so:

`sudo dtc -I dts -O dtb -o  bpi-m2-zero-v1.dtb_4.17.9-m2z`

* Now you can **sync** && **sudo shutdown -h now** , and power cicle the board

You should have now uart3 enabled on next boot.

patrickpoirier51, 2019-08-29:
hello
I edited the file and the setting were already there (4.17.4) , i tried addin alias for serial 3 but no mapping, the good news is that ttyS2 works on pins 9-11-13 

dmesg | grep serial
[    0.213850] 1c28000.serial: ttyS0 at MMIO 0x1c28000 (irq = 45, base_baud = 1500000) is a U6_16550A
[    0.960417] 1c28400.serial: ttyS1 at MMIO 0x1c28400 (irq = 46, base_baud = 1500000) is a U6_16550A
[    0.990253] 1c28800.serial: ttyS2 at MMIO 0x1c28800 (irq = 47, base_baud = 1500000) is a U6_16550A

let me know if there is some test I can do

Thanks for this great work

avafinger, 2019-08-29:
**uart3** is **disabled**, i think is multiplexed with spi1.
Change from "disabled" to "okay" i think i should work.

CON2-P08 | UART3-TX | PA13-EINT13 | SPI1-CS
CON2-P10 | UART3-RX | PA14-EINT14 | SPI1-CLK
CON2-P12 | UART3-CTS | PA16-EINT16 | SPI1-MISO
CON2-P16 | UART3-RTS | PA15-EINT15 | SPI1-MOSI

		uart3: serial@1c28c00 {
			compatible = "snps,dw-apb-uart";
			reg = <0x01c28c00 0x400>;
			interrupts = <GIC_SPI 3 IRQ_TYPE_LEVEL_HIGH>;
			reg-shift = <2>;
			reg-io-width = <4>;
			clocks = <&ccu CLK_BUS_UART3>;
			resets = <&ccu RST_BUS_UART3>;
			dmas = <&dma 9>, <&dma 9>;
			dma-names = "rx", "tx";
			status = "**### disabled**";
		};

avafinger, 2019-08-29:
Why not get a new sd card with kernel 4.20 or 5.3.y?

avafinger, 2019-08-29:
Try to edit the file on Kernel 4.20 or 5.3.y and remove the sp1 alias.

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v16
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v15

patrickpoirier51, 2019-08-29:
Yep it works Thanks 🥇 , sorry I did not looked at details.

As for the kernel, I want to play safe with the librealsense ( an other stuff) as it takes about 8 hours to compile...let me get the thing flying and I'll get back with enhancements
https://github.com/IntelRealSense/librealsense/issues/4408

avafinger, 2019-08-29:
Wow. Nice project. Don't forget to use a **good** heat-sink on the M2 Zero. Please, update here your progress and post a video when you get this flying...

patrickpoirier51, 2019-08-29:
Yep
I installed a fan over the Pi type heatsink while building to keep core around 30 deg.
As you can see, the system will receive sufficient airflow while flying ;-)

patrickpoirier51, 2019-09-02:
It' s alive !! :-)
https://www.youtube.com/watch?v=aQcRiFnSpGA

avafinger, 2019-09-02:
@patrickpoirier51 
Nice work!
Can you explain the role of the bpi m2z  in the whole system? Is it possible to use any other cheap board like BPI M64 or yet a more powerful NanoPi M4 (which has dual camera sensors?) instead of the more expensive Nvidia NANO? (Yes, a newbie question here).

patrickpoirier51, 2019-09-02:
Actually the bpi m2z is a replacement of the Nvidia NANO because I was looking for a more lightweight and compact system to fly on a Q330 size.

You can read all the detailed blog here : 
https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-4-non-ros-bridge-to-mavlink-in-python/44001

meisterbasti, 2020-02-06:
I have a question about your Ubuntu 19.10 distri. Does the Linux support the GPIO ?
If yes how to command it ? The official documentation of the GPIO is not that good.

avafinger, 2020-02-07:
@meisterbasti 

Yes, there is support for GPIO. 
There some ways to control GPIO in the mainline kernel, please read:
https://linux-sunxi.org/GPIO

```
CONFIG_ARCH_HAVE_CUSTOM_GPIO_H=y
CONFIG_GPIOLIB=y
CONFIG_GPIOLIB_FASTPATH_LIMIT=512
CONFIG_OF_GPIO=y
# CONFIG_DEBUG_GPIO is not set
CONFIG_GPIO_SYSFS=y

```

meisterbasti, 2020-02-18:
If I try to command the GPIO via sysfs as described in your link the only thing I get is:

`-bash: /sys/class/gpio/export: Permission denied` 

> Yes, there is support for GPIO.
> There some ways to control GPIO in the mainline kernel, please read:
> https://linux-sunxi.org/GPIO
> 
> ```
> CONFIG_ARCH_HAVE_CUSTOM_GPIO_H=y
> CONFIG_GPIOLIB=y
> CONFIG_GPIOLIB_FASTPATH_LIMIT=512
> CONFIG_OF_GPIO=y
> # CONFIG_DEBUG_GPIO is not set
> CONFIG_GPIO_SYSFS=y
> ```

The code you posted, is this the current settings of the OS or do I have to set it to the values you posted ?

avafinger, 2020-02-18:
Yes you can use WiringPi or WiringPi2, if use change the environment as described there.
I will post the fix on that thread. Not sure which Image he is using.

**Permission denied**

You need to be root to access the gpio pins. use **sudo** or **sudo su**

meisterbasti, 2020-02-18:
If I do the math for the Pin shown in the schematics for Pin "PA21"

[image]

> E.g for PH18 this would be ( 8 - 1) * 32 + 18 = 224 + 18 = 242 (since 'h' is the 8th letter).

This would be:

(1 - 1) * 32 + 21 = 21

But the only result is: 

[image]

avafinger, 2020-02-19:
Here is what you should do (follow verbatim):

```
Welcome to Ubuntu 19.10 (GNU/Linux 5.3.18 armv7l)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

Last login: Wed Feb 19 00:01:30 2020
ubuntu@bpi-m2z:~$ sudo su
[sudo] password for ubuntu: 
root@bpi-m2z:/home/ubuntu# ls /sys/class/gpio/export
/sys/class/gpio/export
root@bpi-m2z:/home/ubuntu# cat /sys/kernel/debug/pinctrl/*/pinmux-pins
Pinmux settings per pin
Format: pin (name): mux_owner gpio_owner hog?
pin 0 (PA0): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 1 (PA1): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 2 (PA2): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 3 (PA3): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 4 (PA4): 1c28000.serial (GPIO UNCLAIMED) function uart0 group PA4
pin 5 (PA5): 1c28000.serial (GPIO UNCLAIMED) function uart0 group PA5
pin 6 (PA6): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 7 (PA7): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 8 (PA8): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 9 (PA9): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 10 (PA10): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 11 (PA11): 1c2ac00.i2c (GPIO UNCLAIMED) function i2c0 group PA11
pin 12 (PA12): 1c2ac00.i2c (GPIO UNCLAIMED) function i2c0 group PA12
pin 13 (PA13): 1c28c00.serial (GPIO UNCLAIMED) function uart3 group PA13
pin 14 (PA14): 1c28c00.serial (GPIO UNCLAIMED) function uart3 group PA14
pin 15 (PA15): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 16 (PA16): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 17 (PA17): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 18 (PA18): 1c2b000.i2c (GPIO UNCLAIMED) function i2c1 group PA18
pin 19 (PA19): 1c2b000.i2c (GPIO UNCLAIMED) function i2c1 group PA19
pin 20 (PA20): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 21 (PA21): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 64 (PC0): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 65 (PC1): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 66 (PC2): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 67 (PC3): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 68 (PC4): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 69 (PC5): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 70 (PC6): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 71 (PC7): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 72 (PC8): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 73 (PC9): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 74 (PC10): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 75 (PC11): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 76 (PC12): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 77 (PC13): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 78 (PC14): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 79 (PC15): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 80 (PC16): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 96 (PD0): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 97 (PD1): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 98 (PD2): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 99 (PD3): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 100 (PD4): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 101 (PD5): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 102 (PD6): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 103 (PD7): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 104 (PD8): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 105 (PD9): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 106 (PD10): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 107 (PD11): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 108 (PD12): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 109 (PD13): (MUX UNCLAIMED) (GPIO UNCLAIMED)
[... 37 lines cut ...]
pin 202 (PG10): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 203 (PG11): (MUX UNCLAIMED) 1c20800.pinctrl:203
pin 204 (PG12): (MUX UNCLAIMED) 1c20800.pinctrl:204
pin 205 (PG13): (MUX UNCLAIMED) 1c20800.pinctrl:205
Pinmux settings per pin
Format: pin (name): mux_owner gpio_owner hog?
pin 352 (PL0): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 353 (PL1): gpio-regulator 1f02c00.pinctrl:353 function gpio_out group PL1
pin 354 (PL2): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 355 (PL3): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 356 (PL4): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 357 (PL5): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 358 (PL6): (MUX UNCLAIMED) 1f02c00.pinctrl:358
pin 359 (PL7): (MUX UNCLAIMED) 1f02c00.pinctrl:359
pin 360 (PL8): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 361 (PL9): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 362 (PL10): leds 1f02c00.pinctrl:362 function gpio_out group PL10
pin 363 (PL11): 1f02000.ir (GPIO UNCLAIMED) function s_cir_rx group PL11
root@bpi-m2z:/home/ubuntu# echo 21 > /sys/class/gpio/export
root@bpi-m2z:/home/ubuntu# ls /sys/class/gpio/export
/sys/class/gpio/export
root@bpi-m2z:/home/ubuntu# ls /sys/class/gpio/
export  gpio21  gpiochip0  gpiochip352  unexport
root@bpi-m2z:/home/ubuntu# 
```

meisterbasti, 2020-02-19:
Password for su is „ubuntu“ right ?

avafinger, 2020-02-19:
yep, that's right.

meisterbasti, 2020-02-19:
I will try when I’m back Home from work. 
Do you think I should update kernel from 5.3.5 to 3.5.18?

If yes can you advise how to ?

avafinger, 2020-02-19:
Yes, you should update.
I think the error was in 5.3.5 and i fixed it later.
I tested the install for the 5.3.18 and it worked:

`ii  linux-image-5.3.18                1.0-35                            armhf        Linux kernel 5.3.18
`

Just make sure you have free space in /boot partition

avafinger, 2020-02-19:
Instructions:

Download one of the deb packages (i suggest Server if you are going to  use GPIO), and run the install:

* GBM (or Wayland)
linux-image-5.3.18_1.0-35.deb

* Server
linux-image-5.3.18-server_1.0-37.deb

* fbdev
linux-image-5.3.18-fbdev_1.0-38.deb

Type in the shell:

```
ubuntu@bpi-m2z:~$ wget --no-check-certificate https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/download/v2.9/linux-image-5.3.18-server_1.0-37.deb
sudo dpkg -i linux-image-5.3.18-server_1.0-37.deb
sync
sudo reboot

```

I have tested the **linux-image-5.3.18 1.0-35.deb**
If you get any error during the install, DON'T reboot, dowload the one i  tested and worked and install that one.

meisterbasti, 2020-02-19:
I think I have the GBM (Wayland) because of #32 

will try the `linux-image-5.3.18_1.0-35.deb`

meisterbasti, 2020-02-19:
Good news ! The update to kernel 5.3.18 was successful and the GPIO over sysfs ist working !

Now a question about the wiringPi2: Do I have to change the environment in

`/etc/environment`

or do I have to 

> Here is the solution:
> 
> Copy the files from https://github.com/BPI-SINOVOIP/BPI-files/tree/master/others/bpi-service to your system
> Change bpi-m3 to bpi-m2z in this file var/lib/bananapi/board.sh
> Install https://github.com/BPI-SINOVOIP/BPI-WiringPi2
> Have fun with wiringPi

avafinger, 2020-02-19:
Please, refer to the last answer on that thread: http://forum.banana-pi.org/t/banana-pi-m2-zero-wiringpi2/5517

You can direct the question to the user there, i am not sure which image was in use.