avafinger/bananapi-zero-ubuntu-base-minimal #11: Overlayfs

stavgremlin, 2018-11-21:
Can you please add overlayfs support to your kernel? I use 4.18.4 kernel. Also, if it is possible, i need ds1307 rct clock and enc28j60 drivers.
I could not use newest kernels, because i beed framebuffer support.

avafinger, 2018-11-21:
I don't get what you mean by "I could not use newest kernels, because i beed framebuffer support."?
All the kernels have framebuffer support, including 4.20.0-rc3. Please explain.

Regarding Overlays, i will see if it is possible to add it to next build. I Will add the driver support.

stavgremlin, 2018-11-22:
And what means thet mali does not works and removed? Information about framebuffer with 1920x1080 resolution was only in v2 and v3 kernel updates.
I mean FS overlay. Also usable are unionfs or aufs, but overlayfs support is in vanilla kernel.

avafinger, 2018-11-22:
Ok, will set those on next 4.18.y build.

stavgremlin, 2018-11-22:
And could you add this modules to all kernels? In plans, is to update kernel when mali would works.

avafinger, 2018-11-24:
Here is the new linux-image-4.18.20-m2z-experimental_1.0-17.deb with overlayfs , rtc-ds1307 and enc28j60 driver as modules.
There is also a 4.19.4 if you want to try out, but not sure if mali is working properly.

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/commit/03552cde975473b40f53c8ace80e0e1be27a9c76

closed 2018-11-24 by avafinger

avafinger, 2018-11-27:
Kernel 4.19.5 LTS with mali working: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/commit/7a1c530f8f82bd65217dcdc1e85d4593efc7bc0d

stavgremlin, 2018-11-28:
Thanks a lot, 4.18.20 works fine with overlayfs. ds1307 and enc 28j60 not tested yet. Will try with 4.19.5.

stavgremlin, 2019-02-05:
I've tried to use fbdev on kernels 4.18.20 and 4.19.5, and could not find device /dev/fb0
Does this kernels supports framebuffer device?
And please, could you add support of EMMC flash, for use this kernel with banana pi zero p2. I've tried, but i can see only micro sd flash.

avafinger, 2019-02-05:
> Does this kernels supports framebuffer device?

4.19 and 4.20 support framebuffer. You should see /dev/fb0 amd mali loaded.

Can your post your kernel 4.19.5 bootlog?

> And please, could you add support of EMMC flash

I dan't have the board but i can try.
Please, grab a complete bootlog (u-boot + kernel) and host the file somewhere.
Also print the output of:
`ls /dev/mmcblk*`

avafinger, 2019-02-06:
You should have in your bootlog something like:

[    1.588177] Console: switching to colour frame buffer device 240x67
[    1.627754] sun4i-drm display-engine: fb0: DRM emulated frame buffer device
[    1.635190] [drm] Initialized sun4i-drm 1.0.0 20150629 for display-engine on minor 0

and
[    8.174976] mali: loading out-of-tree module taints kernel.
[    8.191738] Allwinner sunXi mali glue initialized
[    8.195547] Mali: 
[    8.195556] Found Mali GPU Mali-400 MP r1p1
[    8.208468] Mali: 
[    8.208477] 2+0 PP cores initialized
[    8.209602] Mali: 
[    8.209607] Mali device driver loaded

stavgremlin, 2019-02-06:
ls /dev/mmcblk*
/dev/mmcblk0  /dev/mmcblk0p1  /dev/mmcblk0p2  /dev/mmcblk0p3

stavgremlin, 2019-02-06:
About framebuffer - maybe this is hardware problem, sorry. On another board /dev/fb0 appears.

avafinger, 2019-02-06:
> /dev/mmcblk0 /dev/mmcblk0p1 /dev/mmcblk0p2 /dev/mmcblk0p3

Looks something else, did you boot from SD CARD?

stavgremlin, 2019-02-06:
For now i boot from micro-sd card. And i don't see internel emmc 8 Gb.

stavgremlin, 2019-02-06:
dmesg | grep mmc
[    0.000000] Kernel command line: console=ttyS0,115200 earlyprintk rootfstype=ext4 root=/dev/mmcblk0p2 ro rootwait fsck.repair=yes no_console_suspend consoleblank=0 panic=10
[    2.088481] sunxi-mmc 1c0f000.mmc: Linked as a consumer to regulator.2
[    2.120922] sunxi-mmc 1c0f000.mmc: initialized, max. request size: 16384 KB
[    2.128288] sunxi-mmc 1c10000.mmc: Linked as a consumer to regulator.2
[    2.135530] sunxi-mmc 1c10000.mmc: allocated mmc-pwrseq
[    2.164773] sunxi-mmc 1c10000.mmc: initialized, max. request size: 16384 KB
[    2.198678] mmc0: host does not support reading read-only switch, assuming write-enable
[    2.215014] mmc0: new high speed SDHC card at address 59b4
[    2.224844] mmcblk0: mmc0:59b4 USDU1 7.31 GiB
[    2.237718] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    2.245224]  mmcblk0: p1 p2 p3
[    2.254327] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.262797] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.275109] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    2.287249] mmc1: queuing unknown CIS tuple 0x81 (9 bytes)
[    2.378212] mmc1: new high speed SDIO card at address 0001
[    3.465925] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[   20.823902] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.clm_blob failed with error -2
[   20.823917] brcmfmac mmc1:0001:1: Falling back to syfs fallback for: brcm/brcmfmac43430-sdio.clm_blob
[   24.337392] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   25.841538] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[   36.187679] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   36.401982] EXT4-fs (mmcblk0p3): re-mounted. Opts: (null)

stavgremlin, 2019-02-06:
dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.5-m2z-otg-gadget (alex@svn) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #2 SMP Tue Nov 27 16:40:08 -02 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Banana Pi M2 Zero
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x55c00000
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x400 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu @(ptrval) s35212 r8192 d22132 u65536
[    0.000000] pcpu-alloc: s35212 r8192 d22132 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyS0,115200 earlyprintk rootfstype=ext4 root=/dev/mmcblk0p2 ro rootwait fsck.repair=yes no_console_suspend consoleblank=0 panic=10
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 374212K/524288K available (8192K kernel code, 340K rwdata, 1948K rodata, 1024K init, 266K bss, 19004K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                   lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 341 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 267 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000007] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000019] Switching to timer-based delay loop, resolution 41ns
[    0.000213] Console: colour dummy device 80x30
[    0.000262] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000276] pid_max: default: 32768 minimum: 301
[    0.000431] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000444] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001120] CPU: Testing write buffer coherency: ok
[    0.001609] CPU0: update cpu_capacity 1024
[    0.001621] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002175] Setting up static identity map for 0x40100000 - 0x40100060
[    0.002342] rcu: Hierarchical SRCU implementation.
[    0.003096] smp: Bringing up secondary CPUs ...
[    0.013884] CPU1: update cpu_capacity 1024
[    0.013890] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.024766] CPU2: update cpu_capacity 1024
[    0.024771] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.035537] CPU3: update cpu_capacity 1024
[    0.035543] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.035627] smp: Brought up 1 node, 4 CPUs
[    0.035655] SMP: Total of 4 processors activated (192.00 BogoMIPS).
[    0.035662] CPU: All CPU(s) started in HYP mode.
[    0.035667] CPU: Virtualization extensions available.
[    0.036708] devtmpfs: initialized
[    0.043046] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.043290] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.043314] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.047851] pinctrl core: initialized pinctrl subsystem
[... 310 lines cut ...]
[   37.914959] NET: Registered protocol family 10
[   37.946315] Segment Routing with IPv6
[   44.239150] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   44.386845] br0: port 1(eth0) entered blocking state
[   44.386864] br0: port 1(eth0) entered disabled state
[   44.387355] device eth0 entered promiscuous mode
[   44.460974] device br0 entered promiscuous mode
[   44.477128] br0: port 1(eth0) entered blocking state
[   44.477147] br0: port 1(eth0) entered forwarding state
[... cut ...]
[  116.336636] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  117.925139] netlink: 'wpa_supplicant': attribute type 213 has an invalid length.
[  118.081161] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  125.939222] br0: port 2(wlan0) entered blocking state
[  125.939240] br0: port 2(wlan0) entered disabled state
[  125.939797] device wlan0 entered promiscuous mode
[  125.939925] br0: port 2(wlan0) entered blocking state
[  125.939938] br0: port 2(wlan0) entered forwarding state

avafinger, 2019-02-06:
[bpi-m2-zero-v4.dtb.zip](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/2837667/bpi-m2-zero-v4.dtb.zip)

Please, try to boot with this DTB and print the dmesg.
Let's see if it can detect the eMMC.
**ls /dev/mmc***

avafinger, 2019-02-06:
The previous DTB was for 4.20, the one below is for 4.19

[bpi-m2-zero-v4.dtb_eMMC.zip](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/2837688/bpi-m2-zero-v4.dtb_eMMC.zip)

Instructions:
* cd /boot
  Unzip the file in /boot directory
* type in shell:
```
cd /boot
sudo ln -sf bpi-m2-zero-v4.dtb_eMMC bpi-m2-zero.dtb
sync
sudo shutdown -h now

```

stavgremlin, 2019-02-06:
With this dtb appears internal emmc as /dev/mmcblk2

avafinger, 2019-02-06:
**Good**. You can now manage to use the script to burn eMMC with 4.17 and then copy 4.19 from SD CARD and try to boot from eMMC.

We might need an updated u-boot, but you could try with the current u-boot since i have already a board booting from eMMC with this u-boot.

The steps would be:
* clone only the needed files for the **format_sd_mainline.sh** and **flash_sdcard_m2z_v2.sh**
  (look inside the files to know which files you need)
  something like:
  cd /tmp
  mkdir -p eMMC
  copy the two scripts and the files to this dir

* run the cmds inside eMMC dir
  **sudo chmod +x format_sd_mainline.sh
  sudo chmod +x flash_sdcard_m2z_v2.sh
  sudo format_sd_mainline.sh /dev/mmcblk2
  sudo flash_sdcard_m2z_v2.sh /dev/mmcblk2**
 
after this you could mount the eMMC partitions to work out the changes:
**cd /tmp
mkdir -p boot_emmc
mkdir -p rootfs_emmc
sudo mount -t ext4 ./boot_emmc /dev/mmcblk2p1
sudo mount -t ext4 ./rootfs_emmc /dev/mmcblk2p2**

You should see **boot_emmc** and **rootfs_emmc** so you can copy and modify the contents easily.

The challenge here will be  to find out if booting from **eMMC** will still be **/dev/mmcblk2** or it will change to **/dev/mmcblk0**

if it is **/dev/mmcblk0** then i think will boot normally.
if not you will need to change **bpi-m2-zero_boot.cmd** to reflect root=/dev/mmcblk**2**p2 and recompile it.

You may need to change also this:
 \# setup boot_device
 fdt set mmc${boot_mmc} boot_device <**1**> => to <0> or <2>

I  have no idea which device eMMC will be with this u-boot. You have to find out yourself.

My first guess would be you should **first try** only whith this change (comment the line):
**#** fdt set mmc${boot_mmc} boot_device <1>

You you have to change the order of the /dev/mmcblk? you must update /etc/fstab accordingly.

**You better have a TTL debug available handy, in case you get stuck at boot.!!**

mariusares, 2020-07-05:
> Thanks a lot, 4.18.20 works fine with overlayfs. ds1307 and enc 28j60 not tested yet. Will try with 4.19.5.

Hi
Did you managed to boot from eMMC?