avafinger/bananapi-zero-ubuntu-base-minimal #14: Bluetooth 4.20-rc3 not working

thiagobucca, 2018-12-10:
Hello, i'm trying the instructions on the main page to get Bluetooth working, however i'm not getting it to work, i'm stuck in the following line:

`brcm_patchram_plus -d --patchram /lib/firmware/ap6212/bcm43438a1.hcd --enable_hci --bd_addr 11:22:33:44:55:66 --no2bytes --tosleep 5000 /dev/ttyS1
`
result is:

`:44:55:66 --no2bytes --tosleep 5000 /dev/ttyS1
option patchram with arg /lib/firmware/ap6212/bcm43438a1.hcd
file /lib/firmware/ap6212/bcm43438a1.hcd could not be opened, error 2`

it seems that don't have this file **bcm43438a1.hcd**

checking in the directory i only have **bcm43438a0.hcd**

i'm on kernel 4.20-rc3

thiagobucca, 2018-12-12:
Thanks, it worked, however when i reboot its not working again, is there a way to made this persistant across reboots?

thiagobucca, 2018-12-12:
Sure, thank you. I'll be happy to test when its ready.

avafinger, 2019-01-16:
@thiagobucca 
You can use the same trick i used for the NanoPi M4 to enable Bluetooth at every boot.
You don't need to install the service described there just the trick in rc.local: 

Refer to:
https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal#bluetooth

* create the **/etc/rc.local**
* add the complete **brcm_patchram_plus** command

PS. you may need the complete path to brcm_patchram_plus

thiagobucca, 2019-01-20:
No go yet, same output.

Here's my rc.local file:

`#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.
#pulseaudio -D
#brcm_patchram_plus -d --enable_hci --no2bytes --use_baudrate_for_download --tosleep 200000 --baudrate 1500000 --patchram /lib/firmware/ap6212/bcm43438a1.hcd --enable_lpm /dev/ttyS0 > $
rfkill unblock bluetooth
brcm_patchram_plus -d --patchram /lib/firmware/ap6212/bcm43438a1.hcd --enable_hci --bd_addr 11:22:33:44:55:66 --no2bytes --tosleep 5000 /dev/ttyS1
exit 0`

avafinger, 2019-01-20:
the rc.local is correct. I will try another approach if it works i post it here.

avafinger, 2019-01-23:
@thiagobucca  
Here is the proper way to enable the BT

the content of /etc/rc.local:

```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

#enable BT
brcm_patchram_plus -d --patchram /lib/firmware/ap6212/bcm43438a1.hcd --enable_hci --bd_addr 11:22:33:44:55:66 --no2bytes --tosleep 5000 /dev/ttyS1  > /tmp/bt.log 2>&1 &

exit 0
#!/bin/bash

```
You can check the content of /tmp/bt.log for any errors.

avafinger, 2019-01-23:
A more elegant way of doing:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal#enbale-bluetooth-on-next-boot-permanently

closed 2019-01-23 by avafinger

thiagobucca, 2019-01-25:
Still didn't worked for me, same error. Can you upload your bcm43438a1.hcd file? I got mine somewhere else and i think maybe this can be the source of me issue.

avafinger, 2019-02-03:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/firmware-v1.0