avafinger/bananapi-zero-ubuntu-base-minimal #15: Kernel Compilation Guide

diantahoc, 2019-01-09:
Hello,

I love you work and appreciate what you are doing. However I searched the repo to find a guide on how to build this from source but I cannot find it. If it exists, could you point it out? Otherwise I ask you for guidance on how to build this kernel from source. Any bash script/basic commands/general tips would do the trick.

Thanks,

agates, 2019-06-27:
I would like to see this as well!

avafinger, 2019-07-22:
@diantahoc @agates 

How to is here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal#build-your-own-kernel

jluquee, 2019-12-12:
hello you have the way to compile the kernel from ubuntu intel 64bits for the banana pi m2z, it is not to take so long to compile from arm

Thank you

avafinger, 2019-12-13:
Same commands and export the following:

export ARCH=arm 
export CROSS_COMPILE=arm-linux-gnueabihf-

Make sure you have the cross compiler set up or change the CROSS_OMPILE=your cross compiler

closed 2019-12-13 by avafinger