avafinger/bananapi-zero-ubuntu-base-minimal #16: Non-standard pinmux for i2c1

asymingt, 2019-01-24:
Hi @avafinger 

I'd like to modify the device tree to pinmux a few things differently out the 40-pin expansion header. I'd like to mux i2c1 SCL and SDA to pins 11 and 12 respectively, while keeping uart3 working with software flow control using TX and RX on pins 8 and 10. Firstly, is this possible? 

Using this [link ](http://wiki.banana-pi.org/Banana_Pi_BPI-M2_ZERO#GPIO_PIN_define) as a reference, I changed the following block in the device tree, under the assumption that the pin ordering is "SCL", "SDA" based on the pattern used in i2c0:

```
 i2c1 {
  pins = "PA1", "PA16";
  function = "i2c1";
  phandle = <0x22>;
};
```

I then deleted this block to force software flow control:

```
uart3_rts_cts {
  pins = "PA15", "PA16";
  function = "uart3";
};
```

I then tried to run `sudo i2cdetect -y 1` but the scan takes a very long time. Presumably, I have overlooked something here. Perhaps I need to update the uart3 block or interrupt field of the i2c1 block. Any help would be greatly appreciated.

asymingt, 2019-01-24:
I think I might have an answer to my own question. It appears as if each pin only supports a certain number of functions, so I cannot just arbitrarily route SDA and SCL to a specific pin.

avafinger, 2019-01-24:
I think **SDA** and **SCL** are routed physically to Pin3 and Pin5 respectively.
Pin12 you could choose one of these: UART3-CTS or EINT16 or SPI1-MISO

StuartIanNaylor, 2019-02-26:
Would it not just cause confusion to move from the standard 3/5 pi0 pins?
PS its fantastic to see a v4 kernel would you say v9 is the stable stable :)

Stuart