avafinger/bananapi-zero-ubuntu-base-minimal #18: no EEMC support ?

allairgoo, 2019-05-21:
I downoaded the 8 Gb image , installed on a BPI-P2, all is OK but..

no EMMC viewed in fdisk -l only the SD-CARD (and the eemc on board with 8 Gb is correctly working with other distribs)

root@bpi-m2z:~# fdisk -l Disk /dev/mmcblk0: 7.5 GiB, 8068792320 bytes, 15759360 sectors Units: sectors of 1 * 512 = 512 bytes Sector size (logical/physical): 512 bytes / 512 bytes I/O size (minimum/optimal): 512 bytes / 512 bytes Disklabel type: dos Disk identifier: 0x58b49123

Device Boot Start End Sectors Size Id Type /dev/mmcblk0p1 49152 253951 204800 100M 83 Linux /dev/mmcblk0p2 253952 15532031 15278080 7.3G 83 Linux

avafinger, 2019-05-21:
I don't have the board, you can try this: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/11#issuecomment-460817529

allairgoo, 2019-05-21:
Thank You for your fast answer. I have updated the kernel to 4.20 and used your DTB :  Fdisk shows now the internal EMMC !

closed 2019-05-21 by allairgoo