avafinger/bananapi-zero-ubuntu-base-minimal #19: Stability issues with latest release

amazingproducer, 2019-06-12:
I've been trying to use this project since it has ethernet support, but it seems to crash horribly after about a day of use. By horribly, I mean that it won't turn back on when I reset it. Here's some output from the serial port during the last event: 

```
[Jun11 07:11] Booting Linux on physical CPU 0x0
[  +0.000000] Linux version 4.20.0-m2z-otg-gadget (alex@svn) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #1 SMP Fri Dec 28 09:38:17 -02 2018
[  +0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[  +0.000000] CPU: div instructions available: patching division code
[  +0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[  +0.000000] OF: fdt: Machine model: Banana Pi M2 Zero
[  +0.000000] Memory policy: Data cache writealloc
[  +0.000000] cma: Reserved 128 MiB at 0x55c00000
[  +0.000000] On node 0 totalpages: 131072
[  +0.000000]   Normal zone: 1024 pages used for memmap
[  +0.000000]   Normal zone: 0 pages reserved
[  +0.000000]   Normal zone: 131072 pages, LIFO batch:31
[  +0.000000] psci: probing for conduit method from DT.
[  +0.000000] psci: Using PSCI v0.1 Function IDs from DT
[  +0.000000] random: get_random_bytes called from start_kernel+0xa0/0x3f4 with crng_init=0
[  +0.000000] percpu: Embedded 16 pages/cpu @(ptrval) s35020 r8192 d22324 u65536
[  +0.000000] pcpu-alloc: s35020 r8192 d22324 u65536 alloc=16*4096
[  +0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[  +0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[  +0.000000] Kernel command line: 
[  +0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[  +0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[  +0.000000] Memory: 374204K/524288K available (8192K kernel code, 341K rwdata, 1960K rodata, 1024K init, 268K bss, 19012K reserved, 131072K cma-reserved, 0K highmem)
[  +0.000000] Virtual kernel memory layout:
                  vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                  fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                  vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                  lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                  pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                  modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                    .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                    .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                    .data : 0x(ptrval) - 0x(ptrval)   ( 342 kB)
                     .bss : 0x(ptrval) - 0x(ptrval)   ( 269 kB)
[  +0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[  +0.000000] rcu: Hierarchical RCU implementation.
[  +0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[  +0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[  +0.000000] GIC: Using split EOI/Deactivate mode
[  +0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[  +0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[  +0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[  +0.000000] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[  +0.000012] Switching to timer-based delay loop, resolution 41ns
[  +0.000185] Console: colour dummy device 80x30
[  +0.000378] printk: console [tty0] enabled
[  +0.000048] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[  +0.000025] pid_max: default: 32768 minimum: 301
[  +0.000154] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[  +0.000024] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[  +0.000733] CPU: Testing write buffer coherency: ok
[  +0.000437] CPU0: update cpu_capacity 1024
[  +0.000025] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[  +0.000530] Setting up static identity map for 0x40100000 - 0x40100060
[  +0.000154] rcu: Hierarchical SRCU implementation.
[  +0.000718] smp: Bringing up secondary CPUs ...
[  +0.010756] CPU1: update cpu_capacity 1024
[  +0.000007] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[  +0.010835] CPU2: update cpu_capacity 1024
[  +0.000006] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[  +0.010754] CPU3: update cpu_capacity 1024
[  +0.000007] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[  +0.000078] smp: Brought up 1 node, 4 CPUs
[  +0.000082] SMP: Total of 4 processors activated (192.00 BogoMIPS).
[  +0.000014] CPU: All CPU(s) started in HYP mode.
[  +0.000011] CPU: Virtualization extensions available.
[  +0.000956] devtmpfs: initialized
[... 369 lines cut ...]
[75458.023666] 3fe0: 00005000 a09fe8a0 b5fe17ad b5fdfdd4 60010030 ffffffff 00000000 00000000
[75458.031858] [<c08dd808>] (_test_and_set_bit) from [<c0132fa4>] (queue_work_on+0x24/0x4c)
[75458.039956] [<c0132fa4>] (queue_work_on) from [<d49be540>] (0xd49be540)
[75458.046574] Code: f591f000 e1912f9f e0120003 01822003 (e181cf92) 
[75458.052670] ---[ end trace 8ae3c577c4edc8ef ]---
[75458.057288] Kernel panic - not syncing: Fatal exception in interrupt
[75458.063659] CPU3: stopping
[75458.066381] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G      D    O      4.20.0-m2z-otg-gadget #1
[75458.074989] Hardware name: Allwinner sun8i Family
[75458.079707] [<c010e87c>] (unwind_backtrace) from [<c010b978>] (show_stack+0x10/0x14)
[75458.087456] [<c010b978>] (show_stack) from [<c08de9ec>] (dump_stack+0x88/0x9c)
[75458.094684] [<c08de9ec>] (dump_stack) from [<c010dabc>] (handle_IPI+0x1c4/0x1e4)
[75458.102086] [<c010dabc>] (handle_IPI) from [<c04b1f9c>] (gic_handle_irq+0x8c/0x90)
[75458.109663] [<c04b1f9c>] (gic_handle_irq) from [<c0101a0c>] (__irq_svc+0x6c/0x90)
[75458.117144] Exception stack(0xdf489f60 to 0xdf489fa8)
[75458.122199] 9f60: 00000000 017a2b94 dfb9d5e0 c0117960 ffffe000 c0d04d30 c0d04d74 00000008
[75458.130378] 9f80: 00000000 c0d04d08 c0c5d330 00000000 df75bc64 df489fb0 c01089fc c0108a00
[75458.138553] 9fa0: 600f0013 ffffffff
[75458.142051] [<c0101a0c>] (__irq_svc) from [<c0108a00>] (arch_cpu_idle+0x38/0x3c)
[75458.149456] [<c0108a00>] (arch_cpu_idle) from [<c01453b0>] (do_idle+0x224/0x25c)
[75458.156857] [<c01453b0>] (do_idle) from [<c0145684>] (cpu_startup_entry+0x18/0x1c)
[75458.164429] [<c0145684>] (cpu_startup_entry) from [<4010244c>] (0x4010244c)
[75459.105314] SMP: failed to stop secondary CPUs
[75459.109763] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

amazingproducer, 2019-06-12:
I've been trying to use the device as a motion detection server, and have only installed what was required to build and run the motion daemon. It's entirely possible that motion is contributing to this issue, but it persists even when motion isn't running/is uninstalled.

amazingproducer, 2019-06-12:
Here's an example of what happens when I use the reset button after the device crashes like above:

```
U-Boot SPL 2017.09-h2 (Jul 06 2018 - 10:16:12)
DRAM: 512 MiB
Trying to boot from MMC1
BPI: skip sunxi_mmc_getcd pin 166(1)

U-Boot 2017.09-h2 (Jul 06 2018 - 10:16:12 -0300) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Banana Pi BPI-M2-Zero
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
BPI: skip sunxi_mmc_getcd pin 166(1)
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
No controllers found
Autoboot in 0 seconds, press <Space> to stop
BPI: skip sunxi_mmc_getcd pin 166(1)
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1219 bytes read in 162 ms (6.8 KiB/s)
## Executing script at 43100000
5696568 bytes read in 514 ms (10.6 MiB/s)
982455 bytes read in 112 ms (8.4 MiB/s)
26872 bytes read in 205 ms (127.9 KiB/s)
## Flattened Device Tree blob at 48000000
   Booting using the fdt blob at 0x48000000
   Loading Device Tree to 49ff6000, end 49fff8f7 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.20.0-m2z-otg-gadget (alex@svn) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #1 SMP Fri Dec 28 09:38:17 -02 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Banana Pi M2 Zero
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x55c00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x3f4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu @(ptrval) s35020 r8192 d22324 u65536
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 374204K/524288K available (8192K kernel code, 341K rwdata, 1960K rodata, 1024K init, 268K bss, 19012K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 342 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 269 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[... 587 lines cut ...]
[   93.535130] 7dc0: c0c5d978 00000026 df447fb0 c01669c4 c0c5d978 c0166f98 c0d05114 c0d1a9d0
[   93.543310] 7de0: e080200c e0802000 df447e10 c04b1f5c c011d5f0 600f0113 ffffffff df447e44
[   93.551489] 7e00: bf000000 df446000 0000000b c0101a0c 00000046 00000000 00041400 00041400
[   93.559668] 7e20: c0d55b88 00000000 c0a23c30 00000000 bf000000 ffffe000 0000000b df458000
[   93.567847] 7e40: 00000007 df447e60 c011d5ec c011d5f0 600f0113 ffffffff 00000051 bf000000
[   93.576026] 7e60: df447e74 00041400 df447ee0 c010bc7c c0a23c30 00000000 00010000 00010000
[   93.584205] 7e80: df446210 0000000b c0a253b0 600f0193 005e2230 c0d09d0c 0000000b c0112788
[   93.592385] 7ea0: 005e2230 df447ee0 df446000 e0803000 00000044 c0112ee8 005e2230 0000000b
[   93.600564] 7ec0: 00000000 c01567ec 005e2230 000f0193 ffffffff df447f14 00000000 c0101b64
[   93.608743] 7ee0: df581900 00000001 00041400 03442204 c0d04d08 00000001 df581900 df581900
[   93.616922] 7f00: 00000000 df40c000 e0803000 00000044 c0d8e9b4 df447f34 c0167798 005e2230
[   93.625101] 7f20: 000f0193 ffffffff 00000051 bf000000 005e2230 c0167798 00000004 00041400
[   93.633280] 7f40: df581900 df581964 00000000 c01677fc df581900 c0d05114 00000000 c016b160
[   93.641459] 7f60: c0c5d978 00000018 00000000 c01669c4 c0c5d978 c0166f98 c0d05114 c0d1a9d0
[   93.649638] 7f80: e080200c e0802000 df447fb0 c04b1f5c 004cad04 800f0030 ffffffff 10c5387d
[   93.657817] 7fa0: 10c5387d beff9134 005e2230 c0101db0 005e1aa0 beff913c 00000000 00000000
[   93.665997] 7fc0: 005c95cc 0000062a beff9134 005e1aa0 fffffffc beff9134 005e2230 00000044
[   93.674176] 7fe0: 005e1aa8 beff90c8 004cacfb 004cad04 800f0030 ffffffff 00000000 00000000
[   93.682355] [<c0167798>] (handle_irq_event_percpu) from [<df447e10>] (0xdf447e10)
[   93.689837] Code: bad PC value
[   93.692894] ---[ end trace 67c37ad97e7e9125 ]---
[   93.697509] Kernel panic - not syncing: Fatal exception in interrupt
[   93.703865] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---

```

avafinger, 2019-06-12:
It looks like you have a bad SD card, even though the problem could be memory or the wifi. Can you try again with a brand new sd card? Please get a well-known brand and report back if you still have the same problem. Provide the log in case the crash persists.

If you have a linux HOST PC you can check the sd card by doing this:
* mount the SD CARD and find the correct device name, something like this.. /dev/sdX or /dev/mmcblkY
* unmount manually with 'sudo umount /media/*' or the correct path
* do a check on both partition:
**sudo fsck.ext4 -f /dev/sdX oy mmcblkpY**

amazingproducer, 2019-06-12:
Both partitions passed fsck. I'm not using the wifi interface for anything and haven't set it up to connect to a network. Could this be contributing to the problem?

avafinger, 2019-06-12:
I don't think so. In any case, please add the comments on the wifi section in /etc/network/interfaces

####allow-hotplug wlan0
####iface wlan0 inet dhcp
#	wpa-ssid "foxy"
#	wpa-psk "1234567890"
#address 192.168.254.55
#netmask 255.255.0.0
#gateway 192.168.254.254
#dns-nameservers 8.8.8.8 8.8.4.4
####wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
# Disable power saving on compatible chipsets (prevents SSH/connection dropouts over WiFi)
#wireless-mode Managed
#wireless-power off

This would prevent trying to connect to an AP, lets narrow things.

And if you have a spare sd card, please try with that card.

avafinger, 2019-06-12:
There is something about CIFS. Can you share what you are doing?

`CIFS: Attempting to mount //192.168.1.1/monitor`

amazingproducer, 2019-06-12:
I'm mounting a share to save video collected from rtsp streams when motion is detected.

amazingproducer, 2019-06-12:
I've commented out the stuff in /etc/network/interfaces, and deleted all the links to wpa_supplicant and hostpad in the if-up/down directories. I didn't catch the first crash because I was using the wrong serial baudrate, but I was able to capture the second boot. I have not been able to pull another card to use yet.

```
U-Boot SPL 2017.09-h2 (Jul 06 2018 - 10:16:12)
DRAM: 512 MiB
Trying to boot from MMC1
BPI: skip sunxi_mmc_getcd pin 166(1)

U-Boot 2017.09-h2 (Jul 06 2018 - 10:16:12 -0300) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Banana Pi BPI-M2-Zero
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
BPI: skip sunxi_mmc_getcd pin 166(1)
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
No controllers found
Autoboot in 0 seconds, press <Space> to stop
BPI: skip sunxi_mmc_getcd pin 166(1)
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1219 bytes read in 161 ms (6.8 KiB/s)
## Executing script at 43100000
5696568 bytes read in 510 ms (10.7 MiB/s)
982455 bytes read in 112 ms (8.4 MiB/s)
26872 bytes read in 204 ms (127.9 KiB/s)
## Flattened Device Tree blob at 48000000
   Booting using the fdt blob at 0x48000000
   Loading Device Tree to 49ff6000, end 49fff8f7 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.20.0-m2z-otg-gadget (alex@svn) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #1 SMP Fri Dec 28 09:38:17 -02 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Banana Pi M2 Zero
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x55c00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x3f4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu @(ptrval) s35020 r8192 d22324 u65536
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 374204K/524288K available (8192K kernel code, 341K rwdata, 1960K rodata, 1024K init, 268K bss, 19012K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 342 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 269 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[... 384 lines cut ...]
[    4.790287] [<c0263f14>] (load_elf_binary) from [<c0216df0>] (search_binary_handler.part.3+0x94/0x21c)
[    4.799595] [<c0216df0>] (search_binary_handler.part.3) from [<c0217838>] (__do_execve_file+0x61c/0x724)
[    4.809067] [<c0217838>] (__do_execve_file) from [<c0217ba0>] (sys_execve+0x34/0x3c)
[    4.816813] [<c0217ba0>] (sys_execve) from [<c0101000>] (ret_fast_syscall+0x0/0x54)
[    4.824467] Exception stack(0xde963fa8 to 0xde963ff0)
[    4.829520] 3fa0:                   0057c6c4 00586660 00586660 be89c328 be89cefc b6d66b50
[    4.837699] 3fc0: 0057c6c4 00586660 0057c8c4 0000000b 00586660 0057c8ec 00000000 b6bbf854
[    4.845875] 3fe0: 0057ab84 be89c2fc 005178ff b6ced1a6
[    4.850930] Code: e3a02000 e1a03000 e280cd2a e92d4070 (e5931034) 
[    4.857025] ---[ end trace 8ecd2e6549341454 ]---
```
From this capture, it seems I'm having trouble with ipv6. I've created a blacklist of modules related to previous crashes that I don't intend to use in my project, declared at the end of /etc/modprobe.d/blacklist.conf.

```
# User blacklist for debugging
blacklist ov5640
blacklist v4l2_fwnode
blacklist v4l2_common
blacklist videodev
blacklist sunxi_cir
blacklist brcmfmac
blacklist brcmutil
blacklist g_serial
blacklist ipv6
```

avafinger, 2019-06-12:
Are you using **mali** (GPU / OpenGLES) for something?

amazingproducer, 2019-06-12:
I don't think so. Should I blacklist mali(0)?

avafinger, 2019-06-12:
Hmm, Try and see if you have them loaded to see if blacklist works. 
lsmod

amazingproducer, 2019-06-12:
huh. It booted after adding the ipv6 to blacklist.conf. lsmod output:
```
Module                  Size  Used by
md4                    16384  0
nls_iso8859_1          16384  1
cifs                  462848  2
fscache               192512  1 cifs
mali                  229376  0
```

avafinger, 2019-06-12:
I think is crashing randomly, a sign of a bad sd card.

I would ask you to re-install systemd, there was a corrupted systemd on v1, the base for all versions.
But to re-install you will need a connection to the net or mount the rootfs on a HOST PC (with linux).

Please, read:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal#first-thing-you-should-do-on-the-first-login-for-v1-only

amazingproducer, 2019-06-12:
I built out an ethernet port for it and am using that. I'll try reinstalling systemd.

amazingproducer, 2019-06-12:
Okay, systemd is reinstalled. Shall I reboot, or would you like additional data first?

avafinger, 2019-06-12:
Reboot and grab the boot log....

amazingproducer, 2019-06-12:
```
U-Boot SPL 2017.09-h2 (Jul 06 2018 - 10:16:12)
DRAM: 512 MiB
Trying to boot from MMC1
BPI: skip sunxi_mmc_getcd pin 166(1)

U-Boot 2017.09-h2 (Jul 06 2018 - 10:16:12 -0300) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Banana Pi BPI-M2-Zero
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
BPI: skip sunxi_mmc_getcd pin 166(1)
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
No controllers found
Autoboot in 0 seconds, press <Space> to stop
BPI: skip sunxi_mmc_getcd pin 166(1)
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1219 bytes read in 161 ms (6.8 KiB/s)
## Executing script at 43100000
5696568 bytes read in 510 ms (10.7 MiB/s)
982455 bytes read in 112 ms (8.4 MiB/s)
26872 bytes read in 204 ms (127.9 KiB/s)
## Flattened Device Tree blob at 48000000
   Booting using the fdt blob at 0x48000000
   Loading Device Tree to 49ff6000, end 49fff8f7 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.20.0-m2z-otg-gadget (alex@svn) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #1 SMP Fri Dec 28 09:38:17 -02 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Banana Pi M2 Zero
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x55c00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x3f4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu @(ptrval) s35020 r8192 d22324 u65536
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 374204K/524288K available (8192K kernel code, 341K rwdata, 1960K rodata, 1024K init, 268K bss, 19012K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 342 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 269 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[... 423 lines cut ...]
[  OK  ] Started LSB: Start NTP daemon.
[   21.225041] rc.local[451]: 12 Jun 19:38:58 ntpdate[559]: step time server 192.168.1.1 offset -1.355430 sec
[   24.124143] rc.local[451]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /usr/local/etc/motion/motion.conf
[   24.161425] rc.local[451]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file /usr/local/etc/motion/mejiro.conf
[   24.193549] rc.local[451]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file /usr/local/etc/motion/jubei.conf
[   24.220424] rc.local[451]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file /usr/local/etc/motion/kurogumo.conf
[   24.251973] rc.local[451]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[   24.308107] rc.local[451]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
[   24.324633] rc.local[451]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[   24.368753] rc.local[451]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[   24.407471] rc.local[451]: [0:motion] [NTC] [ALL] become_daemon: Motion going to daemon mode
[  OK  ] Started /etc/rc.local Compatibility.
[  OK  ] Started Serial Getty on ttyS0.
[  OK  ] Started Getty on tty0.
[  OK  ] Reached target Login Prompts.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Graphical Interface.
[  OK  ] Started Stop ureadahead data collection 45s after completed startup.
         Starting Update UTMP about System Runlevel Changes...
[  OK  ] Started Update UTMP about System Runlevel Changes.

Ubuntu 16.04.4 LTS tengu ttyS0

tengu login: 
```

avafinger, 2019-06-12:
Working?

amazingproducer, 2019-06-12:
It usually dies at around a day of working. I'll run it and check in with you tomorrow.

avafinger, 2019-06-12:
Have you checked the free mem just before it crashes?
`free -m`

amazingproducer, 2019-06-12:
I'll keep a watch on it or something.

amazingproducer, 2019-06-13:
Well, it was running okay. The last info I got from monitoring memory was:
```
Every 2.0s: free -m                                     Thu Jun 13 03:33:29 2019

              total        used        free      shared  buff/cache   available
Mem:            495          62         140           4         292         406
Swap:             0           0           0
``` 
But then this showed up on the serial line:
```
[28493.040992] Unable to handle kernel paging request at virtual address dfba0180
[28493.048287] pgd = 78cec0a3
[28493.051011] [dfba0180] *pgd=5fa1141e(bad)
[28493.055134] Internal error: Oops: 8000000d [#1] SMP ARM
[28493.060397] Modules linked in: md4 nls_iso8859_1 cifs fscache mali(O)
[28493.066870] CPU: 3 PID: 21771 Comm: ml1:mejiro Tainted: G        W  O      4.20.0-m2z-otg-gadget #1
[28493.075919] Hardware name: Allwinner sun8i Family
[28493.080637] PC is at 0xdfba0180
[28493.083800] LR is at clockevents_program_event+0x128/0x170
[28493.089297] pc : [<dfba0180>]    lr : [<c0189f4c>]    psr: 60010193
[28493.095572] sp : cdc65eb8  ip : 00000020  fp : dfb9a50c
[28493.100802] r10: dfb9a588  r9 : 00000000  r8 : 00000000
[28493.106028] r7 : 00000000  r6 : 00928bbc  r5 : 00000000  r4 : dfba0180
[28493.112554] r3 : 00000000  r2 : dfba0180  r1 : dfba0180  r0 : 00000000
[28493.119081] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
[28493.126314] Control: 10c5387d  Table: 544f806a  DAC: 00000051
[28493.132072] Process ml1:mejiro (pid: 21771, stack limit = 0xc2179ea5)
[28493.138522] Stack: (0xcdc65eb8 to 0xcdc66000)
[28493.142888] 5ea0:                                                       03fdb400 000019ea
[28493.151073] 5ec0: dfba0180 7fffffff ffffffff dfb9a680 dfb9a588 c018c11c 00000000 7fffffff
[28493.159255] 5ee0: 7fffffff dfb9a500 dfb9a500 20010193 00000003 c017d0ac 20010193 0000000f
[28493.167436] 5f00: 03fdb400 000019ea 00000100 c0d03080 dfb9a5d8 dfb9a5b0 dfb9a650 dfb9a630
[28493.175616] 5f20: 03654164 000019ea c0c5d9c0 df408a00 c0d05114 00000013 df41c040 df40c000
[28493.183801] 5f40: e0803000 ab19342f ab193288 c06c20e4 df408a00 c016bca0 c0c5d978 00000013
[28493.191986] 5f60: 00000000 00000000 df40c000 c01669c4 c0c5d978 c0166f98 c0d05114 c0d1a9d0
[28493.200179] 5f80: e080200c e0802000 cdc65fb0 c04b1f5c 0002772c 80010030 ffffffff 10c5387d
[28493.208366] 5fa0: 10c5387d ab1936af ab19342f c0101db0 00000001 00000000 aea7fbb8 00000000
[28493.216548] 5fc0: ab193008 ab193288 ab193508 000000c3 00000280 ab1936af ab19342f ab193288
[28493.224729] 5fe0: acc58b47 aea7fb58 ab1931b0 0002772c 80010030 ffffffff 00000000 00000000
[28493.232927] [<c0189f4c>] (clockevents_program_event) from [<c018c11c>] (tick_program_event+0x5c/0x98)
[28493.242157] [<c018c11c>] (tick_program_event) from [<c017d0ac>] (hrtimer_interrupt+0x14c/0x2c8)
[28493.250878] [<c017d0ac>] (hrtimer_interrupt) from [<c06c20e4>] (arch_timer_handler_phys+0x28/0x30)
[28493.259862] [<c06c20e4>] (arch_timer_handler_phys) from [<c016bca0>] (handle_percpu_devid_irq+0x78/0x138)
[28493.269450] [<c016bca0>] (handle_percpu_devid_irq) from [<c01669c4>] (generic_handle_irq+0x24/0x34)
[28493.278512] [<c01669c4>] (generic_handle_irq) from [<c0166f98>] (__handle_domain_irq+0x5c/0xb4)
[28493.287231] [<c0166f98>] (__handle_domain_irq) from [<c04b1f5c>] (gic_handle_irq+0x4c/0x90)
[28493.295605] [<c04b1f5c>] (gic_handle_irq) from [<c0101db0>] (__irq_usr+0x50/0x80)
[28493.303093] Exception stack(0xcdc65fb0 to 0xcdc65ff8)
[28493.308148] 5fa0:                                     00000001 00000000 aea7fbb8 00000000
[28493.316336] 5fc0: ab193008 ab193288 ab193508 000000c3 00000280 ab1936af ab19342f ab193288
[28493.324512] 5fe0: acc58b47 aea7fb58 ab1931b0 0002772c 80010030 ffffffff
[28493.331142] Code: 00000000 00000000 00000000 ffffffe0 (dfba0180) 
[28493.337246] ---[ end trace f0a0b6cc063b2b06 ]---
[28493.341868] Kernel panic - not syncing: Fatal exception in interrupt
[28493.348244] CPU0: stopping
[28493.350982] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D W  O      4.20.0-m2z-otg-gadget #1
[28493.359591] Hardware name: Allwinner sun8i Family
[28493.364322] [<c010e87c>] (unwind_backtrace) from [<c010b978>] (show_stack+0x10/0x14)
[28493.372076] [<c010b978>] (show_stack) from [<c08de9ec>] (dump_stack+0x88/0x9c)
[28493.379306] [<c08de9ec>] (dump_stack) from [<c010dabc>] (handle_IPI+0x1c4/0x1e4)
[28493.386707] [<c010dabc>] (handle_IPI) from [<c04b1f9c>] (gic_handle_irq+0x8c/0x90)
[28493.394280] [<c04b1f9c>] (gic_handle_irq) from [<c0101a0c>] (__irq_svc+0x6c/0x90)
[28493.401762] Exception stack(0xc0d01f18 to 0xc0d01f60)
[28493.406814] 1f00:                                                       00000000 02242c94
[28493.414993] 1f20: dfb6d5e0 c0117960 ffffe000 c0d04d30 c0d04d74 00000001 00000000 c0d04d08
[28493.423172] 1f40: c0c5d330 00000000 00000000 c0d01f68 c01089fc c0108a00 60070013 ffffffff
[28493.431357] [<c0101a0c>] (__irq_svc) from [<c0108a00>] (arch_cpu_idle+0x38/0x3c)
[28493.438761] [<c0108a00>] (arch_cpu_idle) from [<c01453b0>] (do_idle+0x224/0x25c)
[28493.446163] [<c01453b0>] (do_idle) from [<c0145684>] (cpu_startup_entry+0x18/0x1c)
[28493.453741] [<c0145684>] (cpu_startup_entry) from [<c0c00cc8>] (start_kernel+0x3d4/0x3f4)
[... 9 lines cut ...]
[28493.520443] 7f60: 00000000 00ae5df8 dfb8d5e0 c0117960 ffffe000 c0d04d30 c0d04d74 00000004
[28493.528622] 7f80: 00000000 c0d04d08 c0c5d330 00000000 00000000 df487fb0 c01089fc c0108a00
[28493.536796] 7fa0: 60010013 ffffffff
[... cut ...]
[28493.636324] 5f80: 00000000 c0d04d08 c0c5d330 00000000 defe9a64 df485fb0 c01089fc c0108a00
[28493.644498] 5fa0: 600f0013 ffffffff
[28493.647994] [<c0101a0c>] (__irq_svc) from [<c0108a00>] (arch_cpu_idle+0x38/0x3c)
[28493.655394] [<c0108a00>] (arch_cpu_idle) from [<c01453b0>] (do_idle+0x224/0x25c)
[28493.662794] [<c01453b0>] (do_idle) from [<c0145684>] (cpu_startup_entry+0x18/0x1c)
[28493.670365] [<c0145684>] (cpu_startup_entry) from [<4010244c>] (0x4010244c)
[28493.677331] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

avafinger, 2019-06-13:
I don't know what process is this:  **ml1:mejiro** , Is this related to Motion or this https://github.com/aliegjln/linux/blob/master/ml1.0.c

Anyway, i have kernel 4.20.17 running on my A64 and seems very stable, i will port it to H2+ and you can give it a try. I will try to build the new Image this weekend.

amazingproducer, 2019-06-13:
I found another card and tried it with that one overnight. The log is incomplete, but it shows similar results:
```
[31312.874350] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[31365.817366] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[31365.823295] rcu:     3-...0: (1 GPs behind) idle=3f2/1/0x40000000 softirq=611169/611170 fqs=122 
[31365.831815] rcu:     (detected by 0, t=493893 jiffies, g=180485, q=1027406)
[31365.838518] Sending NMI from CPU 0 to CPUs 3:
[31375.842880] rcu: rcu_sched kthread starved for 6296 jiffies! g180485 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=0
[31375.853221] rcu: RCU grace-period kthread stack dump:
[31375.858269] rcu_sched       I    0    10      2 0x00000000
[31375.863766] [<c08f4fe0>] (__schedule) from [<c08f5420>] (schedule+0x40/0xa0)
[31375.870818] [<c08f5420>] (schedule) from [<c08f8df8>] (schedule_timeout+0x170/0x294)
[31375.878565] [<c08f8df8>] (schedule_timeout) from [<c0175d38>] (rcu_gp_kthread+0x588/0x9b4)
[31375.886833] [<c0175d38>] (rcu_gp_kthread) from [<c0139d78>] (kthread+0x148/0x150)
[31375.894320] [<c0139d78>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[31375.901540] Exception stack(0xdf481fb0 to 0xdf481ff8)
[31375.906592] 1fa0:                                     00000000 00000000 00000000 00000000
[31375.914770] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[31375.922948] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[31428.865955] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[31428.871883] rcu:     3-...0: (1 GPs behind) idle=3f2/1/0x40000000 softirq=611169/611170 fqs=123 
[31428.880403] rcu:     (detected by 0, t=500198 jiffies, g=180485, q=1027406)
[31428.887106] Sending NMI from CPU 0 to CPUs 3:
[31438.891468] rcu: rcu_sched kthread starved for 6296 jiffies! g180485 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=0
[31438.901809] rcu: RCU grace-period kthread stack dump:
[31438.906857] rcu_sched       I    0    10      2 0x00000000
[31438.912353] [<c08f4fe0>] (__schedule) from [<c08f5420>] (schedule+0x40/0xa0)
[31438.919405] [<c08f5420>] (schedule) from [<c08f8df8>] (schedule_timeout+0x170/0x294)
[31438.927153] [<c08f8df8>] (schedule_timeout) from [<c0175d38>] (rcu_gp_kthread+0x588/0x9b4)
[31438.935422] [<c0175d38>] (rcu_gp_kthread) from [<c0139d78>] (kthread+0x148/0x150)
[31438.942909] [<c0139d78>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[31438.950128] Exception stack(0xdf481fb0 to 0xdf481ff8)
[31438.955181] 1fa0:                                     00000000 00000000 00000000 00000000
[31438.963359] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[31438.971536] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[31491.914543] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[31491.920472] rcu:     3-...0: (1 GPs behind) idle=3f2/1/0x40000000 softirq=611169/611170 fqs=124 
[31491.928993] rcu:     (detected by 0, t=506503 jiffies, g=180485, q=1027407)
[31491.935696] Sending NMI from CPU 0 to CPUs 3:
[31501.940058] rcu: rcu_sched kthread starved for 6296 jiffies! g180485 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=0
[31501.950399] rcu: RCU grace-period kthread stack dump:
[31501.955447] rcu_sched       I    0    10      2 0x00000000
[31501.960943] [<c08f4fe0>] (__schedule) from [<c08f5420>] (schedule+0x40/0xa0)
[31501.967995] [<c08f5420>] (schedule) from [<c08f8df8>] (schedule_timeout+0x170/0x294)
[31501.975743] [<c08f8df8>] (schedule_timeout) from [<c0175d38>] (rcu_gp_kthread+0x588/0x9b4)
[31501.984011] [<c0175d38>] (rcu_gp_kthread) from [<c0139d78>] (kthread+0x148/0x150)
[31501.991498] [<c0139d78>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[31501.998718] Exception stack(0xdf481fb0 to 0xdf481ff8)
[31502.003769] 1fa0:                                     00000000 00000000 00000000 00000000
[31502.011948] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[31502.020125] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[31554.963132] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[31554.969061] rcu:     3-...0: (1 GPs behind) idle=3f2/1/0x40000000 softirq=611169/611170 fqs=125 
[31554.977581] rcu:     (detected by 1, t=512808 jiffies, g=180485, q=1027407)
[31554.984284] Sending NMI from CPU 1 to CPUs 3:
[31564.988646] rcu: rcu_sched kthread starved for 6296 jiffies! g180485 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=0
[31564.998988] rcu: RCU grace-period kthread stack dump:
[31565.004036] rcu_sched       I    0    10      2 0x00000000
[31565.009532] [<c08f4fe0>] (__schedule) from [<c08f5420>] (schedule+0x40/0xa0)
[31565.016584] [<c08f5420>] (schedule) from [<c08f8df8>] (schedule_timeout+0x170/0x294)
[31565.024332] [<c08f8df8>] (schedule_timeout) from [<c0175d38>] (rcu_gp_kthread+0x588/0x9b4)
[31565.032600] [<c0175d38>] (rcu_gp_kthread) from [<c0139d78>] (kthread+0x148/0x150)
[31565.040087] [<c0139d78>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[31565.047307] Exception stack(0xdf481fb0 to 0xdf481ff8)
[31565.052359] 1fa0:                                     00000000 00000000 00000000 00000000
[31565.060537] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[31565.068714] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[31618.011721] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[31618.017647] rcu:     3-...0: (1 GPs behind) idle=3f2/1/0x40000000 softirq=611169/611170 fqs=126 
[... cut ...]
[35158.793890] [<c08f8df8>] (schedule_timeout) from [<c0175d38>] (rcu_gp_kthread+0x588/0x9b4)
[35158.802158] [<c0175d38>] (rcu_gp_kthread) from [<c0139d78>] (kthread+0x148/0x150)
[35158.809645] [<c0139d78>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[35158.816865] Exception stack(0xdf481fb0 to 0xdf481ff8)
[35158.821917] 1fa0:                                     00000000 00000000 00000000 00000000
[35158.830095] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[35158.838272] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
```

amazingproducer, 2019-06-13:
I tried turning it on:
```
U-Boot SPL 2017.09-h2 (Jul 06 2018 - 10:16:12)
DRAM: 512 MiB
Trying to boot from MMC1
BPI: skip sunxi_mmc_getcd pin 166(1)

U-Boot 2017.09-h2 (Jul 06 2018 - 10:16:12 -0300) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Banana Pi BPI-M2-Zero
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
BPI: skip sunxi_mmc_getcd pin 166(1)
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
No controllers found
Autoboot in 0 seconds, press <Space> to stop
BPI: skip sunxi_mmc_getcd pin 166(1)
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1219 bytes read in 160 ms (6.8 KiB/s)
## Executing script at 43100000
5696568 bytes read in 510 ms (10.7 MiB/s)
982455 bytes read in 112 ms (8.4 MiB/s)
26872 bytes read in 203 ms (128.9 KiB/s)
## Flattened Device Tree blob at 48000000
   Booting using the fdt blob at 0x48000000
   Loading Device Tree to 49ff6000, end 49fff8f7 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.20.0-m2z-otg-gadget (alex@svn) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #1 SMP Fri Dec 28 09:38:17 -02 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Banana Pi M2 Zero
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x55c00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x3f4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu @(ptrval) s35020 r8192 d22324 u65536
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 374204K/524288K available (8192K kernel code, 341K rwdata, 1960K rodata, 1024K init, 268K bss, 19012K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 342 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 269 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[... 442 lines cut ...]
[    5.269623] 1fa0:                   004ab008 00020000 004a64bc 000a0000 000001b6 000001b6
[    5.274670] 7fa0:                   00504008 00020000 0050078c 000a0000 000001b6 000001b6
[    5.279717] 1fc0: 004ab008 00020000 00000008 00000005 00000000 00000000 004a9eb4 00000000
[    5.287888] 7fc0: 00504008 00020000 00000008 00000005 00000000 00000000 00502eb4 00000000
[    5.296057] 1fe0: 00000000 beb3639c b6dc5dc9 b6e00d36
[    5.304228] 7fe0: 00000000 beb4e35c b6e84dc9 b6ebfd36
[    5.312402] Code: 0a000088 e5942010 e5943000 e5922008 (e5922118) 
[    5.328580] CPU3: stopping
[    5.331290] CPU: 3 PID: 158 Comm: systemd-dbus1-g Tainted: G      D           4.20.0-m2z-otg-gadget #1
[    5.340590] Hardware name: Allwinner sun8i Family
[    5.345299] [<c010e87c>] (unwind_backtrace) from [<c010b978>] (show_stack+0x10/0x14)
[    5.353046] [<c010b978>] (show_stack) from [<c08de9ec>] (dump_stack+0x88/0x9c)
[    5.360272] [<c08de9ec>] (dump_stack) from [<c010dabc>] (handle_IPI+0x1c4/0x1e4)
[    5.367671] [<c010dabc>] (handle_IPI) from [<c04b1f9c>] (gic_handle_irq+0x8c/0x90)
[    5.375243] [<c04b1f9c>] (gic_handle_irq) from [<c0101a0c>] (__irq_svc+0x6c/0x90)
[    5.382724] Exception stack(0xdef31bc8 to 0xdef31c10)
[    5.387777] 1bc0:                   00000000 c0d55b88 00000080 600e0113 c0d5595c c0d0942c
[    5.395956] 1be0: 0000000b def31cd0 bf000000 ffffe000 00000005 df7ac500 00000000 def31c18
[    5.404132] 1c00: c010bb00 c010bb34 600e0113 ffffffff
[    5.409186] [<c0101a0c>] (__irq_svc) from [<c010bb34>] (die+0x1b8/0x34c)
[    5.415890] [<c010bb34>] (die) from [<c0112f54>] (__do_kernel_fault.part.0+0x64/0x74)
[    5.423727] [<c0112f54>] (__do_kernel_fault.part.0) from [<c0112c7c>] (do_bad_area+0x0/0x84)
[    5.432166] [<c0112c7c>] (do_bad_area) from [<00000000>] (  (null))
[    5.438442] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

avafinger, 2019-06-13:
The might have corrupt the file system.

Please, use fsck again. Check board temperature if not too high!
Change /etc/fstab  and add the following:

```
# UNCONFIGURED FSTAB FOR BASE SYSTEM
# /dev/mmcblk0p1	/boot	vfat	defaults			0		2
/dev/mmcblk0p1	/boot	ext4	defaults			0		2
/dev/mmcblk0p2	/	ext4	defaults,noatime,nodiratime,commit=600,errors=remount-ro,x-gvfs-hide		0		1
```

Do not change the partition numbers, the above is from another board.

amazingproducer, 2019-06-13:
I added some stuff to ramdisk, hoping to decrease disk writes, so my current fstab looks like this:
```
/dev/mmcblk0p1  /boot   ext4    defaults,noatime,commit=1800    0               2
/dev/mmcblk0p2  /       ext4    defaults,noatime,commit=1800    0               1
tmpfs   /var/tmp                tmpfs   size=10M,nodev,nosuid   0       0
tmpfs   /var/cache/samba        tmpfs   size=5M,nodev,nosuid    0       0

```
I'll run fsck and make amendments as prescribed.

avafinger, 2019-06-13:
add **nodiratime**

amazingproducer, 2019-06-13:
Just for the root partition?
Also, fsck passed for both partitions.

avafinger, 2019-06-13:
only root.
That's something weird, why the board would not boot after the crash? There would be one explanation that comes to my mind, the file system is corrupt or got corrupt.
kernel 4.20.17 has a lot of fixes, perhaps the one you are facing.

Care to tell the camera you are using?

amazingproducer, 2019-06-13:
The cameras are separate devices on the network providing rtsp streams for the bpi m2z to process. Oops. I added nodiratime to both partitions and got this at boot time:
```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.20.0-m2z-otg-gadget (alex@svn) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #1 SMP Fri Dec 28 09:38:17 -02 2018
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Banana Pi M2 Zero
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x55c00000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x3f4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu @(ptrval) s35020 r8192 d22324 u65536
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 374204K/524288K available (8192K kernel code, 341K rwdata, 1960K rodata, 1024K init, 268K bss, 19012K reserved, 131072K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 342 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 269 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000008] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000020] Switching to timer-based delay loop, resolution 41ns
[    0.000205] Console: colour dummy device 80x30
[    0.000565] printk: console [tty0] enabled
[    0.000612] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000638] pid_max: default: 32768 minimum: 301
[    0.000793] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000816] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001551] CPU: Testing write buffer coherency: ok
[    0.001987] CPU0: update cpu_capacity 1024
[    0.002012] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002547] Setting up static identity map for 0x40100000 - 0x40100060
[    0.002708] rcu: Hierarchical SRCU implementation.
[    0.003435] smp: Bringing up secondary CPUs ...
[    0.014187] CPU1: update cpu_capacity 1024
[    0.014194] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.025028] CPU2: update cpu_capacity 1024
[    0.025034] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.035782] CPU3: update cpu_capacity 1024
[    0.035789] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.035872] smp: Brought up 1 node, 4 CPUs
[    0.035956] SMP: Total of 4 processors activated (192.00 BogoMIPS).
[    0.035970] CPU: All CPU(s) started in HYP mode.
[    0.035981] CPU: Virtualization extensions available.
[    0.036934] devtmpfs: initialized
[    0.042941] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.043180] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.043220] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.047723] pinctrl core: initialized pinctrl subsystem
[    0.048781] NET: Registered protocol family 16
[... 385 lines cut ...]
[    4.498276] [<c0145684>] (cpu_startup_entry) from [<4010244c>] (0x4010244c)
[    4.505250] CPU3: stopping
[    4.507978] CPU: 3 PID: 162 Comm: systemd Tainted: G      D           4.20.0-m2z-otg-gadget #1
[    4.516604] Hardware name: Allwinner sun8i Family
[    4.521333] [<c010e87c>] (unwind_backtrace) from [<c010b978>] (show_stack+0x10/0x14)
[    4.529105] [<c010b978>] (show_stack) from [<c08de9ec>] (dump_stack+0x88/0x9c)
[    4.536358] [<c08de9ec>] (dump_stack) from [<c010dabc>] (handle_IPI+0x1c4/0x1e4)
[    4.543782] [<c010dabc>] (handle_IPI) from [<c04b1f9c>] (gic_handle_irq+0x8c/0x90)
[... cut ...]
[    4.614765] Exception stack(0xde849f00 to 0xde849f48)
[    4.619840] 9f00: 000deeee 800f0013 00000000 01bdddc0 c0d04d08 def22000 bede7328 bede7efc
[    4.628045] 9f20: ded23800 00000000 ffffff9c deeedc00 00000003 de849f50 c0119664 c01137e8
[    4.636240] 9f40: 600f0013 ffffffff
[    4.639756] [<c0101978>] (__dabt_svc) from [<c01137e8>] (pool_allocator_alloc+0x5c/0x84)
[    4.647872] [<c01137e8>] (pool_allocator_alloc) from [<00000000>] (  (null))
[    4.654949] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

avafinger, 2019-06-13:
Do you have another board with you?
I don't think it would make a diference adding to /boot
`/dev/mmcblk0p2  /       ext4    defaults,nodiratime,noatime,commit=1800    0               1`

The current OS Image has mali (CMA with reserved memory), ov5640 and then we have a memory constraint, i will build a new Image without all this stuff and you can try with that.

amazingproducer, 2019-06-13:
I don't have another board, unfortunately, so I can't rule out a hardware failure that way.
I know it's not the best choice for the sd card's health, but would a swap file help with this kind of memory issue?

avafinger, 2019-06-13:
i would use zram instead of swap. I think this would help unless FS is the issue.
Suggestion:
* start all over again with ZRAM and with minimum changes and without motion
* Let it run for a day, just add your samba mounting point and your NFS stuff.

I am not an experienced samba user, i am not even sure samba is stable in this kernel.

amazingproducer, 2019-06-13:
Okay, I'll start over with ZRAM and without motion. Should I keep my current blacklist, or let the default modules loose again?

avafinger, 2019-06-13:
Keep your blacklist.

avafinger, 2019-06-13:
Just push some copy files to check if does not crash while doing large transfer data, something like that.

amazingproducer, 2019-06-13:
The script-built FS caused a crash when I didn't use the blacklist. Interesting. I'll upgrade systemd and kernel to 4.20 now.

amazingproducer, 2019-06-13:
Okay, I'm back on 4.20 with zram and cifs-utils installed. I mounted some stuff and moved some stuff. I'll leave it on and get back to you.

amazingproducer, 2019-06-14:
Last night, I rebooted and it wouldn't come back up. I made several attempts, each time failing at a different point in the startup process. I'll try again from scratch with the card that I'm more confident is good.

avafinger, 2019-06-14:
Can you try this kernel here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v13

This kernel has a fix for:
 * FS / CIFS
 * mm

After you install the /boot should have:

```
lrwxrwxrwx  1 root   root        20 Jan 15  2018 boot.scr -> bpi-m2-zero_boot.scr
lrwxrwxrwx  1 root   root        18 Jun 14 21:10 zImage -> zImage_4.20.17-m2z*
-rwxr-xr-x  1 root   root   5701544 Jun 14 21:09 zImage_4.20.17-m2z*
lrwxrwxrwx  1 root   root        30 Jun 14 21:10 bpi-m2-zero.dtb -> bpi-m2-zero-v4.dtb_4.20.17-m2z

```

Let me know if you have any problem installing it.
Please give some feedback on how it goes.

avafinger, 2019-06-14:
```
ubuntu@bpi-m2z:~$ uname -ra
Linux bpi-m2z 4.20.17-m2z #1 SMP Fri Jun 14 16:10:26 -03 2019 armv7l armv7l armv7l GNU/Linux
ubuntu@bpi-m2z:~$ free -m
              total        used        free      shared  buff/cache   available
Mem:            495          24         243           7         226         440
Swap:           247           0         247
ubuntu@bpi-m2z:~$ 
```

```
ubuntu@bpi-m2z:~$ lsmod
Module                  Size  Used by
lzo                    16384  16
lzo_compress           16384  1 lzo
zram                   24576  4
sunxi_cir              16384  0
rfcomm                 40960  0
hci_uart               36864  0
btintel                16384  1 hci_uart
bluetooth             327680  5 hci_uart,btintel,rfcomm
ecdh_generic           28672  1 bluetooth
brcmfmac              188416  0
brcmutil               16384  1 brcmfmac
g_serial               16384  0
ipv6                  405504  18

```

```
ubuntu@bpi-m2z:~$ dmesg| grep zram
[   10.744810] zram: Added device: zram0
[   10.746439] zram: Added device: zram1
[   10.751337] zram: Added device: zram2
[   10.780294] zram: Added device: zram3
[   10.923802] zram0: detected capacity change from 0 to 64958464
[   10.988244] Adding 63432k swap on /dev/zram0.  Priority:5 extents:1 across:63432k SS
[   10.990133] zram1: detected capacity change from 0 to 64958464
[   11.010781] Adding 63432k swap on /dev/zram1.  Priority:5 extents:1 across:63432k SS
[   11.012623] zram2: detected capacity change from 0 to 64958464
[   11.054779] Adding 63432k swap on /dev/zram2.  Priority:5 extents:1 across:63432k SS
[   11.056652] zram3: detected capacity change from 0 to 64958464
[   11.078088] Adding 63432k swap on /dev/zram3.  Priority:5 extents:1 across:63432k SS

```

amazingproducer, 2019-06-15:
Do I just jump from 4.17.y to 4.20.17?

amazingproducer, 2019-06-15:
So far, so good. ZRAM and cifs are working fine, and the device reliably reboots. I'll see how it fares with some light use - but no motion server.

amazingproducer, 2019-06-15:
It's looking pretty stable so far! Without installing motion, I tried some simple tasks to see if they'd complete:
1. use ffmpeg to transcode all files in a cifs share's subdirectory, saving them elsewhere on that share. 
2. use ffmpeg to transcode an hour of an mjpeg stream received via http, saving the output to cifs share.

Both tasks completed without error.

avafinger, 2019-06-15:
Good to hear this. Make a backup of your SD card and then start pushing..
You are using a good heatsink, right?

Monitor the CPU Temp:
```
Every 5.0s: ./h2-monitor.sh                             Sat Jun 15 18:52:18 2019

CPU1 freq      : 1008 MHz
CPU2 freq      : 816 MHz
CPU3 freq      : 648 MHz
CPU4 freq      : 816 MHz
CPU count      : 4
Governor       : ondemand
Core voltage   : 1.1 V
SOC Temp       : 53 C

```
If Temp is above 80C you will be in trouble...

Attached the monitor
[h2-monitor.sh.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/3293442/h2-monitor.sh.tar.gz)

avafinger, 2019-06-15:
`sudo ./h2-monitor.sh -w`

amazingproducer, 2019-06-15:
I made a little script in .bashrc to change the color of the username in my console's prompt based on the CPU temperature. The heatsink and fan are from an old video card, and keep it at about 37C most of the time. 

I'll make an image at this point and try h2-monitor.sh

amazingproducer, 2019-06-15:
I appended this to my .bashrc:
```
# enable color coding of username based on CPU temperature
cpu_temp()
{
txtcyn='[1;36m' # Cyan
txtgrn='[1;32m' # Green
txtylw='[1;33m' # Yellow
txtred='[1;31m' # Red
HEAT=`/bin/cat /sys/class/thermal/thermal_zone0/temp`
HEAT_COLOR="$txtred"
if [ $HEAT -le 38000 ]
then
   HEAT_COLOR="$txtcyn"
fi

if [ $HEAT -le 45000 -a $HEAT -gt 38000 ]
then
   HEAT_COLOR="$txtgrn"
fi

if [ $HEAT -le 50000 -a $HEAT -gt 45000 ]
then
   HEAT_COLOR="$txtylw"
fi

if [ $HEAT -gt 50000 ]
then
   HEAT_COLOR="$txtred"
fi
echo -e "$HEAT_COLOR"
}
PROMPT_COMMAND='PS1="${debian_chroot:+($debian_chroot)}\[\033$(cpu_temp)\]\u\[\033[1;32m\]@\h:\[\033[01;34m\]\w\[\033[00m\]\$ "'
```

amazingproducer, 2019-06-15:
Post-boot output of h2-monitor.sh:
```
CPU1 freq      : 816 MHz 
CPU2 freq      : 480 MHz 
CPU3 freq      : 1008 MHz 
CPU4 freq      : 648 MHz 
CPU count      : 4  
Governor       : ondemand  
Core voltage   : 1.1 V 
SOC Temp       : 28 C 
```

amazingproducer, 2019-06-16:
It just crashed after several hours of running the motion server - but since I was logging to ram, I don't have the output. I'll try monitor the serial port more diligently.

amazingproducer, 2019-06-16:
I've continued testing, and had some failures with motion - but not as catastrophic as the original failures. After rebooting, the device resumed normal function. I suspect at this point, any additional issues I'm experiencing are unrelated to the new release.

Thanks for maintaining this project; it's really useful for the bandwidth-constrained hobbyist - and thanks for helping me with this issue.

avafinger, 2019-08-08:
@amazingproducer 

**Mainline Kernel 5.3.0-rc3 has been released**

Maybe there is some improvement on the CIFS side, you can check if works better for you.

avafinger, 2019-09-03:
@amazingproducer 

Mainline Kernel 5.3.0-rc7 has been released and deb package installs has been tested and ok..
Sorry for the broken rc6 install.

amazingproducer, 2019-09-04:
I'll try it out as soon as I can. Thanks!