avafinger/bananapi-zero-ubuntu-base-minimal #20: slow wifi with mainline firmware

thiagobucca, 2019-07-03:
Hello,

After applying this update: Kernel 4.20 - H2 firmware with mainline Wifi&BT drivers
Wifi is behaving painfully slow, i can't even wget the newest kernel from console.

Pinging from local network is also dropping several packets

`64 bytes from 192.168.10.118: icmp_seq=415 ttl=64 time=14.695 ms
Request timeout for icmp_seq 416
Request timeout for icmp_seq 417
Request timeout for icmp_seq 418
Request timeout for icmp_seq 419
Request timeout for icmp_seq 420
Request timeout for icmp_seq 421
Request timeout for icmp_seq 422
64 bytes from 192.168.10.118: icmp_seq=421 ttl=64 time=2347.756 ms
64 bytes from 192.168.10.118: icmp_seq=422 ttl=64 time=1347.680 ms
64 bytes from 192.168.10.118: icmp_seq=424 ttl=64 time=14.122 ms
64 bytes from 192.168.10.118: icmp_seq=425 ttl=64 time=13.439 ms
64 bytes from 192.168.10.118: icmp_seq=426 ttl=64 time=1939.792 ms
64 bytes from 192.168.10.118: icmp_seq=427 ttl=64 time=938.632 ms
64 bytes from 192.168.10.118: icmp_seq=428 ttl=64 time=553.320 ms
64 bytes from 192.168.10.118: icmp_seq=429 ttl=64 time=474.615 ms
64 bytes from 192.168.10.118: icmp_seq=431 ttl=64 time=14.953 ms
Request timeout for icmp_seq 432
`
Anyone also experiencing this?

avafinger, 2019-07-18:
@thiagobucca 
Can you share more info about the wifi issue?

Maybe there is a need to update the wifi firmware if there is one newer?

avafinger, 2019-07-18:
Here is my test about the Wifi. Pinging and accessing the **M2Z** (**192.168.254.101**) from a **PC Box** (**192.168.254.11**).
I don't see anything unusual.

Check if you have another Wifi with the same IP or MAC, and ping in reverse to see if the board can return the icmp packs, maybe a routing problem?

**PC Box**
```
alex@db:~$ ping 192.168.254.101
PING 192.168.254.101 (192.168.254.101) 56(84) bytes of data.
64 bytes from 192.168.254.101: icmp_seq=1 ttl=64 time=35.3 ms
64 bytes from 192.168.254.101: icmp_seq=2 ttl=64 time=60.7 ms
64 bytes from 192.168.254.101: icmp_seq=3 ttl=64 time=79.0 ms
64 bytes from 192.168.254.101: icmp_seq=4 ttl=64 time=102 ms
64 bytes from 192.168.254.101: icmp_seq=5 ttl=64 time=29.6 ms
64 bytes from 192.168.254.101: icmp_seq=6 ttl=64 time=46.2 ms
64 bytes from 192.168.254.101: icmp_seq=7 ttl=64 time=69.4 ms
64 bytes from 192.168.254.101: icmp_seq=8 ttl=64 time=90.0 ms
64 bytes from 192.168.254.101: icmp_seq=9 ttl=64 time=115 ms
64 bytes from 192.168.254.101: icmp_seq=10 ttl=64 time=33.7 ms
64 bytes from 192.168.254.101: icmp_seq=11 ttl=64 time=64.1 ms
64 bytes from 192.168.254.101: icmp_seq=12 ttl=64 time=80.6 ms
^C
--- 192.168.254.101 ping statistics ---
12 packets transmitted, 12 received, 0% packet loss, time 11009ms
rtt min/avg/max/mdev = 29.676/67.256/115.795/26.566 ms

```

**ssh to M2Z**
```
alex@db:~$ ssh ubuntu@192.168.254.101
The authenticity of host '192.168.254.101 (192.168.254.101)' can't be established.
ECDSA key fingerprint is SHA256:HL81ibO27vRldHLBULf0LMUpeb4gHGSD0+8r0k39a8s.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.254.101' (ECDSA) to the list of known hosts.
ubuntu@192.168.254.101's password: 
Welcome to Ubuntu 16.04.6 LTS (GNU/Linux 4.20.17 armv7l)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
Last login: Sun Jun 23 22:26:38 2019
ubuntu@bpi-m2z:~$ uname -ra
Linux bpi-m2z 4.20.17 #1 SMP Wed Jun 19 10:04:44 -03 2019 armv7l armv7l armv7l GNU/Linux
ubuntu@bpi-m2z:~$  
ubuntu@bpi-m2z:~$ lsmod
Module                  Size  Used by
lzo                    16384  16
lzo_compress           16384  1 lzo
zram                   24576  4
snd_soc_hdmi_codec     16384  1
dw_hdmi_i2s_audio      16384  0
dw_hdmi_cec            16384  0
sun8i_drm_hdmi         20480  0
snd_soc_simple_card    16384  0
sun8i_codec_analog     24576  1
sunxi_cir              16384  0
snd_soc_simple_card_utils    16384  1 snd_soc_simple_card
dw_hdmi                28672  2 sun8i_drm_hdmi,dw_hdmi_i2s_audio
cec                    49152  2 dw_hdmi_cec,dw_hdmi
sun8i_adda_pr_regmap    16384  1 sun8i_codec_analog
mali                  217088  0
sunxi_cedrus           36864  0
sun4i_i2s              24576  2
sun4i_codec            36864  3
videobuf2_dma_contig    20480  1 sunxi_cedrus
v4l2_mem2mem           20480  1 sunxi_cedrus
videobuf2_memops       16384  1 videobuf2_dma_contig
videobuf2_v4l2         20480  2 sunxi_cedrus,v4l2_mem2mem
videobuf2_common       36864  3 sunxi_cedrus,v4l2_mem2mem,videobuf2_v4l2
videodev              147456  4 sunxi_cedrus,videobuf2_common,v4l2_mem2mem,videobuf2_v4l2
sun8i_mixer            36864  0
media                  28672  5 sunxi_cedrus,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
sun4i_drm              16384  1
sun4i_frontend         16384  1 sun4i_drm
sun4i_tcon             28672  1 sun4i_drm
sun8i_tcon_top         16384  3 sun8i_drm_hdmi,sun4i_drm,sun4i_tcon
rfcomm                 40960  0
hci_uart               36864  0
btintel                16384  1 hci_uart
bluetooth             327680  5 hci_uart,btintel,rfcomm
ecdh_generic           28672  1 bluetooth
brcmfmac              188416  0
brcmutil               16384  1 brcmfmac
g_serial               16384  0
ipv6                  405504  18
ubuntu@bpi-m2z:~$ ping 192.168.254.11
PING 192.168.254.11 (192.168.254.11) 56(84) bytes of data.
64 bytes from 192.168.254.11: icmp_seq=1 ttl=64 time=6.03 ms
64 bytes from 192.168.254.11: icmp_seq=2 ttl=64 time=11.0 ms
64 bytes from 192.168.254.11: icmp_seq=3 ttl=64 time=7.58 ms
64 bytes from 192.168.254.11: icmp_seq=4 ttl=64 time=7.33 ms
64 bytes from 192.168.254.11: icmp_seq=5 ttl=64 time=6.96 ms
64 bytes from 192.168.254.11: icmp_seq=6 ttl=64 time=5.34 ms

--- 192.168.254.11 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5008ms
rtt min/avg/max/mdev = 5.342/7.390/11.083/1.823 ms
^Cubuntu@bpi-m2z:~$ dmesg|grep err
[    2.428608] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    2.796204] EXT4-fs (mmcblk0p2): warning: mounting fs with errors, running e2fsck is recommended
[    5.419231] EXT4-fs (mmcblk0p2): re-mounted. Opts: commit=600,errors=remount-ro
[    5.637812] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.clm_blob failed with error -2
[    7.094367] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-11), device may have limited channels available
ubuntu@bpi-m2z:~$ dmesg|grep brc
[    5.395362] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.628915] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.637812] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.clm_blob failed with error -2
[    5.648053] brcmfmac mmc1:0001:1: Falling back to syfs fallback for: brcm/brcmfmac43430-sdio.clm_blob
[    7.094367] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-11), device may have limited channels available
[    7.100307] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Mar 30 2016 11:30:56 version 7.45.77.h8.4 FWID 01-ee8a6268
ubuntu@bpi-m2z:~$ 

```

avafinger, 2019-07-18:
I think the board never turns off, check the filesystem integrity.

umount an check both partitions:
sudo fsck.ext4 -f /dev/sdc1
sudo fsck.ext4 -f /dev/sdc2

thiagobucca, 2019-07-27:
i just flashed the latest image you provided and the issue remains

some more information:

dmesg output

`[    2.416212] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    5.446073] EXT4-fs (mmcblk0p2): re-mounted. Opts: commit=600,errors=remount-ro
[    5.613533] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.clm_blob failed with error -2
[    6.885245] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-11), device may have limited channels available
`

still dropping several packets, here's the ping from my macbook:

`4 bytes from 192.168.10.115: icmp_seq=280 ttl=64 time=57.809 ms
64 bytes from 192.168.10.115: icmp_seq=281 ttl=64 time=73.273 ms
64 bytes from 192.168.10.115: icmp_seq=282 ttl=64 time=95.087 ms
64 bytes from 192.168.10.115: icmp_seq=283 ttl=64 time=116.032 ms
64 bytes from 192.168.10.115: icmp_seq=284 ttl=64 time=110.996 ms
Request timeout for icmp_seq 285
Request timeout for icmp_seq 286
64 bytes from 192.168.10.115: icmp_seq=287 ttl=64 time=17.114 ms
64 bytes from 192.168.10.115: icmp_seq=288 ttl=64 time=956.804 ms
Request timeout for icmp_seq 289
Request timeout for icmp_seq 290
Request timeout for icmp_seq 291
Request timeout for icmp_seq 292
Request timeout for icmp_seq 293
Request timeout for icmp_seq 294
Request timeout for icmp_seq 295
Request timeout for icmp_seq 296
64 bytes from 192.168.10.115: icmp_seq=297 ttl=64 time=19.913 ms
64 bytes from 192.168.10.115: icmp_seq=298 ttl=64 time=119.488 ms
64 bytes from 192.168.10.115: icmp_seq=299 ttl=64 time=47.660 ms
Request timeout for icmp_seq 300
Request timeout for icmp_seq 301
Request timeout for icmp_seq 302
`

i started fresh and the just connected to my wifi, i have several other SBC boards and the issue doesn't occur so it maybe something with this specific board.

any other info i cloud provide?

thiagobucca, 2019-07-27:
i started fresh with the latest image you provided and everything is working fine now, thanks.

closed 2019-07-27 by thiagobucca