avafinger/bananapi-zero-ubuntu-base-minimal #21: [Q] How to flash last version?

namero999, 2019-07-19:
Hello! First of all, thanks so much for this effort. The official support is not even close to what you have done here! Thanks!!

I have a question regarding how to flash your image. It is not clear to me if I have to start from V1 and the perform a sequence of updates till the last version, or if it is possible to get/build an image with the last version and flash directly that one.

What do you suggest?

avafinger, 2019-07-19:
I will provide a fresh IMG, i think would be the best way after so many changes.
I think i can do it tomorrow...

avafinger, 2019-07-20:
@namero999 
You can try both ways of Burning the SD CARD from here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v15

wpiecek, 2019-07-20:
I try to use a new image (both - img and from modules) and on BPI-M2Z they hang. Same image on Orange Pi (same H2+) boot correct and work stable (except no wifi).

namero999, 2019-07-20:
Thanks man, much appreciated. I will test it tonight.

avafinger, 2019-07-20:
@wpiecek 

> I try to use a new image (both - img and from modules) and on BPI-M2Z they hang. Same image on Orange Pi (same H2+) boot correct and work stable (except no wifi).

That's interesting. Did i understand it correctly, you booted this IMG on the OrangePi and it works fine except for the wifi? Which OrangePi board and what is the Wifi chip?

Can you grab the bootlog from BPI-M2Z using the TTL debug?
Did you try the old IMG also?

IIs it a new board? Can you attach an Image of your BPI-M2Z with a good resolution so we can check Board version, etc.....

wpiecek, 2019-07-20:
[banana_pi-m2z.zip](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/3413623/banana_pi-m2z.zip)
[orange_pi-zero.zip](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/3413624/orange_pi-zero.zip)

Logs attached. 
For banana, there is something strange - they hold not always is same state. I try reboot few times and sometimes they gan got few steps away - I catch two logs. Mostly, they hang after 'starting time synchronization' (from memory). But once, they start reboot all time until I poweroff. 
When they hold, led is off.

For Orange - boot correct. Attached.

wpiecek, 2019-07-20:
[image]
[image]
[image]
[image]
and there is picture of both boards. Banana have a heat sink - i can remove if they can help.

wpiecek, 2019-07-20:
ok, i tested on another bpi m2z (yeap, I have another) and they work.
I'm not planned to solder serial pins, but because they work, I can dump everything from working machine. 
Any ideas?

avafinger, 2019-07-20:
No need to remove the heat sink. I have the same HW revision as yours, but your board looks pretty bad compared to mine.

Looking at the boot log i can see many things that could be the cause:

a) codec-analog failed to register
b) hdmi sub-systems failed (hdmi + mali (GPU) + cedrus (VPU))
c) Network failed
d) IR

the correct events that failed on your board are:
```
[    7.434588] sun4i-codec 1c22c00.codec: ASoC: codec-analog@1f015c0 not registered
[    7.442139] sun4i-codec 1c22c00.codec: Failed to register our card
[    7.454310] Allwinner sunXi mali glue initialized
[    7.471724] sun4i-codec 1c22c00.codec: ASoC: Failed to create component debugfs directory
[    7.473935] sun4i-codec 1c22c00.codec: Codec <-> 1c22c00.codec mapping ok
[    7.477558] Registered IR keymap rc-empty
[    7.477733] rc rc0: sunxi-ir as /devices/platform/soc/1f02000.ir/rc/rc0
[    7.477903] input: sunxi-ir as /devices/platform/soc/1f02000.ir/rc/rc0/input5
[    7.479521] Mali: 
[    7.479532] Found Mali GPU Mali-400 MP r1p1
[    7.483001] rc rc0: lirc_dev: driver sunxi-ir registered at minor = 0, raw IR receiver, no transmitter
[    7.483406] sunxi-ir 1f02000.ir: initialized sunXi IR driver
[    7.494992] sun4i-drm display-engine: bound 1100000.mixer (ops sun8i_mixer_platform_driver_exit [sun8i_mixer])
[    7.498132] sun4i-drm display-engine: bound 1c0c000.lcd-controller (ops sun4i_tcon_platform_driver_exit [sun4i_tcon])
[    7.498233] sun8i-dw-hdmi 1ee0000.hdmi: 1ee0000.hdmi supply hvcc not found, using dummy regulator
[    7.498329] sun8i-dw-hdmi 1ee0000.hdmi: Linked as a consumer to regulator.0
[    7.499631] Mali: 
[    7.499639] 2+0 PP cores initialized
[    7.500565] Mali: 
[    7.500569] Mali device driver loaded
[    7.516885] sun8i-dw-hdmi 1ee0000.hdmi: Detected HDMI TX controller v1.32a with HDCP (sun8i_dw_hdmi_phy)
[    7.526046] sun8i-dw-hdmi 1ee0000.hdmi: registered DesignWare HDMI I2C bus driver
[    7.526659] sun4i-drm display-engine: bound 1ee0000.hdmi (ops sun8i_dw_hdmi_ops [sun8i_drm_hdmi])
[    7.526681] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    7.526685] [drm] No driver support for vblank timestamp query.
[    8.157486] Console: switching to colour frame buffer device 240x67
[    8.197288] sun4i-drm display-engine: fb0: DRM emulated frame buffer device
[    8.213549] [drm] Initialized sun4i-drm 1.0.0 20150629 for display-engine on minor 0
[    8.253390] asoc-simple-card soc:sound: i2s-hifi <-> 1c22800.i2s mapping ok
[    9.046781] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)

```

Judging by how your board looks i would risk saying you run the board at extremely high temperature that may have caused some damages, but it is just a guess, do you have a spare BPI-M2Z to test with?

Maybe @namero999  could test on his board and give some feed-back.

BTW, OrangePi ZERO have XR819 wifi chip , the mainline support for this chip is not good despite many efforts from kernel developers.

One thing you could try:
a) Remove eth0 (add comments) to the file: /etc/network/interfaces
```
# interfaces(5) file used by ifup(8) and ifdown(8)
auto lo
iface lo inet loopback

# auto eth0
# iface eth0 inet dhcp

```

and also disable wlan0 and see if boots

namero999, 2019-07-21:
The image boots, I could get a prompt. Unfortunately I was without external keyboard. I will acquire a ttl converter a try again today. One thing I noticed is that the red led blinks... Is that normal?

avafinger, 2019-07-21:
@namero999 
**That means it is working!** 

Red blinking means kernel is up and running. Just do an **ssh** to the board IP like so:
`ssh ubuntu@xxx.xxx.xxx where xxx.xxx.xxx is the IP,`
Credentials: ubuntu/ubuntu

namero999, 2019-07-21:
[boot.zip](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/3414641/boot.zip)

Works like a charm. I will now play with it and report any issue with networking (I haven't connected to the network yet)

namero999, 2019-07-21:
I successfully edited `/etc/network/interfaces` and had the wifi working. With an airgain antenna and connected to a nearby hotspot, it peaked 2.31 MB/s and did not lose connection.

One thing I noticed though, is that it couldn't connect to an open network. Not sure if I have to comment out `wpa-psk` or leave it there but with an empty value...

Have you noticed anything strange in the bootlog? As far as I can tell, this image works great for my needs! (headless ssh server)

I will experiment more, the final goal is to have an influxdb + grafana server running.

In the meantime, THANK YOU SO MUCH! This should be the official image and banana pi should acknowledge and appreciate your work!

namero999, 2019-07-21:
For now the only issue I could find is the login at the desktop environment (not really an issue for me, but I wanted to test anyway). Following your instructions, I installed LXDE and at reboot it shows correctly the login screen. But after entering the password, I get `Failed to create session`. Same happens with guest session.

I tried to do this https://cialu.net/how-to-solve-failed-to-start-session-with-lightdm-and-xfce/ (it is the same error I get) but it didn't help.

Not a blocker for me, but I just wanted to let you know!

avafinger, 2019-07-21:
> But after entering the password, I get Failed to create session

I think this is a known issue.
Try to select **OpenBox** , login, then logout, shutdown, reboot and login with LXDE, that will fix it.

avafinger, 2019-07-21:
Just remember that Mali is set to GBM which will NOT work with Desktop.

namero999, 2019-07-21:
One thing I noticed in my bootlog, when comparing to your snippet, is that my sound device failed to initialize.

```
[    7.438054] sun4i-codec 1c22c00.codec: ASoC: codec-analog@1f015c0 not registered
[    7.445623] sun4i-codec 1c22c00.codec: Failed to register our card
[    7.449714] sun4i-codec 1c22c00.codec: ASoC: codec-analog@1f015c0 not registered
[    7.461511] sun4i-codec 1c22c00.codec: Failed to register our card
```
I also don't have any line regarding Mali, which instead appears in your log.

Is there anything that I can do to debug the problem? I'm definitely not an expert in these kind of things...

I booted the zero without an HDMI cable pugged in, although I'm not sure this affects anything...

namero999, 2019-07-21:
> > But after entering the password, I get Failed to create session
> 
> I think this is a known issue.
> Try to select **OpenBox** , login, then logout, shutdown, reboot and login with LXDE, that will fix it.

I think I tried that with the same result, but I'll try again...

avafinger, 2019-07-21:
`[    7.461511] sun4i-codec 1c22c00.codec: Failed to register our card`

Hmm, this error is very similar to @wpiecek has seen, but i think you should boot with HDMI plugged in, and also could be an issue with your mini HDMI to HDMI. I use a mini/micro HDMI to HDMI extension with high quality, you should try with another one or try with another Monitor and see what gives.

namero999, 2019-07-22:
Indeed I tried to boot the machine plugged with an HDMI cable to a monitor and I couldn't find the warning in the logs anymore, plus I can confirm that the audio works!

I couldn't fix the session problem instead. I tried to install a full fledged lubuntu desktop environment (in the hope that it would pull more required packages) and this way the login worked.

Anyway I will go on experimenting with the rom and report here any interesting finding! :)

namero999, 2019-08-28:
For the records, I gathered some notes relative to this image here https://github.com/namero999/banana-pi-zero-notes

closed 2019-08-28 by namero999