avafinger/bananapi-zero-ubuntu-base-minimal #24: WiringPi or RPi.GPIO

lbernstone, 2019-08-08:
Is it possible to use WiringPi or RPi.GPIO with your images?

avafinger, 2019-08-08:
Yes if you use WiringPi for the mainline kernel that has support for m2z. WiringPi from Rpi check for hardware compatibility and you need to disable this "check" in the code. There may be other changes needed.
Check the differences here: https://github.com/BPI-SINOVOIP/BPI-WiringPi2

xqdzn, 2019-08-08:
Hi @lbernstone,
You can just compile it from source then append these line to `/etc/environment`
[code]
BOARD=bpi-m2z
BOARD_AUTO=bpi-m2z
BOARD_OLD=bpi-m64
[/code]
Cheers!

lbernstone, 2019-08-09:
I am able to compile wiringPi (and don't get the linker error), but it does not actually change the pin states.  I run `gpio readall`, and it shows me all the pins set as INPUTs.  I try to set a pin to OUTPUT as root (I did a for x in {1..277} just to be thorough), and then when I read the state again, nothing has changed.

lbernstone, 2019-08-09:
@xqdzn You were on the right track, but wrong location.  It looks in /var/lib/bananapi/boards.sh for the info:
```
BOARD=bpi-m2z
BOARD_AUTO=bpi-m2z
BOARD_OLD=bpi-m64
```

Now to figure out how to get rpi.gpio to work for a user :smile:

closed 2019-08-09 by lbernstone