avafinger/bananapi-zero-ubuntu-base-minimal #25: dtb overlay?

lbernstone, 2019-09-01:
Is there any way to use dtb overlays with this image, or do they need to be merged into the main dtb?

avafinger, 2019-09-01:
Currently, there is no support for dtb overlays. I will try to add support on next build with U-boot and Kernel update RC6.

avafinger, 2019-09-01:
Here is the support for DT Overlays. Assuming you asked for and known how to use DTO.
There is support for DTO on boot time.

Please, make a backup and try this:

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v1.7

avafinger, 2019-09-01:
Updated with a sample, I think this should work.

avafinger, 2019-09-04:
Install fix.
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v1.8

lbernstone, 2019-10-19:
Sorry it took me so long to get back to this.
So, in order to use an overlay, I am going to have to rebuild the whole Uboot image?  This seems like a much more intensive process than just recreating the dtb.  Is it possible to use a wildcard in the ext4load line so it looks something like:
`ext4load mmc 0 ${fdtovaddr} overlays/*.dto`
and then any files added into the overlays folder will be included?

avafinger, 2019-10-19:
You need to update u-boot and kernel as described here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v1.8 (or later)
and https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v1.7

Not sure if * will load all, but you can give it a try.

No need to rebuild anything, it is already built..

avafinger, 2019-10-19:
or you can store the files in a variable like this:

```
overlay=file1.dto file2.dto file3.dto
for ov in ${overlay}; do
    echo overlaying ${ov}...
    load ${devtype} ${devnum}:${bootpart} ${fdtovaddr} /boot/overlays/${ov}.dtb && fdt apply ${fdtovaddr}
done
```

lbernstone, 2019-10-20:
I would have to run mkimage every time to change the overlays.  This is not really useful for my use case (changing a couple lines in the dt).  The overlays work, so I'll go ahead and close the issue.

closed 2019-10-20 by lbernstone

avafinger, 2019-10-20:
> I would have to run mkimage every time to change the overlays. 

No need to run **mkimage** at any time. Just change **dts** and recompile, that's all you need.