avafinger/bananapi-zero-ubuntu-base-minimal #26: OTG port not working on 5.3.0-rc8 kernel

thiagobucca, 2019-09-14:
Hello again @avafinger 

Can you please confirm if the USB OTG port is working on latest released kernel? I was previously using a bluetooth 4.0 dongle but after upgrading the kernel it doesn't show anymore in lsusb or whatsoever.

avafinger, 2019-09-14:
Hi @thiagobucca 

Can you please tell which kernel version was working?
I have not changed OTG configuration.

```
&usb_otg {
	dr_mode = "otg";
	status = "okay";
};

```

BTW, why are you using a bluetooth 4.0 dongle, the BT from the board isn't working for you?
You can always revert to the previous kernel by doing this:
* symlink to the kernel version you what to restore:
```
cd /boot
sudo ln -sf zImage_5.3.0-rc7-m2z zImage
sudo ln -sf bpi-m2-zero-v4.dtb_5.3.0-rc7-m2z  bpi-m2-zero.dtb

```

check before reboot:

```
ls -la
lrwxrwxrwx  1 root root      32 Set  3 09:14 bpi-m2-zero.dtb -> bpi-m2-zero-v4.dtb_5.3.0-rc7-m2z
lrwxrwxrwx  1 root root      20 Set  3 09:14 zImage -> zImage_5.3.0-rc7-m2z

```

* reboot
`sync && sudo reboot`

Maybe you can symlink to the working kernel and grab dmesg **before** and **after** you attach the usb dongle, might be related to kernel config.

thiagobucca, 2019-09-14:
I just found out that my problem has nothing to do with the image, i'm sorry. I just checked my board and the OTG port has a loose connector, after resoldering it worked fine.

I use a bluetooth adapter because onboard bluetooth cannot detect my miflora BLE sensors that i use with my plants, not the best option but it works.

Anyway, thanks for the attention and sorry again for the false alarm.

closed 2019-09-14 by thiagobucca

avafinger, 2019-09-14:
Don't know much about **miflora BLE sensors**, but BT on bpi m2z is 4.1.
You should see something when you do a :

`sudo hcitool lescan`

Maybe some kernel config missing, but CONFIG_BT_LE=y is there.

thiagobucca, 2019-09-15:
Actually i tried to use onboard BT but following the instructions on the main page, my board never go past the following line:

writing
01 03 0c 00 

i noticed i didn't have the file named  bcm43438a1.hcd on /lib/firmware/ap6212 path so i downloaded it from SINOVOIP's github, but i really don't know if this is the problem.

avafinger, 2019-09-15:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/tree/firmware-v1.0

avafinger, 2019-09-24:
@thiagobucca 
You can test a fix for the BT and HDMI, please refer to: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/27#issuecomment-534787251