avafinger/bananapi-zero-ubuntu-base-minimal #27: Kernel 5.3.1 and bluetooth

Bahir, 2019-09-24:
Board: BPI M2 Zero.

I'm trying to setup bluetooth.
The same command worked fine with 4.20.17

`brcm_patchram_plus -d --patchram /lib/firmware/ap6212/bcm43438a1.hcd --enable_hci --bd_addr b8:27:eb:b7:f9:60 --no2bytes --tosleep 5000 /dev/ttyS1
`

With 5.3.1 it produced infinite output

`writing
01 03 0c 00
writing
01 03 0c 00
writing
01 03 0c 00
writing
01 03 0c 00
writing
01 03 0c 00`

And is not finished with "Done setting line discpline"

avafinger, 2019-09-24:
ok, i just gave it a new try, firmware (bt) is missing for the mainline and not to mention that HDMI is broken.
I will need more time to find the fix
In case you want to search for the bt firmware and put it 
in the right place, attached is the correct BT dtb configuration. (but HDMI is broken).

[bpi-m2-zero-v4.dtb_5.3.1.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/3647652/bpi-m2-zero-v4.dtb_5.3.1.tar.gz)
copy it to /boot and overwrite the old dtb.

```
[  471.282066] Bluetooth: hci0: command 0x1003 tx timeout
[  473.362081] Bluetooth: hci0: command 0x1001 tx timeout
[  475.442152] Bluetooth: hci0: command 0x1009 tx timeout
[  481.282173] Bluetooth: hci0: command 0x1003 tx timeout
[  483.362299] Bluetooth: hci0: command 0x1001 tx timeout
[  485.442222] Bluetooth: hci0: command 0x1009 tx timeout
[  489.714481] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[  489.714496] Bluetooth: BNEP filters: protocol multicast
[  489.714527] Bluetooth: BNEP socket layer initialized
[  523.043304] random: crng init done
[  523.043356] random: 7 urandom warning(s) missed due to ratelimiting

```

avafinger, 2019-09-24:
**Bluetooth** and **HDMI** fixed. No need to patchram for 5.3.1.
I will push the changes asap.

```
ubuntu@bpi-m2z:~$ uname -ra
Linux bpi-m2z 5.3.1 #2 SMP Tue Sep 24 18:23:12 -03 2019 armv7l armv7l armv7l GNU/Linux
ubuntu@bpi-m2z:~$ rfkill list
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
ubuntu@bpi-m2z:~$ hcitool devi
Devices:
	hci0	AA:AA:AA:AA:AA:AA
ubuntu@bpi-m2z:~$ 

```

avafinger, 2019-09-25:
Here is the fix: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.2

Bahir, 2019-09-25:
Thank you all is good except...

```
bluetoothd: Can't open input device: No such file or directory (2)
bluetoothd: AVRCP: failed to init uinput for 00:16:44:FD:6B:A0
```

Looks like uinput kernel module missing.

```
Device Drivers  --->
   Input device support  --->
      [*]   Miscellaneous devices  --->
         <M>   User level driver support
```

Could you please make this module?

avafinger, 2019-09-25:
done.
[rootfs_modules_5.3.1.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/3654277/rootfs_modules_5.3.1.tar.gz)

Install with:
sudo tar -xvpzf rootfs_modules_5.3.1.tar.gz -C / --numeric-ow

avafinger, 2019-10-05:
I have played with BT and unfortunately, there is some issue.
I think we have to use the patchram, the old way.

Here is my failed attempt to connect  to a JBL speaker and Phone:

```
 bluetoothctl -a
[NEW] Controller CC:B8:A8:A9:E5:67 bpi-m2z [default]
Agent registered
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# agent on
Agent is already registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# scan on
Discovery started
[CHG] Controller CC:B8:A8:A9:E5:67 Discovering: yes
[NEW] Device 8E:06:78:D2:E1:3C CHARGE3
[NEW] Device 00:17:CA:F7:38:18 00-17-CA-F7-38-18
[CHG] Device 00:17:CA:F7:38:18 LegacyPairing: no
[CHG] Device 00:17:CA:F7:38:18 Name: Haier HW-W910
[CHG] Device 00:17:CA:F7:38:18 Alias: Haier HW-W910
[bluetooth]# scan off
[CHG] Device 00:17:CA:F7:38:18 RSSI is nil
[CHG] Device 8E:06:78:D2:E1:3C RSSI is nil
Discovery stopped
[CHG] Controller CC:B8:A8:A9:E5:67 Discovering: no
[bluetooth]# trust 8E:06:78:D2:E1:3C 
[CHG] Device 8E:06:78:D2:E1:3C Trusted: yes
Changing 8E:06:78:D2:E1:3C trust succeeded
[bluetooth]# pair 8E:06:78:D2:E1:3C 
Attempting to pair with 8E:06:78:D2:E1:3C
[CHG] Device 8E:06:78:D2:E1:3C Connected: yes
[CHG] Device 8E:06:78:D2:E1:3C UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device 8E:06:78:D2:E1:3C UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 8E:06:78:D2:E1:3C UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device 8E:06:78:D2:E1:3C Paired: yes
Pairing successful
[CHG] Device 8E:06:78:D2:E1:3C Connected: no
[bluetooth]# connect 8E:06:78:D2:E1:3C 
Attempting to connect to 8E:06:78:D2:E1:3C
Failed to connect: org.bluez.Error.Failed
[bluetooth]# connect 8E:06:78:D2:E1:3C 
Attempting to connect to 8E:06:78:D2:E1:3C
Failed to connect: org.bluez.Error.Failed
[bluetooth]# devices
Device 8E:06:78:D2:E1:3C CHARGE3
Device 00:17:CA:F7:38:18 Haier HW-W910
[bluetooth]# trust 00:17:CA:F7:38:18 
[CHG] Device 00:17:CA:F7:38:18 Trusted: yes
Changing 00:17:CA:F7:38:18 trust succeeded
[bluetooth]# pair 00:17:CA:F7:38:18
Attempting to pair with 00:17:CA:F7:38:18
[CHG] Device 00:17:CA:F7:38:18 Connected: yes
Request confirmation
[agent] Confirm passkey 750807 (yes/no): yes
[Haier HW-W910]# connect 00:17:CA:F7:38:18
Attempting to connect to 00:17:CA:F7:38:18
Failed to pair: org.bluez.Error.AuthenticationFailed
Failed to connect: org.bluez.Error.Failed
[CHG] Device 00:17:CA:F7:38:18 Connected: no

```

The connect fail. Something is missing. i made sure pulseaudio was running.

Bahir, 2019-10-05:
Mine is good with connection and pairing but I can't make it work with sound playing.

```
[    6.837977] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.870887] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.sinovoip,bananapi zero.txt failed with error -2
[    6.882750] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43430-sdio.sinovoip,bananapi zero.txt
[    6.990811] Bluetooth: Core ver 2.22
[    6.999210] Bluetooth: HCI device and connection manager initialized
[    6.999245] Bluetooth: HCI socket layer initialized
[    7.010583] Bluetooth: L2CAP socket layer initialized
[    7.015775] Bluetooth: SCO socket layer initialized
[    7.038011] Bluetooth: HCI UART driver ver 2.3
[    7.044249] Bluetooth: HCI UART protocol H4 registered
[    7.049571] Bluetooth: HCI UART protocol BCSP registered
[    7.054926] Bluetooth: HCI UART protocol ATH3K registered
[    7.060629] Bluetooth: HCI UART protocol Intel registered
[    7.067018] Bluetooth: HCI UART protocol Broadcom registered
[    7.074003] Bluetooth: HCI UART protocol AG6XX registered
[    7.113685] Bluetooth: RFCOMM TTY layer initialized
[    7.118605] Bluetooth: RFCOMM socket layer initialized
[    7.123879] Bluetooth: RFCOMM ver 1.11
[    7.247370] Bluetooth: hci0: BCM: chip id 94
[    7.260078] Bluetooth: hci0: BCM: features 0x2e
[    7.288210] Bluetooth: hci0: BCM43430A1
[    7.292895] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[    9.260053] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    9.993284] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[   11.034933] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0106
[   15.722882] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.722896] Bluetooth: BNEP filters: protocol multicast
[   15.722926] Bluetooth: BNEP socket layer initialized
```

```
root@bpi-m2z:~# rfkill list all
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```
```
root@bpi-m2z:~# hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: 8C:F7:10:4B:29:08  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN
	RX bytes:4300 acl:29 sco:0 events:289 errors:0
	TX bytes:37453 acl:30 sco:0 commands:253 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
	Link policy: RSWITCH SNIFF
	Link mode: SLAVE ACCEPT
	Name: 'bpi-m2z'
	Class: 0x200000
	Service Classes: Audio
	Device Class: Miscellaneous,
	HCI Version: 4.0 (0x6)  Revision: 0x6a
	LMP Version: 4.0 (0x6)  Subversion: 0x2209
	Manufacturer: Broadcom Corporation (15)
```

```
root@bpi-m2z:~# pulseaudio --start

Oct  5 13:43:59 bpi-m2z bluetoothd[463]: Endpoint registered: sender=:1.62 path=/MediaEndpoint/A2DPSource/VENDOR/LDAC
Oct  5 13:43:59 bpi-m2z bluetoothd[463]: Endpoint registered: sender=:1.62 path=/MediaEndpoint/A2DPSource/VENDOR/APTXHD
Oct  5 13:43:59 bpi-m2z bluetoothd[463]: Endpoint registered: sender=:1.62 path=/MediaEndpoint/A2DPSource/VENDOR/APTX
Oct  5 13:43:59 bpi-m2z bluetoothd[463]: Endpoint registered: sender=:1.62 path=/MediaEndpoint/A2DPSource/AAC
Oct  5 13:43:59 bpi-m2z bluetoothd[463]: Endpoint registered: sender=:1.62 path=/MediaEndpoint/A2DPSource/SBC
[... 125 lines cut ...]
txpwrbckof=6
cckbw202gpo=0x5555
legofdmbw202gpo=0x77777777
mcsbw202gpo=0xaaaaaaaa

# OFDM IIR :
ofdmdigfilttype=7
# PAPD mode:
papdmode=2

il0macaddr=00:90:4c:c5:12:38
wl0id=0x431b

#OOB parameters
hostwake=0x40
hostrdy=0x41
usbrdy=0x03
usbrdydelay=100
deadman_to=0xffffffff
# muxenab: 0x1 for UART enable, 0x10 for Host awake
muxenab=0x10
# CLDO PWM voltage settings - 0x4 - 1.1 volt
#cldo_pwm=0x4
glitch_based_crsmin=1
```

avafinger, 2019-10-05:
Interesting. I could see the first time you tried to connect failed. And the next attempt was a success. 
I can't remember where I got the firmware from...
My attempt:

```
sudo bt-device -c 8E:06:78:D2:E1:3C
Connecting to: 8E:06:78:D2:E1:3C
Error: GDBus.Error:org.bluez.Error.AlreadyExists: Already Exists
ubuntu@bpi-m2z:~$ sudo bt-device -c 8E:06:78:D2:E1:3C
Connecting to: 8E:06:78:D2:E1:3C
Error: GDBus.Error:org.bluez.Error.AlreadyExists: Already Exists
ubuntu@bpi-m2z:~$ sudo bluetoothctl
[NEW] Controller CC:B8:A8:A9:E5:67 bpi-m2z [default]
[NEW] Device 8E:06:78:D2:E1:3C CHARGE3
[bluetooth]# remove 8E:06:78:D2:E1:3C
[DEL] Device 8E:06:78:D2:E1:3C CHARGE3
Device has been removed
[bluetooth]# exit
[DEL] Controller CC:B8:A8:A9:E5:67 bpi-m2z [default]
ubuntu@bpi-m2z:~$ sudo bt-device -c 8E:06:78:D2:E1:3C
Connecting to: 8E:06:78:D2:E1:3C
Error: Device not found.
ubuntu@bpi-m2z:~$ sudo bt-device -l 8E:06:78:D2:E1:3C
No devices found
ubuntu@bpi-m2z:~$ sudo bt-device -l                  
No devices found
ubuntu@bpi-m2z:~$ sudo bluetoothctl
[NEW] Controller CC:B8:A8:A9:E5:67 bpi-m2z [default]
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# agent on
Agent registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# scan on
Discovery started
[CHG] Controller CC:B8:A8:A9:E5:67 Discovering: yes
[NEW] Device 8E:06:78:D2:E1:3C CHARGE3
[bluetooth]# scan off
[CHG] Device 8E:06:78:D2:E1:3C RSSI is nil
Discovery stopped
[CHG] Controller CC:B8:A8:A9:E5:67 Discovering: no
[bluetooth]# quit
Agent unregistered
[DEL] Controller CC:B8:A8:A9:E5:67 bpi-m2z [default]
ubuntu@bpi-m2z:~$ sudo bluetoothctl
[NEW] Controller CC:B8:A8:A9:E5:67 bpi-m2z [default]
[NEW] Device 8E:06:78:D2:E1:3C CHARGE3
[bluetooth]# quit
[DEL] Controller CC:B8:A8:A9:E5:67 bpi-m2z [default]
```

Then:

```
ubuntu@bpi-m2z:~$ sudo bt-device -c 8E:06:78:D2:E1:3C
Connecting to: 8E:06:78:D2:E1:3C
Done

```

I read somewhere you must have the device listed here in order to play sound, i need to find this info, but i think is a pulse configuration.

```
ubuntu@bpi-m2z:~$ rfkill list
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

```
 **Play audio to HDMI** 
```
ubuntu@bpi-m2z:~$ sudo aplay -D sysdefault:CARD=1 /usr/share/sounds/alsa/Front_Left.wav 
Home directory not accessible: Permission denied
Playing WAVE '/usr/share/sounds/alsa/Front_Left.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```

 **your command** 
```
ubuntu@bpi-m2z:~$ paplay /usr/share/sounds/alsa/Front_Center.wav
ubuntu@bpi-m2z:~$ ps -e|grep pulse
  873 ?        00:00:00 pulseaudio
ubuntu@bpi-m2z:~$ paplay /usr/share/sounds/alsa/Front_Center.wav
ubuntu@bpi-m2z:~$ ps -e|grep pulse
  873 ?        00:00:00 pulseaudio
```

So, i think the problem is the right pulse configuration that must be done to make a **link** to the hw be available.
Still, we need to investigate.

Configuration for mainline:
```
cat /etc/asound.conf
pcm.!default {
    type plug
    slave {
        pcm "hw:1,0"
    }
}
		
ctl.!default {
   type hw
   card 1
} 
```

avafinger, 2019-10-05:
@Bahir 
Can you try to use this firmware or see the differences? Try the full stack.

https://github.com/RPi-Distro/firmware-nonfree/blob/master/brcm/brcmfmac43430-sdio.txt

Bahir, 2019-10-05:
> Interesting. I could see the first time you tried to connect failed. And the next attempt was a success.
That's because of I forgot to run pulseaudio first time :-)

With this:
https://github.com/RPi-Distro/firmware-nonfree/blob/master/brcm/brcmfmac43430-sdio.txt

1. WIFI stopped to work at all (not associates with AP)
2. BT the same things

avafinger, 2019-10-05:
Did you get all files from https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm or just the brcmfmac43430-sdio.txt ?

Bahir, 2019-10-05:
Yes I've got all of the files. 
It does not work only if I replace my `brcmfmac43430-sdio.txt` with the one from there.

avafinger, 2019-10-05:
Restore the previous firmware and try this and check any improvement on wlan:

```
cd /lib/firmware/brcm/
sudo ln -sf brcmfmac43430-sdio.txt "brcmfmac43430-sdio.sinovoip,bananapi zero.txt"
sync
sudo reboot

```

Bahir, 2019-10-05:
That was exactly my configuration (long quotes story I posted today).

My file `brcmfmac43430-sdio.txt` points to `brcmfmac43430-sdio.txt.orig` that is the copy of `brcmfmac43430-sdio.sinovoip,bananapi zero.txt`.

I posted the content of the file `brcmfmac43430-sdio.txt` and you can check it in the last quote. 
It is the same as `brcmfmac43430-sdio.sinovoip,bananapi zero.txt`.

So, what do you suggest is my original configuration and it has no luck.

avafinger, 2019-10-05:
Not exactly the same:

`[    6.870887] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.sinovoip,bananapi zero.txt failed with error -2
`

with the symlink there will be a direct firmware load and can have different behavior than the fallback, that was a hope.

avafinger, 2019-10-05:
Just for information, i just repeated the exact same instructions with another board (not bpi-m2z) and i confirm the   **paplay /usr/share/sounds/alsa/Front_Center.wav** is enough to play sound in the CHARGE3. No need for the device to be shown in rfkill list.

Anyway, you can install **hcidump** and run it in the second instance of ssh and monitor the messages during play and try to get a clue where it breaks.
You are still ahead of me.

Maybe you can re-install pulse and Bluez and bt tools, and check again until i find out why i can't connect..

Bahir, 2019-10-05:
I changed it as you suggested (made symlinks) now dmesg is
```
[    6.488202] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.580235] Bluetooth: Core ver 2.22
[    6.591334] Bluetooth: HCI device and connection manager initialized
[    6.598184] Bluetooth: HCI socket layer initialized
[    6.604680] Bluetooth: L2CAP socket layer initialized
[    6.609834] Bluetooth: SCO socket layer initialized
[    6.632913] Bluetooth: HCI UART driver ver 2.3
[    6.637435] Bluetooth: HCI UART protocol H4 registered
[    6.642622] Bluetooth: HCI UART protocol BCSP registered
[    6.647936] Bluetooth: HCI UART protocol ATH3K registered
[    6.659153] Bluetooth: HCI UART protocol Intel registered
[    6.665010] Bluetooth: HCI UART protocol Broadcom registered
[    6.672983] Bluetooth: HCI UART protocol AG6XX registered
[    6.710676] Bluetooth: RFCOMM TTY layer initialized
[    6.716213] Bluetooth: RFCOMM socket layer initialized
[    6.716384] Bluetooth: RFCOMM ver 1.11
[    6.766892] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.954734] Bluetooth: hci0: BCM: chip id 94
[    6.969026] Bluetooth: hci0: BCM: features 0x2e
[    7.004613] Bluetooth: hci0: BCM43430A1
[    7.010604] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[    8.780413] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[   11.354717] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0106
[   16.136176] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.136190] Bluetooth: BNEP filters: protocol multicast
[   16.136219] Bluetooth: BNEP socket layer initialized
```

But the behavoiur is the same
1. Strange freezes in network when i activate BT. BTW with 4.20.17 before rampatch were no such freezes but sure they do appear after patchram.
2. Pulseaudio works bad (dirty sound or it crashes itself)
```
root@bpi-m2z:~# aplay -D pulse /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: pcm_write:2053: write error: Input/output error
```

```
root@bpi-m2z:~# paplay /usr/share/sounds/alsa/Front_Center.wav
Connection failure: Connection terminated
Stream error: Connection terminated
```

Also, I tried to install fresh bluez (5.51 from git) and bluez-alsa plugin (https://github.com/Arkq/bluez-alsa)
With this plugin we do no need to run pulseaudio.
The progress is that with bluez-alsa there no crashes (so there no need to run **hcidump**) but sound played so dirty as well as with pulseaudio.

Seems like firmware issue or bluez issue.

avafinger, 2019-10-05:
FYI, i have the same problem with the other board (that had kernel 4.4 and speaker works with BT) now running kernel 5.3.1.
Since both Images (bpi-m2z and the other board) are build from scratch for the kernel 5.3.1 some configuration might be missing (in my case).

avafinger, 2019-10-05:
@Bahir 
Do you have **/etc/bluetooth/audio.conf** and configured?
    [General]
    Enable=Source,Sink,Media,Socket

avafinger, 2019-10-05:
Sometimes i see this: Enable=Source,Sink,Media,Socket,Gateway,Control

avafinger, 2019-10-06:
@Bahir 
After some configurations and installations..... I got the sound coming from the BT speaker now, playing Front_Center is almost clear but in slow motion, something like Froooooonnnnt Ceeeeeenter.

Pulse is still running after several play commands:
`paplay /usr/share/sounds/alsa/Front_Center.wav`
and
`aplay -D pulse /usr/share/sounds/alsa/Front_Center.wav`

The CHARGE3 battery is very low, could be this.... Just turned off now. So there is hope...

avafinger, 2019-10-06:
Forgot to mention i am using **eth0** ,  wlan0 is connected to a second network but not in use.

```
ubuntu@bpi-m2z:~$ ifconfig
eth0      Link encap:Ethernet  HWaddr 66:81:e7:d5:09:ae  
          inet addr:192.168.254.100  Bcast:192.168.255.255  Mask:255.255.0.0
          inet6 addr: fe80::6481:e7ff:fed5:9ae/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1800 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1282 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:155648 (155.6 KB)  TX bytes:225122 (225.1 KB)
          Interrupt:39 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:80 errors:0 dropped:0 overruns:0 frame:0
          TX packets:80 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5920 (5.9 KB)  TX bytes:5920 (5.9 KB)

wlan0     Link encap:Ethernet  HWaddr cc:b8:a8:a9:e5:66  
          inet addr:192.168.254.102  Bcast:192.168.255.255  Mask:255.255.0.0
          inet6 addr: fe80::ceb8:a8ff:fea9:e566/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:33 errors:0 dropped:0 overruns:0 frame:0
          TX packets:23 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2291 (2.2 KB)  TX bytes:3058 (3.0 KB)

ubuntu@bpi-m2z:~$ 

```

Bahir, 2019-10-06:
> @Bahir
> After some configurations and installations..... I get the sound coming out from the BT speaker now, playing Front_Center is almost clear but in slow motion, something like Froooooonnnnt Ceeeeeenter.

Mine is the same effect but with some distortion (like Matrix sound). As for me it not depends on battery level.
What about my  **/etc/bluetooth/audio.conf**. It is.
# This section contains general options
    [General]
    Enable=Source,Sink,Media,Socket

avafinger, 2019-10-06:
> Also, I noticed that when I activate something with BT (pulseaduio, bluetoothctl) the WIFI connection sometimes interrupts responding (strange delays when i type in ssh console).
> And sometimes I need to restart ssh session at all after I activated BT.

Ok, i am now testing with wlan0. BT plays most of the time but sometimes very laggy. There must be a way to parametrize this or change this delay.

To fix the **ssh issue** you have,  edit the fie  **/etc/network/interfaces** and **uncomment** the two lines:
#wireless-mode Managed
#wireless-power off

If there isn't a way to change this lag (BT) i will try to create Ubuntu EON Image if that is possible with armhf. No time frame for this.

avafinger, 2019-10-06:
Small update. I just tried to play some mp3 and the sound is really bad, maybe similar to what you described.

I used sox
`sudo apt-get install sox libsox-fmt-all`:

`play music.mp3`

avafinger, 2019-10-08:
Today I pushed an 8GB Image here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.4

You still need to install ALSA, Bluetooth and then test the latest Bluez stack. So we can see if it has been fixed with the latest and cutting edge.

The Image is a bare minimum Image that can suit every need, Desktop side or Server side

avafinger, 2019-10-09:
Install BT stack:
```
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install dialog alsa-utils alsamixer libasound2 alsa-base
sudo apt-get install dialog alsa-utils libasound2 alsa-base
sudo apt-get install bluez bluez-tools libbluetooth-dev
sudo apt-get install pulseaudio pulseaudio-utils pavucontrol pulseaudio-module-bluetooth

```

then start configuring...

avafinger, 2019-10-09:
Update. 
Playing **aplay -D pulse /usr/share/sounds/alsa/Front_Center.wav** is now fixed in Ubuntu 19.10 but playing mp3 audio no.
I think must be a way to configure **a2dp** but did not find a way.... :(

```
pactl list cards | awk -v card="#2" -v ORS="\n" -v FS="\n" -v RS="" 'split($1,var," ")  var[1] ~ /Card/ && var[2] == card {print}'
Card #2
	Name: bluez_card.8E_06_78_D2_E1_3C
	Driver: module-bluez5-device.c
	Owner Module: 25
	Properties:
		device.description = "CHARGE3"
		device.string = "8E:06:78:D2:E1:3C"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_8E_06_78_D2_E1_3C"
		bluez.class = "0x340404"
		bluez.alias = "CHARGE3"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Profiles:
		a2dp_sink: High Fidelity Playback (A2DP Sink) (sinks: 1, sources: 0, priority: 40, available: yes)
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: no)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink
	Ports:
		headset-output: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): a2dp_sink, headset_head_unit
		headset-input: Headset (priority: 0, latency offset: 0 usec, not available)
			Part of profile(s): headset_head_unit

```

Bahir, 2019-10-11:
Hi!

I installed from scratch v.24 (Ubuntu 19.10/5.3.5)
With bluetooth the same situation.

```
Mac-Pro:~ $ ssh ubuntu@192.168.1.24
Welcome to Ubuntu Eoan Ermine (development branch) (GNU/Linux 5.3.5 armv7l)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

Last login: Fri Oct 11 14:54:30 2019 from 192.168.1.11
ubuntu@bpi-m2z:~$ bluetoothctl
Agent registered
[bluetooth]# connect FC:A8:9A:68:8D:4A
Attempting to connect to FC:A8:9A:68:8D:4A
[CHG] Device FC:A8:9A:68:8D:4A Connected: yes
Connection successful
[JBL E45BT]# info
Device FC:A8:9A:68:8D:4A (public)
	Name: JBL E45BT
	Alias: JBL E45BT
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
[CHG] Device FC:A8:9A:68:8D:4A ServicesResolved: yes
[JBL E45BT]# quit
ubuntu@bpi-m2z:~$ aplay -D pulse /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
ubuntu@bpi-m2z:~$ aplay -D pulse /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: pcm_write:2057: write error: Input/output error
ubuntu@bpi-m2z:~$
```

avafinger, 2019-10-11:
I have: 
`Active Profile: a2dp_sink`
and you
`Active Profile: headset_head_unit`

Any specific reason?

avafinger, 2019-10-11:
ahh ok your JBL is headset, just ignore...

avafinger, 2019-10-11:
I just tried to play the MP3 file on another board and i got the exact same results as bpi-m2z.
I can play Wav sound just fine with:
`sudo paplay /usr/share/sounds/alsa/Front_Left.wav`

**But**... Playing the mp3 (the one i linked) is just noise..

So, i will put it this way... BT is broken on 5.x regarding sound OR we need a codec (the right codec) to play that mp3.

Maybe someone with BT experience read this and share the right way to configure this.

avafinger, 2019-10-11:
I can play this: https://file-examples.com/wp-content/uploads/2017/11/file_example_WAV_1MG.wav

avafinger, 2019-10-11:
Just for the record:
aplay /usr/share/sounds/alsa/Front_Left.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Left.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, 
Mono

aplay file_example_WAV_1MG.wav 
Playing WAVE 'file_example_WAV_1MG.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo

aplay SoundHelix-Song-1.mp3 
Playing raw data 'SoundHelix-Song-1.mp3' : Unsigned 8 bit, Rate 8000 Hz, Mono

avafinger, 2019-10-11:
BT is working on the other board using sox (kernel 5.3.1). I will check a few things and if i get something new i will report here. Sorry for the noise.

```
wget https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.mp3
play ff-16b-2c-44100hz.mp3 

ff-16b-2c-44100hz.mp3:

 File Size: 2.99M     Bit Rate: 128k
  Encoding: MPEG audio    Info: 2016-04-12
  Channels: 2 @ 16-bit   
Samplerate: 44100Hz      
Replaygain: off         
  Duration: 00:03:07.12  

In:100%  00:03:07.12 [00:00:00.00] Out:8.25M [      |      ]        Clip:0 
```

avafinger, 2019-10-11:
FYI:

```
ubuntu@bpi-m2z:~$ play ff-16b-2c-44100hz.mp3 
play WARN alsa: can't encode 0-bit Unknown or not applicable
Floating point exception

```

avafinger, 2019-10-12:
Hurrah, it works! Crystal clear sound.

Something strange happened after the error below and everything started to work now.
I had to redo everything. Just need to know if it will survive a reboot... 

```
ubuntu@bpi-m2z:~$ play file_example_WAV_1MG.wav

file_example_WAV_1MG.wav:

 File Size: 1.07M     Bit Rate: 256k
  Encoding: Signed PCM    
  Channels: 2 @ 16-bit   
Samplerate: 8000Hz       
Replaygain: off         
  Duration: 00:00:33.53  

In:9.16% 00:00:03.07 [00:00:30.46] Out:24.6k [ -====|====- ] Hd:1.7 Clip:0    play FAIL sox: `default' error writing to pulse audio device: Connection terminated: Operation not permitted
Done.
ubuntu@bpi-m2z:~$ ps -e|grep pulse
  775 ?        00:00:00 pulseaudio
ubuntu@bpi-m2z:~$ play ff-16b-2c-44100hz.mp3 

ff-16b-2c-44100hz.mp3:

 File Size: 2.99M     Bit Rate: 128k
  Encoding: MPEG audio    Info: 2016-04-12
  Channels: 2 @ 16-bit   
Samplerate: 44100Hz      
Replaygain: off         
  Duration: 00:03:07.12  

In:67.9% 00:02:07.06 [00:01:00.06] Out:5.60M [-=====|======] Hd:1.5 Clip:0    
Aborted.

ubuntu@bpi-m2z:~$ play ff-16b-2c-44100hz.mp3 

ff-16b-2c-44100hz.mp3:

 File Size: 2.99M     Bit Rate: 128k
  Encoding: MPEG audio    Info: 2016-04-12
  Channels: 2 @ 16-bit   
Samplerate: 44100Hz      
Replaygain: off         
  Duration: 00:03:07.12  

In:67.9% 00:02:07.06 [00:01:00.06] Out:5.60M [-=====|======] Hd:1.5 Clip:0    
Aborted.
ubuntu@bpi-m2z:~$ play ff-16b-2c-44100hz.mp3 

ff-16b-2c-44100hz.mp3:

 File Size: 2.99M     Bit Rate: 128k
  Encoding: MPEG audio    Info: 2016-04-12
  Channels: 2 @ 16-bit   
Samplerate: 44100Hz      
Replaygain: off         
  Duration: 00:03:07.12  

In:3.23% 00:00:06.04 [00:03:01.08] Out:266k  [     =|-     ]        Clip:0    
Aborted.
ubuntu@bpi-m2z:~$ play file_example_WAV_1MG.wav

file_example_WAV_1MG.wav:

 File Size: 1.07M     Bit Rate: 256k
  Encoding: Signed PCM    
  Channels: 2 @ 16-bit   
Samplerate: 8000Hz       
Replaygain: off         
[... 330 lines cut ...]
		alsa.device = "0"
		alsa.card = "1"
		alsa.card_name = "allwinner,hdmi"
		alsa.long_card_name = "allwinner,hdmi"
		alsa.driver_name = "snd_soc_simple_card"
		device.bus_path = "platform-soc:sound"
		sysfs.path = "/devices/platform/soc/soc:sound/sound/card1"
		device.form_factor = "internal"
		device.string = "hw:1"
		device.buffering.buffer_size = "352800"
		device.buffering.fragment_size = "58800"
		device.access_mode = "mmap+timer"
		device.profile.name = "stereo-fallback"
		device.profile.description = "Stereo"
		device.description = "Built-in Audio Stereo"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"
	ports:
		analog-output: Analog Output (priority 9900, latency offset 0 usec, available: unknown)
			properties:
				
	active port: <analog-output>
ubuntu@bpi-m2z:~$ 

```

Bahir, 2019-10-12:
> I have:
> `Active Profile: a2dp_sink`
> and you
> `Active Profile: headset_head_unit`
> 
> Any specific reason?

I have no idea. Now it is SINK

```
Card #4
	Name: bluez_card.FC_A8_9A_68_8D_4A
	Driver: module-bluez5-device.c
	Owner Module: 27
	Properties:
		device.description = "JBL E45BT"
		device.string = "FC:A8:9A:68:8D:4A"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_FC_A8_9A_68_8D_4A"
		bluez.class = "0x240404"
		bluez.alias = "JBL E45BT"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Profiles:
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
		a2dp_sink: High Fidelity Playback (A2DP Sink) (sinks: 1, sources: 0, priority: 40, available: yes)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink
	Ports:
		headset-output: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit, a2dp_sink
		headset-input: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit
```

Bahir, 2019-10-12:
I've noticed that `pactl list card` shows sink as active profile until `aplay` or `paplay` crashed.
After that it is `Active Profile: headset_head_unit`.

And, now I suddenly started to get these errors (without touching any configs and no new packages installed, just played with bluetoothctl connecting headset and played sounds).

```
Oct 12 15:28:13 bpi-m2z pulseaudio[633]: E: [pulseaudio] bluez5-util.c: GetManagedObjects() failed: org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched rules; type="method_call", sender=":1.45" (uid=1000 pid=633 comm="/usr/bin/pulseaudio --daemonize=no ") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=381 comm="/usr/lib/bluetooth/bluetoothd ")
```

Reboot did not help. The error is still here.

avafinger, 2019-10-12:
I think Bluetooth (armhf) is buggy. In Arm64 BT works pretty good and stable.
Unfortunately, my fix did not survive a reboot and now the same error i mentioned above forces a disconnect from pulseaudio. 

You could try:

```
pusleaudio -k
pusleaudio --start
pactl set-card-profile 2 a2dp_sink

```

avafinger, 2019-10-12:
and check for some error:

```
sudo cat /var/log/syslog | grep -i blue
sudo cat /var/log/syslog | grep -i err
sudo cat /var/log/syslog | grep -i fail

```

Bahir, 2019-10-15:
A bit of progress. I connected antenna to bpi and now bluetooth does not crash while playing sound, like `aplay: pcm_write:2053: write error: Input/output error`. One problem found and gone.
But the sound is still barking while playing. Seems like something wrong with baudrate and a channel speed is not enough to pass WAV or it drops packets (I'm not sure, just a guess)

meisterbasti, 2020-02-24:
Hey, with the new Kernel 5.3.18 I was able to pair and connect to a bluetooth speaker. 
But when I try to play a .wav file the sound is very choppy

Why is it that choppy ? I wan't to detect objects with an attached Coral Edge TPU from Google and what to hear the result over the speaker.

```
[bluetooth]# info C8:84:47:1A:DF:08 
Device C8:84:47:1A:DF:08 (public)
	Name: Sound Kick
	Alias: Sound Kick
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
[bluetooth]# connect C8:84:47:1A:DF:08 
Attempting to connect to C8:84:47:1A:DF:08
[CHG] Device C8:84:47:1A:DF:08 Connected: yes
Connection successful
[CHG] Device C8:84:47:1A:DF:08 ServicesResolved: yes
[Sound Kick]# quit
(cv) ubuntu@bpi-m2z:~/Master$ pico2wave --lang=de-DE --wave=/tmp/test.wav "hallo welt hier spricht der räspberri pei"; play /tmp/test.wav;rm /tmp/test.wav
play WARN alsa: can't encode 0-bit Unknown or not applicable

/tmp/test.wav:

 File Size: 90.8k     Bit Rate: 256k
  Encoding: Signed PCM    
  Channels: 1 @ 16-bit   
Samplerate: 16000Hz      
Replaygain: off         
  Duration: 00:00:02.84  

In:100%  00:00:02.84 [00:00:00.00] Out:45.4k [      |      ] Hd:2.3 Clip:0    
Done.
(cv) ubuntu@bpi-m2z:~/Master$ 
```

meisterbasti, 2020-02-26:
The same problem with a connected Bluetooth Headphones...

I also tried out with connected monitor on the banana pi with disabling wlan0. But still the same. The sound is always very choppy 

```
[Teufel AIRY]# info
Device A0:E9:DB:01:71:AF (public)
	Name: Teufel AIRY
	Alias: Teufel AIRY
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
[Teufel AIRY]# quit
(cv) ubuntu@bpi-m2z:~$ play temp.wav 
play WARN alsa: can't encode 0-bit Unknown or not applicable

temp.wav:

 File Size: 229k      Bit Rate: 256k
  Encoding: Signed PCM    
  Channels: 1 @ 16-bit   
Samplerate: 16000Hz      
Replaygain: off         
  Duration: 00:00:07.14  

In:100%  00:00:07.14 [00:00:00.00] Out:114k  [      |      ] Hd:5.4 Clip:0    
Done.
(cv) ubuntu@bpi-m2z:~$ 
```

avafinger, 2020-02-27:
Please, try this possible fix:

* untar the attached file

* copy **bpi-m2-zero-v4.dtb_5.3.18** to  **/boot** and overwrite the file

* make sure you have:
`lrwxrwxrwx  1 root   root        32 Feb 26 13:43 bpi-m2-zero.dtb -> bpi-m2-zero-v4.dtb_5.3.18`

* reboot
**sudo reboot**

[bpi-m2-zero-v4.dtb_5.3.18.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/4261487/bpi-m2-zero-v4.dtb_5.3.18.tar.gz)

meisterbasti, 2020-02-28:
At the beginning it seems to fix the problem. Sound started with good quality but now the sound is interrupted completely and the device in bluetoothctl is no longer connected.

```
(cv) ubuntu@bpi-m2z:~$ bluetoothctl
Agent registered
[Sound Kick]# info
Device C8:84:47:1A:DF:08 (public)
	Name: Sound Kick
	Alias: Sound Kick
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
[Sound Kick]# quit
(cv) ubuntu@bpi-m2z:~$ play temp.wav
play WARN alsa: can't encode 0-bit Unknown or not applicable

temp.wav:

 File Size: 229k      Bit Rate: 256k
  Encoding: Signed PCM    
  Channels: 1 @ 16-bit   
Samplerate: 16000Hz      
Replaygain: off         
  Duration: 00:00:07.14  

In:35.8% 00:00:02.56 [00:00:04.58] Out:41.0k [======|======] Hd:2.2 Clip:0    play WARN alsa: Input/output error
play FAIL sox: `default' Input/output error: Operation not permitted
Done.
(cv) ubuntu@bpi-m2z:~$ bluetoothctl
Agent registered
[bluetooth]# info
Missing device address argument
[bluetooth]# 
```

avafinger, 2020-02-28:
Looks like BT speed is still wrong. I have seen 921600  and 1500000 max speed for this  device.
Currently set to 1500000, maybe it should be 921600.

You could try different speed values.

* Install DTC
`sudo apt-get install device-tree-compiler`

* Decompile **bpi-m2-zero-v4.dtb_5.3.18**
  change **max-speed**= to the new value

* Re-compile **bpi-m2-zero-v4.dtb_5.3.18**

* reboot

PS: make a backup file

meisterbasti, 2020-02-29:
After decompiling

```
(cv) ubuntu@bpi-m2z:/boot$ dtc -I dtb -O dts bpi-m2-zero-v4.dtb_5.3.18 -o /tmp/test/tmp.dts
/tmp/test/tmp.dts: Warning (unit_address_vs_reg): /soc/pinctrl@1c20800/bt_pwr_pin@0: node has a unit name, but no reg property
/tmp/test/tmp.dts: Warning (unit_address_vs_reg): /soc/pinctrl@1f02c00/r_i2c@0: node has a unit name, but no reg property
/tmp/test/tmp.dts: Warning (unit_address_vs_reg): /soc/pinctrl@1f02c00/led_pins@0: node has a unit name, but no reg property
/tmp/test/tmp.dts: Warning (unit_address_vs_reg): /soc/pinctrl@1f02c00/key_pins@0: node has a unit name, but no reg property
/tmp/test/tmp.dts: Warning (unit_address_vs_reg): /soc/pinctrl@1f02c00/regulator_pins@0: node has a unit name, but no reg property
/tmp/test/tmp.dts: Warning (avoid_unnecessary_addr_size): /soc/mixer@1100000/ports/port@1: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
/tmp/test/tmp.dts: Warning (avoid_unnecessary_addr_size): /soc/lcd-controller@1c0c000/ports/port@0: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
/tmp/test/tmp.dts: Warning (avoid_unnecessary_addr_size): /soc/ethernet@1c30000: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
/tmp/test/tmp.dts: Warning (unique_unit_address): /soc/spi@1c68000/spi@0: duplicate unit-address (also used in node /soc/spi@1c68000/spiflash@0)
/tmp/test/tmp.dts: Warning (unique_unit_address): /soc/spi@1c68000/spi@0: duplicate unit-address (also used in node /soc/spi@1c68000/pitft@0)
/tmp/test/tmp.dts: Warning (unique_unit_address): /soc/spi@1c68000/spiflash@0: duplicate unit-address (also used in node /soc/spi@1c68000/pitft@0)
/tmp/test/tmp.dts: Warning (alias_paths): /aliases: aliases property name must include only lowercase and '-'
/tmp/test/tmp.dts: Warning (graph_child_address): /soc/mixer@1100000/ports/port@1: graph node has single child node 'endpoint', #address-cells/#size-cells are not necessary
/tmp/test/tmp.dts: Warning (graph_child_address): /soc/lcd-controller@1c0c000/ports/port@0: graph node has single child node 'endpoint', #address-cells/#size-cells are not necessary
```

I changed the value using "jed Editor" as you said from:

```
bluetooth {
                                compatible = "brcm,bcm43438-bt";
                                max-speed = < 0x16e360 >;
                                clocks = < 0x11 0x01 >;
                                clock-names = "lpo";
                                vbat-supply = < 0x09 >;
                                vddio-supply = < 0x09 >;
                                device-wakeup-gpios = < 0x0c 0x06 0x0d 0x00 >;
                                host-wakeup-gpios = < 0x0c 0x06 0x0b 0x00 >;
                                shutdown-gpios = < 0x0c 0x06 0x0c 0x00 >;

```

to

```
bluetooth {
                                compatible = "brcm,bcm43438-bt";
                                max-speed = < 0xe1000 >;
                                clocks = < 0x11 0x01 >;
                                clock-names = "lpo";
                                vbat-supply = < 0x09 >;
                                vddio-supply = < 0x09 >;
                                device-wakeup-gpios = < 0x0c 0x06 0x0d 0x00 >;
                                host-wakeup-gpios = < 0x0c 0x06 0x0b 0x00 >;
                                shutdown-gpios = < 0x0c 0x06 0x0c 0x00 >;

```

after saving and recompiling with 

`dtc -I dts -O dtb -f tmp.dts -o bpi-m2-zero-v4.dtb_5.3.18`

There were several warnings but the file was created.

```
bpi-m2-zero-v4.dtb_5.3.18: Warning (unit_address_vs_reg): /soc/pinctrl@1c20800/bt_pwr_pin@0: node has a unit name, but no reg property
bpi-m2-zero-v4.dtb_5.3.18: Warning (unit_address_vs_reg): /soc/pinctrl@1f02c00/r_i2c@0: node has a unit name, but no reg property
bpi-m2-zero-v4.dtb_5.3.18: Warning (unit_address_vs_reg): /soc/pinctrl@1f02c00/led_pins@0: node has a unit name, but no reg property
bpi-m2-zero-v4.dtb_5.3.18: Warning (unit_address_vs_reg): /soc/pinctrl@1f02c00/key_pins@0: node has a unit name, but no reg property
bpi-m2-zero-v4.dtb_5.3.18: Warning (unit_address_vs_reg): /soc/pinctrl@1f02c00/regulator_pins@0: node has a unit name, but no reg property
bpi-m2-zero-v4.dtb_5.3.18: Warning (avoid_unnecessary_addr_size): /soc/mixer@1100000/ports/port@1: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
bpi-m2-zero-v4.dtb_5.3.18: Warning (avoid_unnecessary_addr_size): /soc/lcd-controller@1c0c000/ports/port@0: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
bpi-m2-zero-v4.dtb_5.3.18: Warning (avoid_unnecessary_addr_size): /soc/ethernet@1c30000: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
bpi-m2-zero-v4.dtb_5.3.18: Warning (unique_unit_address): /soc/spi@1c68000/spi@0: duplicate unit-address (also used in node /soc/spi@1c68000/spiflash@0)
bpi-m2-zero-v4.dtb_5.3.18: Warning (unique_unit_address): /soc/spi@1c68000/spi@0: duplicate unit-address (also used in node /soc/spi@1c68000/pitft@0)
bpi-m2-zero-v4.dtb_5.3.18: Warning (unique_unit_address): /soc/spi@1c68000/spiflash@0: duplicate unit-address (also used in node /soc/spi@1c68000/pitft@0)
[... 200 lines cut ...]
bpi-m2-zero-v4.dtb_5.3.18: Warning (gpios_property): /soc/spi@1c68000:cs-gpios: cell 4 is not a phandle reference
bpi-m2-zero-v4.dtb_5.3.18: Warning (gpios_property): /soc/spi@1c68000/pitft@0:dc-gpios: cell 0 is not a phandle reference
[... cut ...]
bpi-m2-zero-v4.dtb_5.3.18: Warning (graph_child_address): /soc/mixer@1100000/ports/port@1: graph node has single child node 'endpoint', #address-cells/#size-cells are not necessary
bpi-m2-zero-v4.dtb_5.3.18: Warning (graph_child_address): /soc/lcd-controller@1c0c000/ports/port@0: graph node has single child node 'endpoint', #address-cells/#size-cells are not necessary
(cv) ubuntu@bpi-m2z:/tmp/test$ 
```

after copying back to /boot/ and rebooting the controller wasn't available anymore..

What went wrong ? 
I copied back your initial file before modification and after rebooting the controller was available again but.

meisterbasti, 2020-03-03:
Any ideas ?

avafinger, 2020-03-03:
Nothing is wrong with your procedure. DTB seems to be off a bit somehow, i will investigate.
But looking at the BSP the correct bt speed is 1500000.

Can you try the latest 5.6.0-rc3 to see if you get better results? Make a backup.
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.0

avafinger, 2020-03-04:
Please, check the files in boot, zImage is missing:

`ls -la /boot`

avafinger, 2020-03-04:
You should have something like this:

```
alex@svn:/media/alex/boot$ ls -la
total 17588
drwxrwxr-x   3 alex alex    4096 mar  2 15:56 .
drwxr-x---+ 10 root root    4096 mar  4 16:58 ..
lrwxrwxrwx   1 root root      20 jan 15  2018 boot.scr -> bpi-m2-zero_boot.scr
-rw-rw-r--   1 alex alex   31573 mar  2 15:56 bpi.dts
-rw-r--r--   1 root root    1183 nov 24  2018 bpi-m2-zero_boot.cmd
-rw-r--r--   1 root root    1255 nov 24  2018 bpi-m2-zero_boot.scr
lrwxrwxrwx   1 root root      29 fev 28 21:42 bpi-m2-zero.dtb -> bpi-m2-zero-v4.dtb_5.6.0-rc3+
-rw-r--r--   1 root root   26088 fev  9 10:24 bpi-m2-zero-v4.dtb_5.3.18
-rw-r--r--   1 root root   26114 fev 26 00:52 bpi-m2-zero-v4.dtb_5.3.18-fbdev
-rw-r--r--   1 root root   27017 fev 28 21:42 bpi-m2-zero-v4.dtb_5.6.0-rc3+
-rw-rw-r--   1 alex alex  982455 jan 17  2018 initrd.img
drwx------   2 root root   16384 nov  5  2017 lost+found
lrwxrwxrwx   1 root root      17 fev 28 21:42 zImage -> zImage_5.6.0-rc3+
-rwxr-xr-x   1 root root 5759992 fev  9 10:24 zImage_5.3.18
-rwxr-xr-x   1 root root 5760416 fev  9 16:29 zImage_5.3.18-fbdev
-rwxr-xr-x   1 root root 5346752 fev 28 21:41 zImage_5.6.0-rc3+

```

avafinger, 2020-03-04:
Make sure the symlinks are ok, You can do this in /boot: (or /media/boot in case you have the sd card attached to a PC box)

```
sudo ln -sf zImage_5.6.0-rc3+ zImage
sudo ln -sf bpi-m2-zero-v4.dtb_5.6.0-rc3+ bpi-m2-zero.dtb
```

and then boot it again.

Also, check for SD card errors with fsck...

meisterbasti, 2020-03-04:
> Please, check the files in boot, zImage is missing:
> 
> `ls -la /boot`

check the /boot after the Update of the Kernel before rebooting or after rebooting and receiving the showed log ?

avafinger, 2020-03-04:
The deb install has the wrong symlink, to fix this, you have to do this:

         sudo ln -sf zImage_5.6.0-rc3+ zImage

You cannot boot anymore if you already rebooted, you should attach the sd card to you linux box and go to /boot (SD card) and repair the symlink as described above.

meisterbasti, 2020-03-04:
I don't have any Linux Box beside the BPZ. I'm going to restored the SD card to a backup and will try to update Kernel + your Fix before rebooting

avafinger, 2020-03-04:
You can also test the new Kernel 5.6.0-rc4 from here:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.1

I think i fixed the install issue.

meisterbasti, 2020-03-05:
> You can also test the new Kernel 5.6.0-rc4 from here:
> https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.1
> 
> I think i fixed the install issue.

This version worked well.

```
Welcome to Ubuntu 19.10 (GNU/Linux 5.6.0-rc4+ armv7l)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

 * Multipass 1.0 is out! Get Ubuntu VMs on demand on your Linux, Windows or
   Mac. Supports cloud-init for fast, local, cloud devops simulation.

     https://multipass.run/

 * Latest Kubernetes 1.18 beta is now available for your laptop, NUC, cloud
   instance or Raspberry Pi, with automatic updates to the final GA release.

     sudo snap install microk8s --channel=1.18/beta --classic
Last login: Thu Mar  5 19:31:26 2020
```

I have a question. I have seen this since a few days....
Is this advertisement: 

>   * Multipass 1.0 is out! Get Ubuntu VMs on demand on your Linux, Windows or
>    Mac. Supports cloud-init for fast, local, cloud devops simulation.
> 
>      https://multipass.run/
> 
>  * Latest Kubernetes 1.18 beta is now available for your laptop, NUC, cloud
>    instance or Raspberry Pi, with automatic updates to the final GA release.
> 
>      sudo snap install microk8s --channel=1.18/beta --classic

??

meisterbasti, 2020-03-05:
```
ubuntu@bpi-m2z:~$ aplay temp.wav 
Playing WAVE 'temp.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
aplay: pcm_write:2057: write error: Input/output error
```

Audio is still not working correctly. Has the bt speed been set to 1500000 ?

avafinger, 2020-03-05:
So, nothing changed with the new Kernel 5.6.0-rc4 ?

Maybe the problem is in the BT stack for the 32-bit  version. Let's try 20.04 and see if there is an improvement. BT speed is 1500000 in DT.

Regarding the banner, it is from Ubuntu. I've never touched snap...

[4 more replies]