avafinger/bananapi-zero-ubuntu-base-minimal #29: HDMI display image

neilyoung, 2019-10-23:
I'm having issues with the convenience image. HDMI is pretty much unreadable, since it is shifted left by 3-4 characters, top and bottom by several lines. Monitored on two BPI Zero M2 and with different monitors.

What can I do?

avafinger, 2019-10-23:
Can you give me more information?

* Monitor(s) in use, maker, and specifications
* Any adapter or extension you are using?

Can you post a picture of the unreadable screen?
There is a chance you need to set a custom EDID but you need to identify which screen resolution your monitor(s)  can handle.

For example, here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v1.9
This is set to 1280x720@60.

neilyoung, 2019-10-23:
@avafinger Thanks for the quick follow up. Both are Samsung, consumer models. 

Here a "screenshot". It has most likely be made with another image (Ubuntu 16.04 server), but the symptom is the same with the convenience image. The monitors report 1920x1080 @60 Hz.

[image]

neilyoung, 2019-10-23:
... and no adapter in use. Just the little mini-HDMI to HDMI adapter from a PIZW

neilyoung, 2019-10-23:
Excuse the dumb question: How could i make an image I could flash from the version you quoted?

neilyoung, 2019-10-23:
The shot has been taken from a Samsung UE45D6200

avafinger, 2019-10-23:
Start with this Image:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.4

Regarding your issue, use the TV remote control to fix up:

* Wrong setting
[image]

* Correct setting
[image]

neilyoung, 2019-10-24:
Thanks. Will try and report. I think I already skimmed the TV settings and found nothing which worked. Will repeat.

neilyoung, 2019-10-24:
OK, following the link I was able to flash the image with Etcher. I was **not** able to follow this:

```
First thing you should do before you boot

Edit /etc/network/interfaces and change the portion with your SID and PSK

  allow-hotplug wlan0
  iface wlan0 inet dhcp
      wpa-ssid "your SID"
      wpa-psk "put here your ASCII password"

```

The SD is simply not readable from macOS after flashing. This happens with all BPI images I tested so far. It does not happen with e.g. regular Raspbian images.

MacOS complains, that the media could not be read by this computer. Hence I could not add Wifi support.

Anyway, I could make it run. My TV does not have such a setting, but a "Screen adaptation" setting, by which I could move the position of the display, so that it perfectly fits now. I plugged an USB ETH adapter and by this I'm ready now to make it a "real" Ubuntu :) 

Thanks for your assistance!

neilyoung, 2019-10-24:
Anyway - what can be seen at the first or second glance: This distribution has been compiled with patience, love and accuracy. Well done, nice piece of work. Don't know, why I didn't find it by myself with the very first shot...

avafinger, 2019-10-24:
> I noticed, that it needs several minutes to present the prompt.

This usually indicates you have a poor or bad (quality) sd card, and you might have trouble in the near future. The best thing to do is buy a fast and good sd card.

Mac OS and Windows cannot read linux files without "special assistance".
You can edit the files with **jed**  when you get in the shell command, like this:
`sudo jed /etc/network/interfaces`

neilyoung, 2019-10-24:
That’s a good hint. Is there any card you could recommend?

neilyoung, 2019-10-24:
Sorry accidentally closed issue

neilyoung, 2019-10-24:
Right now I’m using a 32 GB Transcend

neilyoung, 2019-10-24:
OK, I have flashed another card now. I'm wondering: The SSH server seems to NOT be enabled by default, after flashing, right? I supposed, this was the case, but now I guess it has been activated by update && upgrade. Would that be possible?

Other than that: Are you knowing, how I could enable the SPI interface? On an ordinary PI this could be done by raspi-config, but this is not possible here, I guess..

Thanks in advance.

avafinger, 2019-10-24:
> EDIT: Update: The SSH server is running. After a couple of restarts it worked.

Change the file /etc/network/interfaces:

**allow-hotplug** wlan0
```
  iface wlan0 inet dhcp
      wpa-ssid "your SID"
      wpa-psk "put here your ASCII password"

```

to

**auto** wlan0
```
  iface wlan0 inet dhcp
      wpa-ssid "your SID"
      wpa-psk "put here your ASCII password"
```

Two possible things:
* Your sd card is corrupt, if you removed power without doing a **sudo shutdown -h now** you have now a corrupt content. If that is the case you need to do an **fsck** (linux box)

* Your sd card is fake

neilyoung, 2019-10-24:
Hmm. The SD card worked quite well in a couple of RPIs for now. Anyway: Removing the power w/o shutdown: I guess this is the ordinary use case of a PI, don't you agree ? :)

Thanks for the hints. Do you know how to enable SPI?

neilyoung, 2019-10-24:
I have made an fsck on my linux box: Nothing obvious, I would say...

```
decades@ubuntu:~/Documents$ sudo fsck -f /dev/sdc1
fsck from util-linux 2.27.1
e2fsck 1.42.13 (17-May-2015)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
boot: 19/25600 files (10.5% non-contiguous), 3495/25600 blocks
decades@ubuntu:~/Documents$ sudo fsck -f /dev/sdc2
fsck from util-linux 2.27.1
e2fsck 1.42.13 (17-May-2015)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
rootfs: 20583/485760 files (0.2% non-contiguous), 203191/1940480 blocks
decades@ubuntu:~/Documents$ 
```

neilyoung, 2019-10-24:
Is there any way to test, if the SD is "sufficient" for the system?

avafinger, 2019-10-24:
Sd card is clean, could be your PS. Try one with 5v and 2.5A.

To enable SPI you need to decompile the **dtb** , change the SPI node to "okay" and recompile.

neilyoung, 2019-10-24:
The PS is an original PI PS. I would exclude it.
As a matter of fact, meanwhile I cannot SSH at all, even though both interfaces are up and I can ping it (ETH and Wifi).  No idea anymore. FS is clean, did another check right now. 

Anything I could check, supposed I will make it again onto the box?

neilyoung, 2019-10-24:
Ah.. .now. 5 minutes after start - ssh prompt.

neilyoung, 2019-10-24:
Something in here?

```
ubuntu@bpi-m2z:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.3.5 (alex@svn) (gcc version 7.4.1 20181213 [linaro-7.4-2019.02 revision 56ec6f6b99cc167ff0c2f8e1a2eed33b1edc85d4] (Linaro GCC 7.4-2019.02)) #1 SMP Mon Oct 7 21:22:23 -03 2019
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Banana Pi M2 Zero
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x4dc00000
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 15 pages/cpu s31244 r8192 d22004 u61440
[    0.000000] pcpu-alloc: s31244 r8192 d22004 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyS0,115200 earlyprintk rootfstype=ext4 root=/dev/mmcblk0p2 rw rootwait fsck.repair=yes panic=10 no_console_suspend consoleblank=0 drm_leak_fbdev_smem=1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 241880K/524288K available (9216K kernel code, 328K rwdata, 1984K rodata, 1024K init, 269K bss, 20264K reserved, 262144K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2f8/0x488 with crng_init=0
[    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000006] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000017] Switching to timer-based delay loop, resolution 41ns
[    0.000213] Console: colour dummy device 80x30
[    0.000259] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000271] pid_max: default: 32768 minimum: 301
[    0.000412] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000425] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001197] CPU: Testing write buffer coherency: ok
[    0.001650] CPU0: update cpu_capacity 1024
[    0.001659] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002197] Setting up static identity map for 0x40100000 - 0x40100060
[    0.002328] rcu: Hierarchical SRCU implementation.
[    0.002871] smp: Bringing up secondary CPUs ...
[    0.013620] CPU1: update cpu_capacity 1024
[    0.013627] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.024503] CPU2: update cpu_capacity 1024
[    0.024509] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.035277] CPU3: update cpu_capacity 1024
[    0.035283] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.035371] smp: Brought up 1 node, 4 CPUs
[    0.035394] SMP: Total of 4 processors activated (192.00 BogoMIPS).
[    0.035400] CPU: All CPU(s) started in HYP mode.
[    0.035404] CPU: Virtualization extensions available.
[    0.036359] devtmpfs: initialized
[    0.042763] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.042981] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.043007] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.051494] pinctrl core: initialized pinctrl subsystem
[    0.052603] NET: Registered protocol family 16
[    0.054217] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.055239] cpuidle: using governor menu
[    0.055564] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.055572] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.078690] SCSI subsystem initialized
[... 313 lines cut ...]
[    9.668786] usbcore: registered new interface driver asix
[   10.040260] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-11), device may have limited channels available
[   10.044996] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug 29 2016 20:48:16 version 7.45.41.26 (r640327) FWID 01-4527cfab
[   10.065312] asix 1-1:1.0 enx0050b624eec0: renamed from eth1
[   10.088340] asoc-simple-card soc:sound: i2s-hifi <-> 1c22800.i2s mapping ok
[   13.283876] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0106
[   14.583144] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
[... cut ...]
[   16.069016] IPv6: ADDRCONF(NETDEV_CHANGE): enx0050b624eec0: link becomes ready
[   16.073295] asix 1-1:1.0 enx0050b624eec0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   30.268814] random: crng init done
[   30.268841] random: 7 urandom warning(s) missed due to ratelimiting
[   32.481393] vcc3v0: disabling
[   32.481430] vcc5v0: disabling
[   53.593285] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  218.171947] Adding 2097148k swap on /var/swap.  Priority:-2 extents:9 across:2564096k SS
ubuntu@bpi-m2z:~$ 
```

avafinger, 2019-10-24:
Adding **2097148k** swap on /var/swap.  Priority:-2 extents:9 across:256409

I think you are running out of memory.
Don't use **swap**, use **zram**

It took **53** seconds to get wlan0 up.
You have to be careful when using eth1 and wlan0 (if enabled) at the same time. I think this may be delaying the board, choose one of the two or comment out the wlan interface and see the timings again and vice-versa.

Here is the timings you should expect:

```
[   12.136151] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0106
[   13.032178] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
[   13.642996] dwmac-sun8i 1c30000.ethernet eth0: PHY [0.1:01] driver [Generic PHY]
[   13.643021] dwmac-sun8i 1c30000.ethernet eth0: phy: setting supported 00,00000000,000062cf advertising 00,00000000,000062cf
[   13.648170] dwmac-sun8i 1c30000.ethernet eth0: No Safety Features support found
[   13.648195] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
[   13.648207] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[   13.648223] dwmac-sun8i 1c30000.ethernet eth0: configuring for phy/mii link mode
[   13.648245] dwmac-sun8i 1c30000.ethernet eth0: phylink_mac_config: mode=phy/mii/Unknown/Unknown adv=00,00000000,000062cf pause=10 link=0 an=1
[   13.654210] dwmac-sun8i 1c30000.ethernet eth0: phy link up mii/100Mbps/Full
[   13.654226] dwmac-sun8i 1c30000.ethernet eth0: phylink_mac_config: mode=phy/mii/100Mbps/Full adv=00,00000000,00000000 pause=0f link=1 an=0
[   13.654269] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   13.660920] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   22.358788] random: crng init done
[   22.358846] random: 7 urandom warning(s) missed due to ratelimiting
[   23.806143] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   32.483655] vcc3v0: disabling
[   32.483686] vcc5v0: disabling

```

neilyoung, 2019-10-24:
Cool. Thanks for the analysis. Yes, I have added a swap file since I need to compile a VERY big project. Never used zram before, will check. The swap partition is just temporary, will remove it later. 

I guess, your memory hypothesis is right. With my first attempts I was monitoring the behaviour with htop. I noticed, that the CPU load was pretty low, but the memory load was "over the top".

Will keep you posted, how it goes with zram.

neilyoung, 2019-10-24:
There must still be something wrong with SSH. ETH is down now (disconnected). After reboot I can ping Wifi immediately. But the SSH server does not respond until at least 5 minutes are gone. Swap is 0 again. Very strange.

neilyoung, 2019-10-24:
OK, this is the state:

```
ubuntu@bpi-m2z:~$ free
              total        used        free      shared  buff/cache   available
Mem:         506008       41936      402092        2680       61980      453040
Swap:       1285180           0     1285180
ubuntu@bpi-m2z:~$ sudo swapon -s
[sudo] password for ubuntu: 
Filename				Type		Size	Used	Priority
/dev/zram0                             	partition	63248	0	5
/dev/zram1                             	partition	63248	0	5
/dev/zram2                             	partition	63248	0	5
/dev/zram3                             	partition	63248	0	5
/var/swap                              	file    	1032188	0	-2
```
The suggestion for compiling the SDK was to have a swap partition of at least 2GB. I didn't change the default ZRAM config now,  running `make -j3`. htop shows 1.77 CPU load over 15 minutes. Memory is still close to 500 MB utilized.  Not sure if this will work w/o accident....

neilyoung, 2019-10-24:
I cannot see a major advantage of ZRAM over SWAP. Running `make -j2` now, and it is unexpectedly lame. SWAP usage is 500 MB of 1.23 GB.  Mem is at 100 %, CPU at 2.58.

I have compiled the same stack on a single core Raspberry PI Zero W. It took 8 hours there. I was expecting to see a way more speedy build here, but it turns out, it's not the case.

neilyoung, 2019-10-24:
I'm sure there is a problem with the SSH server. Regardless of two or just one active interface: The interface is UP, I can ping it, but SSH is refused for minutes, and most the time it does not get up at all.

avafinger, 2019-10-24:
Follow the advice, use ONLY zram.

**ssh fix here:**
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/27#issuecomment-538750121

neilyoung, 2019-10-24:
Is this the suggestion?

> To fix the ssh issue you have, edit the fie /etc/network/interfaces and uncomment the two lines:
> #wireless-mode Managed
> #wireless-power off

I'm not using Wifi anymore. Just ETH. Same issues.

neilyoung, 2019-10-24:
> Follow the advice, use ONLY zram.

Thanks, but compilation runs WAY better with 2 GB swap.

avafinger, 2019-10-24:
Do you know the specifications of your PS?

Were you able to do :
`sudo apt-get update && sudo apt-get dist-upgrade`

Try top reinstall ssh, type in the shell (not remotely):
`sudo apt-get remove --purge openssh-server && sudo apt-get install openssh-server`
and then **sudo reboot**

neilyoung, 2019-10-24:
The power supply is the default RPI supply: 5V / 2.5A

Thanks for the suggestion. Will try tomorrow. Main issue tonight is to get the compilation through. Thanks for today. Your help is very much appreciated.

avafinger, 2019-10-24:
Ok, my last two advice

* check the FS every boot
` sudo tune2fs -c 1 -i 1 /dev/mmcblk0p2`

* start over again with a fresh image

neilyoung, 2019-10-24:
I suppose I have to do the latter. SSH does not respond anymore after the uninstall/re-install from shell :)
Not my day I guess.

neilyoung, 2019-10-24:
I think I will get a new SD tomorrow. It is pretty hard to get an 8GB anymore, but I purchased an expensive one. Hope, the PI is not that bitchy with this one.

https://www.amazon.de/gp/product/B00B978X8Y/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

neilyoung, 2019-10-24:
...ah no, reinstall not necessary. It just took 5 minutes again until SSH came up :) Additional coffee break.

neilyoung, 2019-10-24:
Hmm. Was trying to remove swap in favour of zram. Wondering, why the zram blocks are so small... Just 64 MB?
Would you say, it's ok?

```
ubuntu@bpi-m2z:~$ cat /proc/swaps
Filename				Type		Size	Used	Priority
/dev/zram0                              partition	63248	0	5
/dev/zram1                              partition	63248	0	5
/dev/zram2                              partition	63248	0	5
/dev/zram3                              partition	63248	0	5
ubuntu@bpi-m2z:~$ free
              total        used        free      shared  buff/cache   available
Mem:         506008       40800      403400        2676       61808      454176
Swap:        252992           0      252992
```

avafinger, 2019-10-24:
Hope this **400x** worth it...

If you do have a ubuntu linux box you should also try this way just in case: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v15

neilyoung, 2019-10-24:
> Hope this 400x worth it...

Sorry, what do you mean?

avafinger, 2019-10-24:
Speed: 400x!

>  why the zram blocks are so small.

You have only 512 Mbytes of RAM.

neilyoung, 2019-10-24:
Thanks for the other image link. Will try tomorrow (really now :)) Maybe a kernel step back is not that bad idea

> You have only 512 Mbytes of RAM.

Ok, I have heard of zram today for the very first time, sorry.

neilyoung, 2019-10-24:
Nope, zram does really not work...
I really wonder why this image has such a problem with the swap partition. Again: I did run the entire compilation with the same swap configuration on a single core RPI Zero. Except that it took 8 hours there was no problem at all...

Here is the result of the compiler. I was warned: Swap file partition too small - crash.

And it happened.

```
usr/include/python3.7m/pythread.h:97:17: note: declared here
   97 | PyAPI_FUNC(int) PyThread_set_key_value(int key, void *value) Py_DEPRECATED(3.7);
      |                 ^~~~~~~~~~~~~~~~~~~~~~
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:102: CMakeFiles/realsense2.dir/src/ds5/ds5-motion.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:76: CMakeFiles/realsense2.dir/src/ds5/ds5-timestamp.cpp.o] Error 1
```

neilyoung, 2019-10-24:
What BTW qualifies a "cheap SD card"? I have never read such a warning with Raspbian :)

> DON'T use cheap SD CARD, use a brand new and trusted SD CARD

neilyoung, 2019-10-24:
Now I got your 400x remark... I just purchased over the price :) I saw 64 GB cards with the same price and thought, it must be a good one if 8 GB is the same :)

avafinger, 2019-10-24:
Ok, i got your point. Your compilation runs out of memory, and it should not with zram.
zram can be configured, you can search for additional info and adjust it.

> configuration on a single core RPI Zero

Yes, building with a single core can use far less memory than four core.

The reason to use the new 8GB is just to start with a clean image before you make your changes.
8GB sd card will certainly run out of space soon when you start your build, while 32GB not.

I think you should:
1) Start with this way of burning the SD CARD:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v15

2) Configure the basics, and do an update && dist-upgrade (make it works)

3) Try to configure zram
https://www.cnx-software.com/2018/05/14/running-out-of-ram-in-ubuntu-enable-zram/

Once the SSH is resolved, and zram configured, move on with your changes.

neilyoung, 2019-10-24:
Ok, thanks. Is there a difference between the two ways you are outlining? I mean is there a difference in the result between a Linux step-by-step and a Mac/Windows etcher burn? I'm now already running the make just on a single instance, seeing the percentages going up, 1 by 1, taking an hour for each. I'm really baffled, how weak this chip is, where everybody says, it has the power of a Pi 3... Maybe later at runtime. It completely fails here at build time. Surprising.

neilyoung, 2019-10-24:
I have seen the article you are quoting regarding zram. I didn't manage to make a bigger swap partition than what you have seen a couple of posts above. And this didn't work.

neilyoung, 2019-10-24:
Thanks for your patience, btw.

avafinger, 2019-10-24:
>  Linux step-by-step and a Mac/Windows etcher burn?

The layout of the sd card is preserved in the linux way. 

Try to start over, step by step. Make your changes later.
No one has reported the issues you are having... Here the bpi-m2z works. :)

neilyoung, 2019-10-24:
I don't blame you or your image. It can be my sample. I for instance need to cool down the chip with a fan mounted to my MacBook. The first SD has been burned to ashes literally, in the slot while running the make, optimistically on four threads! Never seen this before with one of my many PIs. Up to now I didn't give a thought to the OS. I burned it and done. It was just a means.

neilyoung, 2019-10-25:
> The layout of the sd card is preserved in the linux way.

Does it automatically expand the memory uses, if I use a 32 GB instead of an 8 GB? I guess, I just used "Expand filesystem" up to now in raspi-config for that, if it is similar.

neilyoung, 2019-10-25:
BTW: The SDK I'm trying to build is this:
https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md

The PI3B is a perfect choice for this, but the PIZero is overloaded a lot at runtime. That's why I was trying to use the BPI, which was suggested as "PI3 in PIZ form factor" :) That has still to be shown, but I don't give up :)

avafinger, 2019-10-25:
Yes, if you use a 32 GB, all 32GB will be available, not the 8GB (from d=the img)

neilyoung, 2019-10-25:
Cool. I will do that than and report.
Good night now, I'm overdue.

avafinger, 2019-10-25:
Good.

You need 2 things to get the best from this board:
* Heat-sink (a good one)
  Yes, i know RPI does not mention you need a heat-sink.
  without the Heat-sink throttle occurs and you end up with 4 cores at 120 Mhz

* A server Image
  This Image has a portion of the memory reserved for the mali for the OpenGLES.
  You can get a better performance using this CMA reserved memory for applications.
  I will try to build a Server Image this week-end (or tonight), then you give another try..

neilyoung, 2019-10-25:
Sorry for the late reply. I let it run on battery (3800 mAH) for 5:30 h now, until it died. I also have measured the temperature of the chips, both between 40 and 45 degrees Celsius during the run.

The SSH problem is gone (maybe the new installation did it). There is yet another odd problem, compared to the "other" PIs: The camera is issuing 200 data sets of around 100 to 200 bytes per second, so it shouldn't cause an overload on USB 2. However, at application level the frame rate achieved was around 120 to 130, so I suppose, something is throttling the USB. My script has always seen the full 200 Hz on the other PIs, even on a PIZW. Does that ring a bell somehow?

Your server offer is very much appreciated, can't wait to flash that :) I would like to switch off BT, can this also be done using the dtb compiler?

Have a nice weekend, in case, we don't talk each other anymore.

neilyoung, 2019-10-25:
Some info regarding power consumption: The PI with the camera plugged but idling consumes about 130 mA. If the script is running, which starts the camera and obtains poses the power consumption is between 360 and 380 mA, but below 400. As a rule of thumb I would consider the couple to consume 400 mA, which _should_ allow for 6h operation on 2.600 mAH 18650B (theoretically, don't believe that, to be tested)

avafinger, 2019-10-25:
Use the device-tree-compiler to enable SPI (i think is spi0) and disable BT.

```
&spi0 {
   status = "okay"; 
}
```

For the BT i think you can remove **rfkill_bt** the node and/or **bt_pwr_pin**.

> The camera is issuing 200 data sets of around 100 to 200 bytes per second, so it shouldn't cause an overload on USB 2. However, at application level the frame rate achieved was around 120 to 130

Can you give more details about your FPS? What camera is in use, and what size of the frame. 120/130 FPS seems a lot, i don't understand what is the problem, details, please.

Check your USB camera speed:
dmesg|grep speed

avafinger, 2019-10-25:
low-speed: 1.5 Mbit/s
high-speed: 12 Mbit/s

[70 more replies]