avafinger/bananapi-zero-ubuntu-base-minimal #30: alsa audio

lbernstone, 2019-10-30:
As you know, I am trying to get RetroPie working on this device.  I am unable to get alsa sound on the HDMI output working with the Ubuntu 19.10 image.  No errors, but no sound.  I have disabled pulseaudio by unlinking the service and removing /etc/alsa/conf.d.  I have tried using both hw devices in speaker-test, but no sounds. My alsa layout looks like this:
```
$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=Codec
    H3 Audio Codec, CDC PCM Codec-0
    Default Audio Device
dmix:CARD=Codec,DEV=0
    H3 Audio Codec, CDC PCM Codec-0
    Direct sample mixing device
dsnoop:CARD=Codec,DEV=0
    H3 Audio Codec, CDC PCM Codec-0
    Direct sample snooping device
hw:CARD=Codec,DEV=0
    H3 Audio Codec, CDC PCM Codec-0
    Direct hardware device without any conversions
plughw:CARD=Codec,DEV=0
    H3 Audio Codec, CDC PCM Codec-0
    Hardware device with all software conversions
sysdefault:CARD=allwinnerhdmi
    allwinner,hdmi, 1c22800.i2s-i2s-hifi i2s-hifi-0
    Default Audio Device
dmix:CARD=allwinnerhdmi,DEV=0
    allwinner,hdmi, 1c22800.i2s-i2s-hifi i2s-hifi-0
    Direct sample mixing device
dsnoop:CARD=allwinnerhdmi,DEV=0
    allwinner,hdmi, 1c22800.i2s-i2s-hifi i2s-hifi-0
    Direct sample snooping device
hw:CARD=allwinnerhdmi,DEV=0
    allwinner,hdmi, 1c22800.i2s-i2s-hifi i2s-hifi-0
    Direct hardware device without any conversions
plughw:CARD=allwinnerhdmi,DEV=0
    allwinner,hdmi, 1c22800.i2s-i2s-hifi i2s-hifi-0
    Hardware device with all software conversions

$ amixer
Simple mixer control 'Line In',0
  Capabilities: pvolume pvolume-joined pswitch cswitch
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: Playback 0 - 7
  Front Left: Playback 7 [100%] [6.00dB] [on] Capture [off]
  Front Right: Playback 7 [100%] [6.00dB] [on] Capture [off]
Simple mixer control 'Line Out',0
  Capabilities: pvolume pvolume-joined pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 31
  Mono:
  Front Left: Playback 31 [100%] [0.00dB] [on]
  Front Right: Playback 31 [100%] [0.00dB] [on]
Simple mixer control 'Line Out Source',0
  Capabilities: penum
  Items: 'Stereo' 'Mono Differential'
  Item0: 'Stereo'
  Item1: 'Stereo'
Simple mixer control 'Mic1',0
  Capabilities: pvolume pvolume-joined pswitch cswitch
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: Playback 0 - 7
  Front Left: Playback 7 [100%] [6.00dB] [on] Capture [off]
  Front Right: Playback 7 [100%] [6.00dB] [on] Capture [off]
Simple mixer control 'Mic1 Boost',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 7
  Mono: 7 [100%] [42.00dB]
Simple mixer control 'Mic2',0
  Capabilities: pvolume pvolume-joined pswitch cswitch
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: Playback 0 - 7
  Front Left: Playback 7 [100%] [6.00dB] [on] Capture [off]
  Front Right: Playback 7 [100%] [6.00dB] [on] Capture [off]
Simple mixer control 'Mic2 Boost',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 7
  Mono: 7 [100%] [42.00dB]
Simple mixer control 'Mixer',0
  Capabilities: cswitch
  Capture channels: Front Left - Front Right
  Front Left: Capture [off]
  Front Right: Capture [off]
Simple mixer control 'Mixer Reversed',0
  Capabilities: cswitch
  Capture channels: Front Left - Front Right
  Front Left: Capture [off]
  Front Right: Capture [off]
Simple mixer control 'ADC Gain',0
  Capabilities: cvolume cvolume-joined
  Capture channels: Mono
  Limits: Capture 0 - 7
  Mono: Capture 3 [43%] [0.00dB]
Simple mixer control 'DAC',0
  Capabilities: pvolume pvolume-joined pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 63
  Mono:
  Front Left: Playback 63 [100%] [0.00dB] [on]
  Front Right: Playback 63 [100%] [0.00dB] [on]
Simple mixer control 'DAC Reversed',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [off]
  Front Right: Playback [off]
```

lbernstone, 2019-10-30:
I do get some issues with the codec device in dmesg:
```
ubuntu@bpi-m2z:~$ dmesg | grep codec
[    7.938570] cedrus 1c0e000.video-codec: Device registered as /dev/video0
[    8.082250] sun4i-codec 1c22c00.codec: ASoC: codec-analog@1f015c0 not registered
[    8.089901] sun4i-codec 1c22c00.codec: Failed to register our card
[    8.121327] sun4i-codec 1c22c00.codec: ASoC: codec-analog@1f015c0 not registered
[    8.128915] sun4i-codec 1c22c00.codec: Failed to register our card
[    8.170637] sun4i-codec 1c22c00.codec: ASoC: codec-analog@1f015c0 not registered
[    8.178185] sun4i-codec 1c22c00.codec: Failed to register our card
[    8.195871] debugfs: Directory '1c22c00.codec' with parent 'H3 Audio Codec' already present!
[    8.211924] sun4i-codec 1c22c00.codec: ASoC: Failed to create component debugfs directory: -17
[    8.227700] sun4i-codec 1c22c00.codec: Codec <-> 1c22c00.codec mapping ok
```

avafinger, 2019-10-30:
Firstly you should not remove the link! And pulseaudio should be enabled, not a requirement but better so.

Step to get Audio (HDMI output)
- Restore the previous Image

- Enable all output (amixer)

- Configure default sound output to HDMI
  /etc/asound.conf

special attention to "Play audio to HDMI" section here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/27#issuecomment-538648533

lbernstone, 2019-10-30:
Didn't see that issue, as my bluetooth is working :smile:.  I'll reset and give it a shot.

avafinger, 2019-10-30:
> sun4i-codec 1c22c00.codec: Codec <-> 1c22c00.codec mapping ok

This is not HDMI, and it was finally mapped at the end, just ignore it.

> Didn't see that issue, as my bluetooth is working smile. I'll reset and give it a shot.

Good, save this configuration for later reference! Maybe you found how it should be configured!

lbernstone, 2019-11-04:
Yeah, it seems like it is all working.  I can see the device and alsamixer takes me directly to the allwinner,hdmi hw device.  There's no errors about an interrupted stream or timeout.  I did have to add the ubuntu user to the audio group along with the asound.conf, but then everything looks like you show in that thread.  There are no options in the mixer for the hw:allwinner,hdmi device, so I can't "enable all output" there.

lbernstone, 2019-11-04:
I just tested with bluetooth.  I am able to pair a speaker, it automatically switches to that sink, plays the audio, but it stutters a lot.

lbernstone, 2019-11-15:
Any ideas how to troubleshoot this?  Do you get sound with the U19.10 image?

avafinger, 2019-11-15:
BT? I got it working without "stuttering". It did not survive a reboot, the same procedure on the next boot disconnect the BT every time i try.

Hdmi-sound works on Ubuntu 19.10. To enable hdmi sound you need to configure asound.conf

```
pcm.!default {
    type plug
    slave {
        pcm "hw:1,0"
    }
}
		
ctl.!default {
   type hw
   card 1
} 
```

Test with:
`sudo aplay -D sysdefault:CARD=1 /usr/share/sounds/alsa/Front_Left.wav `

lbernstone, 2019-11-15:
No sound.  No errors.  I don't think this tv has anything weird going on with it.  Other devices work out of the box.  The old jesse/armbian image gives me sound (but the dtb is compiled into the kernel, so I can't compare).

avafinger, 2019-11-15:
Do you have a spare sd card and a linux box?

avafinger, 2019-11-15:
I can make an Image 8GB with my setup... No linux Box needed.

lbernstone, 2019-11-15:
Sure.  I have lots of stuff kicking around :)

avafinger, 2019-11-15:
I will update kernel and prepare an 8GB Image and you give it a try, it takes some time to prepare one here...

avafinger, 2019-11-16:
Ubuntu 19.10

* mali-fbdev
* hdmi-sound

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.8

avafinger, 2019-11-16:
re-uploading the 7z img file..

avafinger, 2019-11-16:
You can play mp3 like this:

```
play ff-16b-2c-44100hz.mp3 

ff-16b-2c-44100hz.mp3:

 File Size: 2.99M     Bit Rate: 128k
  Encoding: MPEG audio    Info: 2016-04-12
  Channels: 2 @ 16-bit   
Samplerate: 44100Hz      
Replaygain: off         
  Duration: 00:03:07.12  

In:2.63% 00:00:04.92 [00:03:02.19] Out:217k  [   -==|==    ]        Clip:0    

```

lbernstone, 2019-11-16:
Still no sound.  It doesn't make any sense for you to be trying to troubleshoot my hw problems.  I'd like to figure out how to make the kernel myself.  Which is the appropriate mali driver to be using?  DX910-SW-99002-r9p0-01rel0.tgz from https://developer.arm.com/tools-and-software/graphics-and-gaming/mali-drivers/utgard-kernel ?

lbernstone, 2019-11-16:
I am now seeing this in the dmesg log.  Not sure if it is involved with the hdmi sound, but it an i2s error seems like it could be related:
```
[  495.777327] sun4i-i2s 1c22800.i2s: Unsupported MCLK divider: -22
[  495.783367] sun4i-i2s 1c22800.i2s: ASoC: can't set 1c22800.i2s hw params: -22
[  515.687000] sun4i-i2s 1c22800.i2s: Unsupported MCLK divider: -22
[  515.693058] sun4i-i2s 1c22800.i2s: ASoC: can't set 1c22800.i2s hw params: -22
```

avafinger, 2019-11-16:
I don't get any errors while playing music.

Here is the tip, use the Armbian Kernel that you say hdmi-sound works:

* Get the Armbian Kernel
* Get the sunxi-mali from mripard, here: https://github.com/mripard/sunxi-mali
* Get the mali-blobs, r6p2: https://github.com/bootlin/mali-blobs

Good luck.

closed 2020-06-21 by lbernstone