avafinger/bananapi-zero-ubuntu-base-minimal #31: Is it possible to add xorg to this?

ljones0, 2019-11-14:
Hello,

Not sure if xorg can be run on this image but however when I tried to run xorg (and from that a desktop) I get the following error:

/usr/lib/xorg/Xorg: symbol lookup error: /usr/lib/xorg/modules/libglamoregl.so: undefined symbol: gbm_create_devicexinit: giving up

Am I missing an xorg driver at all?  I do have xserver-xorg-video-fbdev installed btw.

ljones

lbernstone, 2019-11-15:
Which image?  You do not want the one with fbdev in the name if you are running X.

avafinger, 2019-11-15:
Make sure you can run **kmscube** in your Image . Get it from here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/edit/v14

libglamoregl is for 3D rendering!
https://www.freedesktop.org/wiki/Software/Glamor/

ljones0, 2019-11-16:
Sorry I was referring to the "Ubuntu 19.10 (EOAN) - fbdev" image. It is the one with fbdev in it so is that the wrong one? I did see others, e.g. "Ubuntu 19.10 - Eoan Ermine with Kernel 5.3.5" and there's a list beneath it which lists things like hdmi, hdmi sound, mali, bt etc but no mention of xorg. Which is the corret image?

ljones

avafinger, 2019-11-16:
I did not know Glamour is set when you install Xorg.

You can try the latest Image with fbdev and install LXDE.
After you boot and login with ubuntu / ubuntu you type in shell:

**sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install lxde**

If you install some other Xorg that Glamour is set by default it will try to use mali OpenGLES2 and for that to work you need mali-gbm. You need Wayland.

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.8

meisterbasti, 2019-12-16:
I have a question about this Linux version.
1st. why does it take so long while powering the banana pi until it is possible to login with ubuntu / ubuntu ?

2nd. After installing lxde like in the comment above noting happens. After rebooting only the black screen with the white underscore ( _ ) appears. Hitting the ENTER key about 100 times doesn't change anything. 

I want to build a Linux version with a actual Kernel for a project at university with the Banana Pi M2 Zero  

[image]

avafinger, 2019-12-16:
I think it is taking so long because it keeps waiting for an IP from DHCP either from eth0 (it is enabled) or wlan0 (wifi). The login prompt appears **only after** it grabs an IP.
You can study /etc/network/interfaces and change eth0 and wlan0 to 'allow-hotplug ' so it does not wait until it gets the IP in any interface. The drawback is it can take a really long time to acquire the IP.

Look into Xorg.0.log for X11 errors for some clues.

meisterbasti, 2019-12-17:
How do I get this log when the system is not responding ?
The underscore is flashing some seconds and than freezes and nothing else happens... 
Sorry for silly questions but I do not have much experience in Linux...

avafinger, 2019-12-17:
https://www.youtube.com/watch?v=AywJsMLogRY

https://www.youtube.com/watch?v=8upT2NymUzg

https://www.adafruit.com/product/954

avafinger, 2019-12-17:
https://www.youtube.com/watch?v=uUTwcuq0Ha4

but be careful the one you choose, some cheap on has > 3.4v that don't work.

meisterbasti, 2019-12-17:
I thought this answer might come.

Ordered the one from Adafruit.

avafinger, 2020-02-16:
@meisterbasti 
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/33