avafinger/bananapi-zero-ubuntu-base-minimal #32: libgdk-3.so.0: undefined symbol: wl_egl_window_destroy

meisterbasti, 2020-01-04:
Hey, 

I am trying to compile OpenCV from source for the Bpi Zero but the compile is not successful. it is aborting with an  "wl_egl_window_destroy" error and several other "wl_egl_window" errors.

[image]

I tried to reinstall libgtk-3-0 but I received a similar error:

`Setting up libgtk-3-0:armhf (3.24.12-1ubuntu1) ...
/usr/lib/arm-linux-gnueabihf/libgtk-3-0/gtk-query-immodules-3.0: symbol lookup error: /lib/arm-linux-gnueabihf/libgdk-3.so.0: undefined symbol: wl_egl_window_destroy`

Any ideas what is causing this error and any suggestions how to fix it ?

Regards
Sebastian

avafinger, 2020-01-07:
Are you using mali fbdev Image or mali GBM Image?

avafinger, 2020-01-07:
Use the GBM version: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.4

```
0008f0bd T wl_egl_window_create
0008f1ad T wl_egl_window_destroy
0008f1e9 T wl_egl_window_get_attached_size
0008f149 T wl_egl_window_insert_surface
0008f119 T wl_egl_window_ref
0008f171 T wl_egl_window_remove_surface
0008f091 T wl_egl_window_resize
0008f125 T wl_egl_window_unref

```

meisterbasti, 2020-01-07:
Thanks for your reply. I’m using the v2.8 version which is the fbdev. 

Can you give me some details what is the difference between this two images, the mali fbdev and mali GBM Image? 
 Is it not possible do install the missing wl_egl_window_* stuff ?
 Sorry for the questions but I am Linux beginner and want to understand.

avafinger, 2020-01-10:
mali fbdev does not have support for X11 / Wayland window stuff.

In theory, you can just use the right mali blob, copying the Wayland blobs from here: https://github.com/bootlin/mali-blobs/tree/master/r6p2/arm over the fbdev blobs.

But you need a way to add more CMA, try to pass cma=256 to the kernel in boot.cmd.
512 MBytes is not enough to have a good Desktop experience, but it may work.

meisterbasti, 2020-02-16:
The GBM Image fixed that issue.

closed 2020-02-16 by meisterbasti