avafinger/bananapi-zero-ubuntu-base-minimal #33: Few questions about the system

meisterbasti, 2020-02-15:
As there are no private messages I want to ask a few questions.

1 .) Is there a way to auto-login the ubuntu user ?

2.) Is there a possibility to permanent deactivate eth0 ? I only need the wlan0 interface...

3.) Last question is while startup:
Here I'm getting the message: "A start job is running for DHCP Client..."

[image]

This makes the booting time very long as you can see it is counting up to 1m 38 sec. resulting in a "Failed to start" message

[image]

Looking into systemctl gives me the info that it is Failed. Therefore I think it is not used. So is it possible to deactivate it so that the booting will go faster ?

[image]

Thanks for your help.

avafinger, 2020-02-15:
> 1 .) Is there a way to auto-login the ubuntu user ?

Yes, there is. Please look at these instructions here, you may need some twist:
https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal#libreelec-or-not-libreelec

> 2.) Is there a possibility to permanent deactivate eth0 ? I only need the wlan0 interface...

You can disable eth0 (which i have it working here) by doing this:

* edit the file: **/etc/network/interfaces**
* Comment the following lines:
**#**auto eth0
**#**iface eth0 inet dhcp
* reboot

You can edit it while running the kernel or do a shutdown and edit the file in a linux box.

> 3.) Last question is while startup:

Here I'm getting the message: "A start job is running for DHCP Client..."

DHCP Client is trying to get a Dynamic IP from the AP or Router, if you proceed with item 2) it will not try to grab an IP for the eth0 interface, but will still try to get IP for the wlan0 interface.

You have to properly set up your wlan0 interface with a password for your AP.

meisterbasti, 2020-02-15:
1.) Worked for me. Thx !!

2.) Does not work for me. I commented it out but noting happened. Also I removed everything that I don't ned but still not working for me.

[image]

`sudo ifconfig eth0 down` switched the interface offline but after reboot it is up again....

avafinger, 2020-02-15:
So, you removed eth0 interface from the files and you still get the eth0 up?

avafinger, 2020-02-15:
try:

```
#auto eth0
iface eth0 inet dhcp
allow-hotplug eth0
```

meisterbasti, 2020-02-16:
yes I removed eth0 from the interface file and it is still up after reboot.

Adding your lines  results in the same. eth0 is still up.

> #auto eth0
iface eth0 inet dhcp
allow-hotplug eth0

I searched for "permanent deactivate eth0" but every solution I tried had no success.

avafinger, 2020-02-16:
Please try:

* edit /etc/dhcp/dhclient.conf
* change timeout 300; to timeout 5;

and

```
allow-hotplug eth0
iface eth0 inet dhcp

```

not sure if the order of lines are important.

avafinger, 2020-02-16:
and you might need:

```
sudo apt-get update
sudo apt-get install ifupdown-extra
```

Please, change the subject to something like: "Board takes 01:30 min to display login prompt" , or something like that.

meisterbasti, 2020-02-16:
Good news. I think that your suggestion worked. I changed the dhclient as you described.

> Please try:
> 
> * edit /etc/dhcp/dhclient.conf
> * change timeout 300; to timeout 5;

I also installed ifupdown-extra. The system is logging-in so much faster. Boot-up time is now as I would expect it.

The second suggestion has no positive effect. Therefore I uncommented it in the interface file again.

But still I do not understand why eth0 always goes "up" after rebooting. Isn't there a command or setting that disable it permanent ?

avafinger, 2020-02-16:
I can only think some other service is doing the work for the eth0, this should be a feature on 19.10, some need to investigate.

dazKind, 2020-05-22:
Hi, 
the wait during boot got on my nerves as well. I'm configuring the devices later in the boot and i figured I could get away with modifying the dhclient.unitfile and supply the `-nw` flag to dhclient and it worked. 

See https://linux.die.net/man/8/dhclient

> -nw
Become a daemon process immediately (nowait) rather than waiting until an IP address has been acquired.

Steps to make this work: 
1. Open the unit-file: `sudo vim /etc/systemd/system/dhclient.service`
2. Add the `-nw` flag: `ExecStart=/sbin/dhclient -4 -q -nw`

P.S.: @avafinger awesome work on this image btw!

Cheers

nordurljosahvida, 2020-10-10:
all of this worked great thanks, `-nw` on dhclient, lower timeout in dhclient.conf, and eth0 settings. now system reboots in 30 seconds from last ping to first ping. how do you get output on display during boot? i only have a blinking cursor.

avafinger, 2020-10-10:
> how do you get output on display during boot? i only have a blinking cursor.

You need a more recent u-boot with hdmi enabled. Try the latest **mainline** u-boot.