avafinger/bananapi-zero-ubuntu-base-minimal #34: Missing linux headers in 4.17.4

aronfeher, 2020-03-23:
Hi!

First and foremost thank you for this amazing work. You saved me a lot of time and effort with this kernel. I've downloaded the given image with the kernel v2 (4.17.4), and the system works fine. The only problem is I can't find the Linux headers so that I can build some kernel space drivers.
In /usr/src/linux-headers-4.17.4-mz there is only an include folder. The apt-cache search Linux-headers-$(uname -r) won't return anything. 

Should I download the Linux kernel, build it, and cross-compile the headers, or it's included in the image, and only I can't find it?
Or could a kernel upgrade (e.g. v11) fix my issue?

Any help is appreciated.

Thank you in advance,
Aron Feher.

avafinger, 2020-03-23:
Hi @aronfeher ,

Do you have a need for this specific kernel version?
You could flash this kernel here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.4
which has the kernel headers.

But you can install 4.17.8 here: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/commit/70ddae856875b7572d83b5d6310018b9229799af
It will install the headers.

The last kernel 4.17 is the kernel 4.17.18.

aronfeher, 2020-03-24:
Thank you for the quick reply @avafinger.
Today I've tried the kernel 4.17.18, 4.20.0-rc3. Sadly I couldn't flash the 5.3.5 Kernel, since my 8GB card has 43MB less space (if I have no choice, I'll build the whole system, but that is a last resort).

And here are what I've found:
- usr/src/linux-headers-4.x.x folder has only one 'include' folder. 
- typing `apt-cache search linux-headers-$(uname -r)` still won't return anything, which from my understanding means that somewhere the headers are broken.
- when I try to compile an SPI driver with the make file
`obj-m+=hello.o
all:
    make -C /lib/modules/$(shell uname -r)/build/ M=$(PWD) modules
clean:
    make -C /lib/modules/$(shell uname -r)/build/ M=$(PWD) clean`
an error occurs: "make[1]: *** /lib/modules/4.20.0-rc3-m2z-otg-gadget/build/: No such file or directory.  Stop."
but listing the modules: `ls /lib/modules/4.20.0-rc3-m2z-otg-gadget` returns `build  kernel         modules.alias.bin  modules.builtin.bin  modules.dep.bin  modules.order    modules.symbols      source
extra  modules.alias  modules.builtin    modules.dep          modules.devname  modules.softdep  modules.symbols.bin
`.

On top of that with kernel 4.20 I can't see the SPI in the /dev.

I'm starting to think that this problem is only because my lack of knowledge. So far I've only wrote drivers on systems with complete setups, so I don't even know where to look.

avafinger, 2020-03-24:
The linux-headers you see in: /usr/src:
linux-headers-5.3.18  linux-headers-5.3.18-fbdev  linux-headers-5.6.0-rc5+
have only the **include** folder.

This is so and can be used to build drivers out-of-tree. the **build** file is a symlink to the kernel source code which you don't have in the provided Images.

You only need linux-headers to build drivers out-of-tree. Seems your driver needs kernel source code. If that is the case you can try to move to kernel 5.6.y.

**Kernel source**: https://github.com/avafinger/linux-5.6.y
**Image**: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.4 then update to **5.6.y**

PS: Not all 8GB card is equal, buy a Sandisk card or > 8GB and manually adjust the rootfs size and reclaim all available space.

or you have to build the Image yourself, all from scratch

.

aronfeher, 2020-03-31:
Sorry for the late reply.

I've changed the kernel to the latest 5.6.y, and enabled all the required peripherals (spi, pwm, camera) in the device tree.
Now I can see every peripheral in the kernel, and I could build drivers out-of-tree.
I still don't know why I couldn't build my drivers, but with this kernel everything is solved.

Thank you again for the help!

closed 2020-03-31 by aronfeher