avafinger/bananapi-zero-ubuntu-base-minimal #35: System is very slow time to time

papaiatis, 2020-03-24:
Got a banana pi m2 zero and followed the instructions in this repo to flash and install ubuntu onto it.

I noticed that from time to time (well, quite often) the system becomes very slow. Downloading packages from the Internet is very slow (~10-40kB/s), typing characters in SSH terminal appears very slowly, some commands just hang for a while (a couple of minutes, even a single `cd some-dir` command). But then it all starts working fine again for a while. 
I suspect this is some kind of I/O issue. I use a SanDisk ULTRA 8GB Class10 card. I also attached an external antenna to the WiFi chip.

Also, in some rare cases, the system gets totally unreachable.

After fresh system setup I only installed Python 3.7 and OctoPrint (which runs in the background) and enabled UART3 in kernel. 
Speaking of kernel, I have 4.17.4-m2z.

Any idea what could be wrong? 

I monitored system resources with `top` for a while and I didn't see high CPU load or low number of available memory.

avafinger, 2020-03-24:
Try this:

edit /etc/network/interfaces and uncomment:

# Disable power saving on compatible chipsets (prevents SSH/connection dropouts over WiFi)

    wireless-mode Managed
    wireless-power off

papaiatis, 2020-03-24:
I have these wifi settings already enabled in `interfaces` file.

avafinger, 2020-03-24:
Wifi has BT embedded and i also enabled eth0 (which i use). There is a report it takes a long time to connect to wifi, maybe you disable eth0 interface or blacklist the eth interface and also BT if you are not using it. Let's see if it helps.

Maybe this can help: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/33#issue-565762320

papaiatis, 2020-03-24:
I do not suspect networking as the root cause of this problem. Rather, I think, it's more like a general I/O problem. Do you know any steps to check I/O performance or issues?

avafinger, 2020-03-24:
iotop
iostat