avafinger/bananapi-zero-ubuntu-base-minimal #36: How to install latest (Kernel 5.6.0-rc5) release on M2Z?

bluepuma77, 2020-03-29:
Hi, my Banana Pi M2 Zero just arrived, as a faster replacement for the Raspberry Pi Zero. I want to run music player Mopidy, interact with an GPIO/SPI RFID-reader and an external USB speaker.

When I look at the latest release [Kernel 5.6.0-rc5](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.2) it states 

> Install - Use a spare sd card to test this release

How would I do that with `linux-image-5.6.0-rc5+_1.0-41.deb`? Shouldn't the release contain an image file to write to the SD card?

I will try with image [5.3.11-fbdev v2.8](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.8) for now, which contains an image within the assets.

Thanks, bluepuma

avafinger, 2020-03-29:
Yes, install img first: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.8
and then download the latest Kernel 5.6.0-rc5 deb and install in it.

bluepuma77, 2020-04-08:
Thanks for the hint :)

I now invested quite some time to tinker with the board, but I admit that I am not really happy. 

I tried your release, also a [Debian](http://forum.banana-pi.org/t/bpi-m2-zero-bpi-p2-zero-new-image-debian-10-lite-with-kernel-4-19-support-grub-2019-10-20/10063) and [Armbian](https://dl.armbian.com/bananapim2zero/). Every time I was navigating blindly, because somehow my 4K monitor would not show any image. With a USB-serial converter I could at least see the logs and could get on the command line.

My method for "flashing" an SD card on Mac:
```
diskutil list
sudo diskutil unmountDisk /dev/disk2
7z e -so ~/Downloads/bpi-m2z-8GB-sd-ubuntu_eoan_kernel_fbdev_5.3.11.img.7z | sudo dd bs=1m of=/dev/rdisk2
```

Your release v2.8 worked fine, recognised my external ethernet adapter. After a `sudo apt update && sudo apt -y upgrade` I get continuous error messages:
```
[ 4544.113606] ieee80211 phy0: brcmf_run_escan: error (-110)
[ 4544.119132] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-110)
```

Then I updated to your latest release with 
```wget --no-check-certificate "https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/download/v3.2/linux-image-5.6.0-rc5+_1.0-41.deb"
sudo dpkg -i linux-image-5.6.0-rc5+_1.0-41.deb```
There seems to be a problem with raising network interfaces and DHCP, also my external ethernet adapter is not recognised anymore. I think at  some point before I had a working WiFi connection, but not anymore.

Maybe I am just spoiled by using all kinds of Raspberry Pi generations where the software usually just workes. Don't get me wrong, I love your work, but this board might just not be for me. I look for a working software solution to put into an IoT device, which is reliably available via network - and a bit faster than the Raspberry Pi Zero W. 

It's just so sad that the manufacturer is putting so little effort into supporting the board on SBC distributions like Armbian or DietPi.

closed 2020-04-08 by bluepuma77

avafinger, 2020-04-08:
> Your release v2.8 worked fine, recognised my external ethernet adapter.

Because this kernel version (v2.8) has all the drivers and kernel 5.6.y does not. stick with v2.8. Or add the drivers for your USB dongle, build the kernel with the right drivers:
https://github.com/avafinger/linux-5.6.y

> to get to login takes like 3 minutes.

The fix would be to remove eth0 support (which I use) or apply this workaround here: 
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/33#issue-565762320

> external ethernet adapter.

Why external ethernet adapter when you have one on the board? The error is related to the firmware of your external ethernet adapter, maybe if you try to figure out which firmware is needed or updated (wrongly?) would fix the warning/error.

Raspberry Pi has a very big company behind it which pays for its support, developers, and projects, just to sell their chips.
I don't get paid for this work (which is expensive), and Armbian and DietPi are community-driven.

I think if you provide info about your USB ethernet and dmesg (prior to the update and after the update) would help to solve the problem, without information, NO one can help you.

bluepuma77, 2020-04-08:
Hi Avafinger, I completely understand that you are doing this on your free time and it's great that YOU ARE the community for Ubuntu on M2Z 😃  

As a single person it's hard to deliver a great product compared to others having larger (volunteer) teams. I personally think Sinovoip should pay you 👍 

I thought I could easily switch from Raspi Zero to the M2Z and continue as usual with more CPU power - but for me it's not as easy as expected, maybe because I know too little about Linux kernel inner workings. I just prefer to spend my time on my python application 😏