avafinger/bananapi-zero-ubuntu-base-minimal #38: wlan0 is not present after updating kernel to 5.6.0-rc7

aronfeher, 2020-04-05:
Hi!

I've started a new issue, since The last one (the header problem) was solved, but a new issue arose, which was different.

I've downloaded torvalds kernel 5.6.0-rc7, and built the complete image with my modules, and headers. I've installed the zImage, modules and linux headers. I've installed the device tree as you stated at the **build your own kernel** section.

When I restarted the BPI everything was seemingly OK - I did not try the WiFi at this stage, so I modified the device tree to have the PWM, SPI, I2C modules, which were present after the restart.

The problem is now wlan0 won't appear.
At boot-up the system failes to start DHCP Client, and fails to Raise network interfaces.
```ifconfig``` shows only ```eth0``` and ```lo```.
Writing ```ifup wlan0``` gives:
``` 
wpa_supplicant: /sbin/wpa_supplicant daemon failed to start
run-parts: /etc/network/if-pre-up.d/wpasupplicant exited with return code 1
ifup: failed to bring up wlan0
```
Do I have to change something in the Kernel, or the Device Tree? Or something after the build?

avafinger, 2020-04-05:
Use this kernel: https://github.com/avafinger/linux-5.6.y

aronfeher, 2020-04-06:
Ok, I've downloaded the kernel. ```git tag -l | less``` Returned nothign, so I presume it's the latest rc7.
After ``` make -j4 sunxi_defconfig``` the configuration was written, but two warnings arose:

`
drivers/mtd/devices/Kconfig:212:warning: config symbol defined without type
drivers/mtd/devices/Kconfig:214:warning: config symbol defined without type
`
and the instruction ```make INSTALL_MOD_PATH=output zImage ```
can't complete because of some scripts/mod/devicetable-offsets.c errors - the last few lines are:
`scripts/mod/devicetable-offsets.c:228:2: note: in expansion of macro ‘DEVID’
  228 |  DEVID(tee_client_device_id);
      |  ^~~~~
scripts/mod/devicetable-offsets.c:5:50: error: invalid application of ‘sizeof’ to incomplete type ‘struct tee_client_device_id’
    5 | #define DEVID(devid) DEFINE(SIZE_##devid, sizeof(struct devid))
      |                                                  ^~~~~~
scripts/mod/devicetable-offsets.c:228:2: note: in expansion of macro ‘DEVID’
  228 |  DEVID(tee_client_device_id);
      |  ^~~~~
scripts/mod/devicetable-offsets.c:7:9: error: ‘OFF_tee_client_device_id_uuid’ undeclared (first use in this function); did you mean ‘tee_client_device_id’?
    7 |  DEFINE(OFF_##devid##_##field, offsetof(struct devid, field))
      |         ^~~~
scripts/mod/devicetable-offsets.c:229:2: note: in expansion of macro ‘DEVID_FIELD’
  229 |  DEVID_FIELD(tee_client_device_id, uuid);
      |  ^~~~~~~~~~~
scripts/mod/devicetable-offsets.c:7:41: error: expected expression before ‘struct’
    7 |  DEFINE(OFF_##devid##_##field, offsetof(struct devid, field))
      |                                         ^~~~~~
scripts/mod/devicetable-offsets.c:229:2: note: in expansion of macro ‘DEVID_FIELD’
  229 |  DEVID_FIELD(tee_client_device_id, uuid);
      |  ^~~~~~~~~~~
scripts/mod/devicetable-offsets.c:5:29: error: ‘SIZE_wmi_device_id’ undeclared (first use in this function); did you mean ‘fsl_mc_device_id’?
    5 | #define DEVID(devid) DEFINE(SIZE_##devid, sizeof(struct devid))
      |                             ^~~~~
scripts/mod/devicetable-offsets.c:231:2: note: in expansion of macro ‘DEVID’
  231 |  DEVID(wmi_device_id);
      |  ^~~~~
scripts/mod/devicetable-offsets.c:5:50: error: invalid application of ‘sizeof’ to incomplete type ‘struct wmi_device_id’
    5 | #define DEVID(devid) DEFINE(SIZE_##devid, sizeof(struct devid))
      |                                                  ^~~~~~
scripts/mod/devicetable-offsets.c:231:2: note: in expansion of macro ‘DEVID’
  231 |  DEVID(wmi_device_id);
      |  ^~~~~
scripts/mod/devicetable-offsets.c:7:9: error: ‘OFF_wmi_device_id_guid_string’ undeclared (first use in this function)
    7 |  DEFINE(OFF_##devid##_##field, offsetof(struct devid, field))
      |         ^~~~
scripts/mod/devicetable-offsets.c:232:2: note: in expansion of macro ‘DEVID_FIELD’
  232 |  DEVID_FIELD(wmi_device_id, guid_string);
      |  ^~~~~~~~~~~
scripts/mod/devicetable-offsets.c:7:41: error: expected expression before ‘struct’
    7 |  DEFINE(OFF_##devid##_##field, offsetof(struct devid, field))
      |                                         ^~~~~~
scripts/mod/devicetable-offsets.c:232:2: note: in expansion of macro ‘DEVID_FIELD’
  232 |  DEVID_FIELD(wmi_device_id, guid_string);
      |  ^~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:101: scripts/mod/devicetable-offsets.s] Error 1
make: *** [Makefile:1112: prepare0] Error 2
`

avafinger, 2020-04-06:
Can you compare the scripts/mod (this kernel) and scripts/mod (from the Linus you just compiled successfully?)

aronfeher, 2020-04-06:
I've run a file compare, and no differences were found. I've checked them visually, they are the same.

aronfeher, 2020-04-08:
Update. I couldn't build the kernel because of some configs. I've made a clean clone of your linux-5.6.y kernel. Left the default sun8i_defconfig, and I could generate every file. 
After reboot I get the same problem: 
A start job is running for network interfaces, and DHCP Client.
Both of the jobs fail.
Only eth0, and lo.

aronfeher, 2020-04-23:
Do you have an image of the linux-5.6.y with working wlan0, so that I could compare the generated files, because I still couldn't figure out the issue with my build.

avafinger, 2020-04-23:
I will build one with 5.6.6 server. You can install Desktop or use it for building kernels.

aronfeher, 2020-05-21:
Hi again!
Sorry for the continuous bother, but I'm still having the same issues.
I download the deb package for the linux-5.6.13-server. and I install it. The system can connect to the Internet, but I can't build kernel modules, since the /usr/lib/modules/build is inaccessible - in ssh it shows up red.

I download the linux-5.6.13 from the official web site. I build it, and place the zImage, dtb, usr files to the correct locations (in this case the /usr/lib/modules/build is accessible and is shown blue), but the wifi disappears.  ifup gives the same message as above.

I even tried to use the 5.6.13 src and lib folders with the 5.6.13-server kernel, but that is not working because the kernel differs.

Could you please write the exact kernel number which was used to generate the linux-5.6.13-server, or a download location? As things are I'm starting to feel like dropping the BPi and using an ESP32 where I can control the hardware, and have internet access :)).

avafinger, 2020-05-21:
It is 5.6.3.
Before you build your kernel, type this: export LOCALVERSION="-server" ,  so you can build your modules and use it with the pre-built one.

If you build your own kernel from https://www.kernel.org/ it should boot but you have to enable BRCM (wifi):

```
CONFIG_WLAN=y
CONFIG_WLAN_VENDOR_BROADCOM=y
CONFIG_B43=m
CONFIG_B43_BCMA=y
CONFIG_B43_SSB=y
CONFIG_B43_BUSES_BCMA_AND_SSB=y
CONFIG_B43_SDIO=y
CONFIG_B43_BCMA_PIO=y
CONFIG_B43_PIO=y
CONFIG_B43_PHY_G=y
CONFIG_B43_PHY_N=y
CONFIG_B43_PHY_LP=y
CONFIG_B43_PHY_HT=y
CONFIG_B43_LEDS=y
CONFIG_B43LEGACY=m
CONFIG_B43LEGACY_LEDS=y
CONFIG_B43LEGACY_DEBUG=y
CONFIG_B43LEGACY_DMA=y
CONFIG_B43LEGACY_PIO=y
CONFIG_B43LEGACY_DMA_AND_PIO_MODE=y
CONFIG_BRCMUTIL=m
CONFIG_BRCMFMAC=m
CONFIG_BRCMFMAC_PROTO_BCDC=y
CONFIG_BRCMFMAC_SDIO=y

```
That will do the trick.

aronfeher, 2020-05-22:
Okay. Now I get it.
I was missing those CONFIG_B43 settings. For some reason I couldn't (and still can't) find them in menuconfig. After adding them manually the wlan0 reappeared, and now I can use the kernel modules and the wifi. Well this was a nice learning experience.

Thank you very much for this invaluable help!

closed 2020-05-22 by aronfeher