avafinger/bananapi-zero-ubuntu-base-minimal #39: wget and curl fail with certificate errors

minnixtx, 2020-04-06:
For instance

<pre><code>wget https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/download/v3.2/linux-image-5.6.0-rc5+_1.0-41.deb</code></pre>

<pre><code>--2020-04-06 11:47:26--  https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/download/v3.2/linux-image-5.6.0-rc5+_1.0-41.deb
Resolving github.com (github.com)... 140.82.114.4
Connecting to github.com (github.com)|140.82.114.4|:443... connected.
ERROR: cannot verify github.com's certificate, issued by ‘CN=DigiCert SHA2 Extended Validation Server CA,OU=www.digicert.com,O=DigiCert Inc,C=US’:
  Unable to locally verify the issuer's authority.
To connect to github.com insecurely, use `--no-check-certificate'.</code></pre>

avafinger, 2020-04-06:
Try: wget --no-check-certificate https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/download/v3.2/linux-image-5.6.0-rc5+_1.0-41.deb

minnixtx, 2020-04-06:
That only works by bypassing wget's built in cert checking. There is something wrong/broken here. I tried reinstalling ca certificates but still same problem. Makes it impossible to use certain scripts that rely on curl as well.

meisterbasti, 2020-04-08:
Had the same problem some time ago.

install curl with 
`sudo apt-get install curl`

download to local pc the "cacert.pem" from:

https://curl.haxx.se/docs/caextract.html

Transfer to banana pi with scp command:

`scp YOUR_FILE ubuntu@192.168.0.129: `

NOTE: The ":" at the end are required  !!

change file from .PEM to .CET with

`openssl x509 -in cacert.pem -inform PEM -out cacert.crt`

put file in right directory:

`sudo cp cacert.crt /usr/local/share/ca-certificates/cacert.crt`

update certificates:

`sudo update-ca-certificates`

This worked for me