avafinger/bananapi-zero-ubuntu-base-minimal #41: Desktop GUI?

gavan1, 2020-05-11:
It’s a non issue, general question.

I’ve ran the dist-upgrade and rebooted. Is this release Include the desktop GUI ?

The BPI just brings me to command prompt. Ubuntu 19.10.

Thank you for these releases.

avafinger, 2020-05-11:
Yes, it is a CLI (Command Line Interface) minimal where you can install a Desktop of your choice and/or add the packages you need and will use.

I suggest if you want a Desktop, use LXDE.
https://help.ubuntu.com/community/LXDE

```
sudo apt-get update
sudo apt-get dist-upgrade.
sudo apt-get install lxde

```

You have only 512 MBytes of DRAM, the Images with GBM and FBDEV use CMA=256 which leaves only around 200 MBytes free for the Desktop. Use the server version you don't need mali / lima support, or change CMA=128 in boot.cmd (/boot).

gavan1, 2020-05-12:
So...
sudo apt-get install lxde  and  CMA=128 edit.

Or.

I missed how to install the server version.

@avafinger 🙏🏽.

avafinger, 2020-05-12:
> I missed how to install the server version.

Just update the kernel:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.5

> Any additional info on how relevant Lubuntu is to this image?

    sudo apt-get install lubuntu-desktop

gavan1, 2020-05-13:
@avafinger lifesaver!

gavan1, 2020-05-14:
Installing Lubuntu wasn’t the best idea.
I’ll be reverting back to just the LXDE install.

@avafinger It’s not very clear how functional Bluetooth support is. Some of your threads show you’ve successfully sent audio.

Have your prepared an image where Bluetooth just works for Keyboard and Mouse?

gavan1, 2020-05-28:
bpi-m2z-8GB-sd-ubuntu_eoan_kernel_fbdev_5.3.11 + linux-image-5.6.13-server_1.0-44 bluetooth worked (keyboard pairing via bluetoothctl commands) 

Will close this.

closed 2020-05-28 by gavan1