avafinger/bananapi-zero-ubuntu-base-minimal #43: Directions to rebuild kernel and/or add kernel modules?

dazKind, 2020-05-27:
I wanna run a couple of bpi m2+ zeros with docker inside an ad-hoc wifi mesh.

I figured the 5.3.5 image would be a good start for a docker-host(small footprint, awesome thermals) but it seems that its missing the support for overlayfs in the kernel. The batman-adv module is another requirement. I looked into rebuilding the kernel myself but I'm quite unclear about the directions and the codebases/patches/configs used to create the supplied image&kernel. I can produce new kernels but I run into issues like missing wifi and the like. 

Happy about any info shedding some light on how to successfully rebuild/recreate the v24+ kernels!

P.S.: I found https://github.com/avafinger/linux-5.6.y and tickered with it but it seems I'm missing something there, too. No clue. 

Cheers

avafinger, 2020-05-27:
> I can produce new kernels but I run into issues like missing wifi and the like.

You should start with this configuration, disable lima and lower CMA. And add support for docker and the likes.
https://github.com/avafinger/linux-5.6.y/blob/master/arch/arm/configs/m2z_lima_defconfig

also:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/38#issuecomment-632062680

dazKind, 2020-05-28:
Awesome! That worked like a charm :)

closed 2020-05-28 by dazKind