avafinger/bananapi-zero-ubuntu-base-minimal #44: Emmc

mariusares, 2020-07-01:
Hi
First of all thank you for your great work. 
It's like a nice surprise ...

I have installed the bare ubuntu with kernel 14.7 from : 
https://mega.nz/#!obIAVQiA!G0CCABkLunCcG8hEqMf7NfjTBK1jgMytt2f0VdtOl08
I can't see the emmc on fdisk ... 
Do you recommend a different kernel version?
Is any chance to have a bare debian 10 os with a stable kernel too? I mean on debian/Raspbian you have 2 kernels. Version 3 what runs ok and smoothie but is no AP network mode compatible and Version 4,4 where load average is pretty high (over 1 on idle)
For those who use i2c, gpio, spi, serial and wifi ap mode, a bare system is great.
Thank you for your time

avafinger, 2020-07-02:
_I can't see the emmc on fdisk ..._
A: BananaPi M2Z does not have eMMC!

_Do you recommend a different kernel version?_
A:  Scroll down the github page and you see the kernel history and new kernel versions, personally i like the kernel 5.3 and 5.7.
Kernel 5.3 has great CPU Temp management and performance, Kernel 5.7 has a lot of bug fixes and it is maintained by the Linux community.

_Is any chance to have a bare debian 10_
A:  Sorry, i am an Ubuntu user but would like to move to Debian, it can happen soon or take years.
The idea behind this Image is to have a CLI (Command Line Interface and minimal) with few installed packages, M2Z has 512 MB of DRAM, users will not be pleased to run a Full Desktop, yet it is possible. You can install LXDE or try Armbian.

*Start with this IMG and upgrade the kernel to 5.7:*
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.4

*Upgrade Kernel:*
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.7

*Kernel and some IMG releases:*
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases

mariusares, 2020-07-02:
I understand, i chose banana pi over raspberry pi because rpi won't implement eMmc on their boards anytime soon, but it's so poor on software support.
Thanks to people like you who color a bit that side of the room :)
I'll try to get your kernel sources to compile them in a debian system ... When I'll get some time to be honest ...

avafinger, 2020-07-02:
This board has eMMC but the layout is a bit different:
http://www.banana-pi.org/m2pz.html

mariusares, 2020-07-02:
This is the board i have right now
Si basically you are suggesting to upgrade the os image from Ubuntu 16.4 to ubuntu 19.* and kernel 5.7
I understand that Ubuntu 16.4 it's LTS correct? 
Can I use the kernel 5.7 and configuration on Ubuntu 16.4?
I probably look stupid but i cant find the download links :)
Thank you

mariusares, 2020-07-02:
I got the links
Sorry I'm on mobile so ...

avafinger, 2020-07-02:
Ubuntu 19.10

avafinger, 2020-07-02:
Ubuntu 19.10 - Eoan Ermine with Kernel 5.3.5

avafinger, 2020-07-02:
_Can I use the kernel 5.7 and configuration on Ubuntu 16.4?_
yes, i think it is possible, but the deb package will install the modules in /usr/lib/modules and i think Ubuntu 16.04 LTS has it in /lib/modules

If after you boot and you don't see the modules with **lsmod** you will have to move /usr/lib/modules/KERNEL_VERSION to /lib/modules/

If this happens you won't be able to ssh to the board and will need to use HDM or the serial debug

mariusares, 2020-07-02:
I'll give it a try.
What I need is Gpio, uart, i2c , spi , eth, wlan+ap mode and stability ...
Let me install it and I'll get back to you
Thank you

mariusares, 2020-07-02:
ok so kernel 5.3 and 5.7 only detects /dev/mmcblk0 ...
there is no way to detect eMMC for some reason
any clue please?

avafinger, 2020-07-02:
Try this:

1 - Update u-boot (only the bootloader) from this tag:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v1.7

2 - make sure eMMC is active
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/18

3 - Make sure the root device is correct for eMMC, change it in /boot/boot.cmd

mariusares, 2020-07-02:
there is no /dev/sd* ... 
dd if=./bootloader_2z_v3.bin conv=notrunc bs=1k seek=8 of=/dev/mmcblk0p1 
so /dev/mmcblk0p1  is the /boot sd card partition ... 
is that OK?

avafinger, 2020-07-02:
/dev/sd* is for USB SDHC reader/writer (if you connect to a PC box running Linux)
/dev/mmcblk0p1 that is OK if you do from M2Z

mariusares, 2020-07-02:
i did this :
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/11#issuecomment-461130750 
basically i can see the eMMC now but i can`t boot from eMMC ...
do you have the kernel+kernel config able to boot from eMMC or we have to do that manually?
i`m on Ubuntu 19.10 now with kernel 5.3 ...
root@bpi-m2z:~# uname -a
Linux bpi-m2z 5.3.5 #1 SMP Mon Oct 7 21:22:23 -03 2019 armv7l armv7l armv7l GNU/Linux
root@bpi-m2z:~# fdisk -l
Disk /dev/mmcblk0: 29.29 GiB, 31439454208 bytes, 61405184 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x2474af91

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       49152   253951   204800  100M 83 Linux
/dev/mmcblk0p2      253952 15777791 15523840  7.4G 83 Linux

Disk /dev/mmcblk2: 7.29 GiB, 7818182656 bytes, 15269888 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x2474af91

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk2p1       49152   253951   204800  100M 83 Linux
/dev/mmcblk2p2      253952 15777791 15523840  7.4G 83 Linux
root@bpi-m2z:~#

mariusares, 2020-07-02:
i`m trying to change some settings here 
# setup boot_device
fdt set mmc${boot_mmc} boot_device <2>

and i get the same result on boot 
U-Boot SPL 2017.09-h2 (Jul 06 2018 - 10:16:12)
DRAM: 512 MiB
Trying to boot from MMC2
MMC Device 1 not found
spl: could not find mmc device. error: -19
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###

mariusares, 2020-07-02:
There is no boot.cmd ... I'll try to decompile boot.scr and edit it in order to take action ...
Let's see the result ...

avafinger, 2020-07-02:
[boot.cmd.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/4866398/boot.cmd.tar.gz)

to be able to boot from eMMC you have to format it and copy the bootloader, boot, and rootfs.
There is a script that does that, search the releases and find the latest and instead of passing /dev/sdX you will pass /dev/mmcblkY

mariusares, 2020-07-02:
well is not working for me unfortunately ...

avafinger, 2020-07-02:
Boot from SD card and see what you get with this command:

**ls /dev/mmcblk***

mariusares, 2020-07-03:
root@bpi-m2z:~# ls /dev/mmc*
/dev/mmcblk0    /dev/mmcblk0p2  /dev/mmcblk2boot0  /dev/mmcblk2p1
/dev/mmcblk0p1  /dev/mmcblk2    /dev/mmcblk2boot1  /dev/mmcblk2p2
root@bpi-m2z:~#
its strange ...
the bootloader should read the sd card and emmc as well ...

avafinger, 2020-07-03:
1 - when booting from **SD CARD**:
/dev/mmcblk0 is your SD CARD
/dev/mmcblk2 is your eMMC

2 - when booting without SD CARD
        
`              /dev/mmcblk? - It changes device  number, you should figure out what number will be, possible 1.`

3 - Update u-boot with: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.4

4 - Use this scripts to format and install kernel and rootfs and update bootloader with the new version:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v15
```
              sudo ./format_sd_mainline.sh  /dev/mmcblk2
              sudo ./flash_sdcard_mz2_v2.sh  /dev/mmcblk2

```
reboot without SD CARD, i think a debug ttl uart is indispensable

avafinger, 2020-07-03:
try: 
`fdt set mmc${boot_mmc} boot_device <boot_mmc>`

mariusares, 2020-07-03:
i`m controlling the board via uart ttl 
the image does not see emmc without modification
the bootloader does not boot from emmc 

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.4 

fdisk -l:
root@bpi-m2z:~# fdisk -l
Disk /dev/mmcblk0: 29.29 GiB, 31439454208 bytes, 61405184 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x2474af91

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       49152   253951   204800  100M 83 Linux
/dev/mmcblk0p2      253952 15777791 15523840  7.4G 83 Linux
root@bpi-m2z:~#

still can`t see the emmc ...
so i have to use 

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/11#issuecomment-461130750
bpi-m2-zero-v4.dtb_eMMC.zip

with this dtb the emmc is /dev/mmcblk2 and i don`t think that would be changed after that ... so ... i don`t know what would be the resolution ...

avafinger, 2020-07-03:
This DTB has eMMC enabled, try with this dtb.
[bpi-m2-zero-v4.dtb_emmc.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/4870567/bpi-m2-zero-v4.dtb_emmc.tar.gz)

* unzip it into /boot
* sudo ln -sf bpi-m2-zero-v4.dtb_emmc bpi-m2-zero-v4.dtb
* sync && sudo reboot

This for kernel 5.7.x-server

mariusares, 2020-07-03:
you mean this one?
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.7

i can only see that 5.7 kernel but no "server" tag on it

avafinger, 2020-07-03:
yes!

mariusares, 2020-07-03:
this is output from sd-card boot 
root@bpi-m2z:~# fdisk -l && uname -a && ls -la /boot && cat /etc/issue*
Disk /dev/mmcblk0: 29.3 GiB, 31439454208 bytes, 61405184 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x58b49123

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       49152   253951   204800  100M 83 Linux
/dev/mmcblk0p2      253952 15532031 15278080  7.3G 83 Linux

Disk /dev/mmcblk2: 7.3 GiB, 7818182656 bytes, 15269888 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x58b49123

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk2p1       49152   253951   204800  100M 83 Linux
/dev/mmcblk2p2      253952 15532031 15278080  7.3G 83 Linux
Linux bpi-m2z 5.7.6 #1 SMP Sun Jun 28 00:11:55 -03 2020 armv7l armv7l armv7l GNU/Linux
total 12220
drwxrwxr-x  3 ubuntu ubuntu    4096 Jul  3 17:39 .
drwxr-xr-x 21 root   root      4096 Jan 15  2018 ..
lrwxrwxrwx  1 root   root        20 Jan 15  2018 boot.scr -> bpi-m2-zero_boot.scr
-rw-r--r--  1 root   root     26886 Jul  6  2018 bpi-m2-zero-v1.dtb_4.17.4-m2z
-rw-r--r--  1 root   root     27076 Jul  6  2018 bpi-m2-zero-v2.dtb_4.17.4-m2z
-rw-r--r--  1 root   root     28817 Jun 28 04:30 bpi-m2-zero-v4.dtb_5.7.6
-rw-rw-r--  1 ubuntu ubuntu   28813 Jul  3 16:15 bpi-m2-zero-v4.dtb_emmc
lrwxrwxrwx  1 root   root        23 Jul  3 17:39 bpi-m2-zero.dtb -> bpi-m2-zero-v4.dtb_emmc
-rw-r--r--  1 root   root       924 Jul  6  2018 bpi-m2-zero_boot.cmd
-rw-r--r--  1 root   root       996 Jul  6  2018 bpi-m2-zero_boot.scr
-rw-rw-r--  1 ubuntu ubuntu  978454 Jul  6  2018 initrd.img
drwx------  2 root   root     16384 Nov  5  2017 lost+found
lrwxrwxrwx  1 root   root        12 Jul  3 17:34 zImage -> zImage_5.7.6
-rwxr-xr-x  1 root   root   5376808 Jul  6  2018 zImage_4.17.4-m2z
-rwxr-xr-x  1 root   root   6000368 Jun 28 03:48 zImage_5.7.6
Ubuntu 16.04.4 LTS \n \l

Ubuntu 16.04.4 LTS
root@bpi-m2z:~#

 dd if=/dev/mmcblk0 of=/dev/mmcblk2 bs=10M status=progress
then i removed the sd-card and boot from eMMC ...
output:

U-Boot SPL 2020.04 (Apr 21 2020 - 14:52:45 +0000)
DRAM: 512 MiB
Trying to boot from MMC2
MMC Device 1 not found
spl: could not find mmc device 1. error: -19
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###

on Ubuntu 19.10 the same ... so that dtb has a problem by booting from emmc ...

avafinger, 2020-07-03:
Who told you this instructions would work?

_dd if=/dev/mmcblk0 of=/dev/mmcblk2 bs=10M status=progress
then i removed the sd-card and boot from eMMC ..._

mariusares, 2020-07-03:
well how can i boot from eMMC ?

avafinger, 2020-07-03:
Trying to boot from MMC**2**
MMC Device **1** not found

boot.cmd must tell the boot loader what device number eMMC is.
If you boot with SD CARD, eMMC has ID number X, if you boot WITHOUT SDCARD eMMC has ID number Y.

I will release a new Kernel and will prepare the script for writing to eMMC and try to guess which ID number we can try.
This takes some time, i am at work now.

mariusares, 2020-07-03:
i replaced 0 to 3 and is the same error 
scr was recompiled every time so ...
Trying to boot from MMC2
MMC Device 1 not found
spl: could not find mmc device 1. error: -19
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###

avafinger, 2020-07-03:
You are doing this on eMMC/boot , right?

mariusares, 2020-07-03:
Correct

mariusares, 2020-07-03:
Ubuntu 16.04 image is more faster than 19.10 ... I mean that particular image. I read that 19.10 is advertised to be the fastest Ubuntu ever but i personally don't see that. In the other hand 16.04 is insane faster ... 
If we manage to boot it from emmc that's something else ...

avafinger, 2020-07-03:
I have this feeling too, but the 19.10 is not optimized.
I will finish the script and you try again, maybe you poke that user who got this working on eMMC.

mariusares, 2020-07-03:
Well I'm looking at different image and if the sd card is inserted then will boot from sd card, if the image is copied to eMMC, will boot from eMMC when SD is not inserted. We speaking about same settings, so i presume that there's different u-boot configuration ...

avafinger, 2020-07-05:
Here are two files with device 1 and device 2.
Please, search the releases for a version which format and flash the files. I have not finished for the new kernel, please, use an old one, just to boot from eMMC.

Test first with bpi-m2-zero_boot_2.scr and then bpi-m2-zero_boot_1.scr

Instructions

* boot from SD CARD and make sure eMMC is visible
* Download the scripts (and old kernel) and the file attached
* format and install using /dev/mmcblk2
* mount eMMC and flash the new bootloader to eMMC
* go to /boot in eMMC/boot (not in the SD CARD)
* unzip the boot_device.scr.tar.gz there
* link with the respective file to taste the boot from eMMC
  sudo ln -sf bpi-m2-zero_boot_2.scr boot.scr (first try) and reboot

do the same for bpi-m2-zero_boot_1.scr

I think the new u-boot (bootloader) it will find the correct device number.

```
 /boot$ ls -la
   lrwxrwxrwx  1 root root      20 Jan 15  2018 boot.scr -> bpi-m2-zero_boot.scr

```
[boot_device.scr.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/4875725/boot_device.scr.tar.gz)

mariusares, 2020-07-06:
lrwxrwxrwx 1 root   root        22 Jul  6 07:36 boot.scr -> bpi-m2-zero_boot_2.scr

U-Boot SPL 2020.04 (Apr 21 2020 - 14:52:45 +0000)
DRAM: 512 MiB
Trying to boot from MMC2
MMC Device 1 not found
spl: could not find mmc device 1. error: -19
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###

lrwxrwxrwx 1 root   root        22 Jul  6 07:53 boot.scr -> bpi-m2-zero_boot_1.scr

U-Boot SPL 2020.04 (Apr 21 2020 - 14:52:45 +0000)
DRAM: 512 MiB
Trying to boot from MMC2
MMC Device 1 not found
spl: could not find mmc device 1. error: -19
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###

mariusares, 2020-07-06:
look like the problem is on u-boot ...
is only looking after MMC1 but not for MMC0 1 or 2 ... basically eMMC became MMC1 when sd card is unplugged

avafinger, 2020-07-06:
Here is the new u-boot with support for eMMC.

[u-boot-sunxi-with-spl.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/4878543/u-boot-sunxi-with-spl.tar.gz)

mariusares, 2020-07-06:
U-Boot 2020.04-rc1-00200-ge1dff2d69e-dirty (Jul 06 2020 - 08:20:51 -0300) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Banana Pi BPI-M2-Zero
DRAM:  512 MiB
MMC:   mmc@1c0f000: 0, mmc@1c10000: 1
Loading Environment from FAT... Card did not respond to voltage select!
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
No working controllers found
Hit any key to stop autoboot:  0
Card did not respond to voltage select!
MMC: no card present
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.
starting USB...
No working controllers found
No ethernet found.
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
starting USB...
No working controllers found
No ethernet found.
No ethernet found.
=> mmc info
MMC: no card present
=>

same result with bpi-m2-zero_boot.scr bpi-m2-zero_boot_1.scr bpi-m2-zero_boot_2.scr

avafinger, 2020-07-06:
Please, explain in detail the steps you did to flash emmc.

Don't  forget to adjust rootfs=/dev/mmcblk?p2 (boot.cmd) and /etc/fstab (in rootfs) or you get stuck with u-boot prompt.

mariusares, 2020-07-06:
just got stuck with the u-boot as you can see
i have tried with /dev/mmcblk2p2 as wel but as you can see there is no mmc present on mmc list 
so that means that is a different pinout in the uboot ...
check this up
http://forum.banana-pi.org/t/problem-trying-to-use-emmc/11253/2

avafinger, 2020-07-06:
Please, try again with this one.
[u-boot-sunxi-with-spl.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/4878660/u-boot-sunxi-with-spl.tar.gz)

mariusares, 2020-07-06:
same issue 
U-Boot 2020.04-rc1-00200-ge1dff2d69e-dirty (Jul 06 2020 - 08:47:09 -0300) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Banana Pi BPI-M2-Zero
DRAM:  512 MiB
MMC:   mmc@1c0f000: 0, mmc@1c10000: 1
Loading Environment from FAT... Card did not respond to voltage select!
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
No working controllers found
Hit any key to stop autoboot:  0
Card did not respond to voltage select!
MMC: no card present
starting USB...
...

mariusares, 2020-07-06:
=> mmc info
MMC: no card present
=> mmc list
mmc@1c0f000: 0
mmc@1c10000: 1
=> mmc dev 1
Card did not respond to voltage select!
=> mmc dev 0
MMC: no card present
=> mmc dev 1
Card did not respond to voltage select!
=>

avafinger, 2020-07-06:
Could be.
Can you try with this old u-boot ?
[bootloader_m2z_v3.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/4878713/bootloader_m2z_v3.tar.gz)

mariusares, 2020-07-06:
i did it ... 
i`ll do it again ..

avafinger, 2020-07-06:
Did you build their u-boot (BPI) to see if you can boot? what is the output?

mariusares, 2020-07-06:
U-Boot SPL 2017.09 (Jan 18 2018 - 19:36:14)
DRAM: 512 MiB
Trying to boot from MMC2
MMC Device 1 not found
spl: could not find mmc device. error: -19
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###

i did it with their image ... so i mounted their /boot image and can`t get trough 
different settings, they have a folder bananapi in the /boot partition ....

avafinger, 2020-07-06:
Sorry, i made a mistake.  I forgot to really enable emmc, this is the last attempt.  Can't think of anything else to try.
[u-boot-sunxi-with-spl.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/4878750/u-boot-sunxi-with-spl.tar.gz)

mariusares, 2020-07-06:
U-Boot 2020.04-rc1-00200-ge1dff2d69e-dirty (Jul 06 2020 - 09:07:24 -0300) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Banana Pi BPI-M2-Zero
DRAM:  512 MiB
MMC:   Device 'mmc@1c11000': seq 1 is in use by 'mmc@1c10000'
mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from FAT... Unable to use mmc 1:1... In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
No working controllers found
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
28813 bytes read in 2 ms (13.7 MiB/s)
## Executing script at 43100000
Bad FIT image format
SCRIPT FAILED: continuing...
MMC: no card present
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.
starting USB...
No working controllers found
No ethernet found.
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
starting USB...
No working controllers found
No ethernet found.
No ethernet found.
=>
this is the output ...

mariusares, 2020-07-06:
=> mmc info
Device: mmc@1c11000
Manufacturer ID: 15
OEM: 100
Name: 8GTF4
Bus Speed: 52000000
Mode: MMC High Speed (52MHz)
Rd Block Len: 512
MMC version 5.1
High Capacity: Yes
Capacity: 7.3 GiB
Bus Width: 8-bit
Erase Group Size: 512 KiB
HC WP Group Size: 8 MiB
User Capacity: 7.3 GiB WRREL
Boot Capacity: 4 MiB ENH
RPMB Capacity: 512 KiB ENH
=>

mariusares, 2020-07-06:
and ....

[    2.480547]   No soundcards found.
[    2.493546] Freeing unused kernel memory: 1024K
[    2.532635] Run /init as init process
Kernel params: console=ttyS0,115200 earlyprintk rootfstype=ext4 root=/dev/mmcblk0p2 rw rootwait fsck.repair=yes panic=10 no_console_suspend consoleblank=0 drm_leak_fbdev_smem=1
Waiting for root system /dev/mmcblk0p2, countdown : 5
Waiting for root system /dev/mmcblk0p2, countdown : 4
Waiting for root system /dev/mmcblk0p2, countdown : 3
Waiting for root system /dev/mmcblk0p2, countdown : 2

Waiting for root system /dev/mmcblk0p2, countdown : 1

 let me setup the /etc/fstab :)

avafinger, 2020-07-06:
root=/dev/mmcblk0p2 => adjust to root=/dev/mmcblk**1**p2
and fstab

mariusares, 2020-07-06:
[    2.532005] Run /init as init process
Kernel params: console=ttyS0,115200 earlyprintk rootfstype=ext4 root=/dev/mmcblk1p2 rw rootwait fsck.repair=yes panic=10 no_console_suspend consoleblank=0 drm_leak_fbdev_smem=1
Waiting for root system /dev/mmcblk1p2, countdown : 5
Waiting for root system /dev/mmcblk1p2, countdown : 4
Waiting for root system /dev/mmcblk1p2, countdown : 3
Waiting for root system /dev/mmcblk1p2, countdown : 2
Waiting for root system /dev/mmcblk1p2, countdown : 1
Rootfs: /dev/mmcblk0p2
[   28.215766] /dev/mmcblk0p2: Can't open blockdev
[   28.223334] /dev/mmcblk0p2: Can't open blockdev
mount: mounting /dev/mmcblk0p2 on /mnt/root failed: No such file or directory
umount: can't unmount /mnt/root: Invalid argument
Dropping to a shell.

/ # [   32.481996] vcc3v0: disabling
[   32.485067] vcc5v0: disabling
[   32.488083] cam-avdd: disabling
[   32.491268] cam-dvdd: disabling

/ #

avafinger, 2020-07-06:
So must be: Waiting for root system /dev/mmcblk**2**p2

Try to check in the shell.

`ls  /dev/mmcbl*`

mariusares, 2020-07-06:
/ # ls /dev/mmc*
/dev/mmcblk2       /dev/mmcblk2boot1  /dev/mmcblk2p2
/dev/mmcblk2boot0  /dev/mmcblk2p1
/ # [   64.482738] cfg80211: failed to load regulatory.db

avafinger, 2020-07-06:
we now know ..
**root= /dev/mmcblk2p2**
**fstab** must reflect the same.

for the record:
which boot.scr is the correct one?

mariusares, 2020-07-06:
[    2.481578] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    2.491546] platform regulatory.0: Falling back to sysfs fallback for: regulatory.db
[    2.492587]   No soundcards found.
[    2.505182] Freeing unused kernel memory: 1024K
[    2.572998] Run /init as init process
Kernel params: console=ttyS0,115200 earlyprintk rootfstype=ext4 root=/dev/mmcblk2p2 rw rootwait fsck.repair=yes panic=10 no_console_suspend consoleblank=0 drm_leak_fbdev_smem=1
Waiting for root system /dev/mmcblk2p2, countdown : 5
Rootfs: /dev/mmcblk2p2
[    2.735318] EXT4-fs (mmcblk2p2): bad geometry: block count 1909760 exceeds size of device (1876992 blocks)
mount: mounting /dev/mmcblk2p2 on /mnt/root failed: Invalid argument
umount: can't unmount /mnt/root: Invalid argument
Waiting for root system /dev/mmcblk2p2, countdown : 4
Rootfs: /dev/mmcblk2p2
[    7.891324] EXT4-fs (mmcblk2p2): bad geometry: block count 1909760 exceeds size of device (1876992 blocks)
mount: mounting /dev/mmcblk2p2 on /mnt/root failed: Invalid argument
umount: can't unmount /mnt/root: Invalid argument
Waiting for root system /dev/mmcblk2p2, countdown : 3
Rootfs: /dev/mmcblk2p2
[   13.028491] EXT4-fs (mmcblk2p2): bad geometry: block count 1909760 exceeds size of device (1876992 blocks)
mount: mounting /dev/mmcblk2p2 on /mnt/root failed: Invalid argument
umount: can't unmount /mnt/root: Invalid argument

[24 more replies]