avafinger/bananapi-zero-ubuntu-base-minimal #47: Sources for Kernel 5.7.6

BilyakA, 2020-08-31:
I am trying to add lima to release v3.7 (Kernel 5.7.6) to make both OV5640 and lima work.

Just wondering: what kernel sources have you used to build that image? I can see only [5.6.y](https://github.com/avafinger/linux-5.6.y) in your repositories that is currently at version 5.6.0-rc7.

Was that vanilla kernel with [this](https://github.com/avafinger/linux-5.6.y/blob/master/arch/arm/configs/m2z_defconfig) config?

avafinger, 2020-08-31:
Use this setting and the DTB. If I remember correctly, I had to change the kernel configuration (that configuration) to make the CPU Temp work in 5.7. in the OV5640, the problem was the PD14, which should be enabled first, maybe it could be done on u-boot . I have no experience with **lima**, but if you follow the route to build the kernel, I suggest you go with 5.9-rc3, most of the patches are already there, you only have to deal with the config and transplant the OV5640 node. i've lost some of the work with a "rm -rf *"  , so i will move on to 5.9 when time permits. 5.9 seems to become LTS.

BilyakA, 2020-08-31:
@avafinger well sorry to hear about rm -rf *

Ok thanks for the pointers.

I've seen you have modified some dts ([this](https://github.com/avafinger/linux-5.6.y/blob/master/arch/arm/boot/dts/sun8i-h3-x.dtsi) and [this](https://github.com/avafinger/linux-5.6.y/blob/master/arch/arm/boot/dts/sunxi-h3-h5-x.dtsi)) to make the [main](https://github.com/avafinger/linux-5.6.y/blob/master/arch/arm/boot/dts/bpi-m2-zero-v4.dts) one. Seems that major differences are only about CPU temp and frequencies. And CSI of course.

Right now I am trying to compile vanilla 5.9-rc3 with your dts (that seems to be more complete) and your settings.