avafinger/bananapi-zero-ubuntu-base-minimal #48: Can't make GBM work

BilyakA, 2020-09-03:
Fresh install [v2.4 image](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.4)
`apt-get update && apt-get dist-upgrade`
Install [v3.7 kernel](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.7)
`apt-get install kmscube git`
`git clone https://github.com/bootlin/mali-blobs.git`
`cd mali-blobs/tree/master/r6p2/arm/wayland`
`pwd > /etc/ld.so.conf.d/00-mali.conf`
`ldconfig`
`ldd /usr/bin/kmscube` to ensure that ligEGL.so.1 => /home/ubuntu/mali-blobs/r6p2/arm/wayland/libgbm.so.1
And after that kmscube tells me:
`kmscube: symbol lookup error: kmscube: undefuned symbol: gbm_bo_unmap`

Without mali-blobs kmscube output follows:
`MESA-LOADER: failed to open sun4i-drm (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: sun4i-drm
MESA-LOADER: failed to open kms_swrast (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: kms_swrast
MESA-LOADER: failed to open swrast (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: swrast
Segmentation fault`

What am I doing wring and how to make any 3d (mali/lima) work?

P.S. I suspect that `undefuned symbol: gbm_bo_unmap` is some problem with mali o kmscube being outdated and MAYBE it is already working but hot can I check that besides kmscube?

avafinger, 2020-09-03:
i think kmscube is originally written for opengles3, you must recompile it and remove the reference to those functions.

Try this kmscube from v1.9:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases?after=v1.9

avafinger, 2020-09-03:
Oops, https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v14

BilyakA, 2020-09-04:
Ok, I have downloaded [kmscube from v14](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/download/v14/kmscube). It required libpng12.so.0 so I have manually compiled and installed it (as described [here](https://askubuntu.com/questions/895897/error-while-loading-shared-libraries-libpng12-so-0)).
Now kmscube gives me this output:
```
failed to initialize
failed to initialize EGL
```

avafinger, 2020-09-04:
There was no need to compile libpng12, was just to install it.
Anyway, you are almost there, do the following:

1. From your  /home/ubuntu/mali-blobs/r6p2/arm/wayland/
2. copy the contents:

        sudo cp -avf * /usr/lib/arm-linux-gnueabihf

ShoesForClues, 2021-06-28:
I'm having a similar issue, except with SDL2. I compiled SDL2 targeting the KMSDRM video driver backend and after installing the mali-blobs like you said, I tried to run one of the SDL2 test programs and I'm getting this error:
```
ubuntu@bpi-m2z:~/sdl_test$ ./testyuv
INFO: 100 iterations in 617ms, 6.17ms each
ERROR: Couldn't create window: kmsdrm not available
```
I'm running this on Ubuntu 20.04 LTS with 5.7.10 kernel version.

avafinger, 2021-06-28:
kmsdrm is available for **root** user (default image) unless you add the rules to run with **ubuntu** user

`sudo ./testyuv`

Don't forget you should compile SDL2 without X11.
```

ubuntu@bpi-m2z:~$ sudo ls -la /dev/dri/by-path/
total 0
drwxr-xr-x 2 root root 60 Feb 23 00:19 .
drwxr-xr-x 3 root root 80 Feb 23 00:19 ..
lrwxrwxrwx 1 root root  8 Feb 23 00:19 platform-display-engine-card -> ../card0
```

ShoesForClues, 2021-06-29:
Still getting the same error.
```
ubuntu@bpi-m2z:~/sdl_test$ sudo LD_LIBRARY_PATH=./:$LD_LIBRARY_PATH ./testyuv
INFO: 100 iterations in 765 ms, 7.65ms each
ERROR: Couldn't create window: No available video device
```
Here's how I configured it:
```
root@DevBox:/SDL# ./configure --disable-video-opengl --enable-video-kmsdrm --disable-video-vulkan --disable-video-x11
```
And here's the config summary :
```
SDL2 Configure Summary:
Building Shared Libraries
Building Static Libraries
Enabled modules : atomic audio video render events joystick haptic sensor power filesystem threads timers file loadso cpuinfo assembly
Assembly Math   :
Audio drivers   : disk dummy oss alsa(dynamic)
Video drivers   : dummy kmsdrm(dynamic) opengl_es1 opengl_es2
Input drivers   : linuxev linuxkd
Enable virtual joystick APIs : YES
Using libsamplerate : NO
Using libudev       : YES
Using dbus          : NO
Using ime           : YES
Using ibus          : NO
Using fcitx         : NO
```
Maybe the issue could be the fact that I compiled this in a separate chroot environment? I've often had success when working with other platforms.

avafinger, 2021-06-29:
> Maybe the issue could be the fact that I compiled this in a separate chroot environment?

I don't think so.
Check if you can run **kmscube** or one of these: https://github.com/avafinger/mali-fbdev-stress-test-tools

If you can't, you still have problems with your mail setup, so no OpenGles available.

avafinger, 2021-06-29:
opps, forget about the https://github.com/avafinger/mali-fbdev-stress-test-tools

ShoesForClues, 2021-06-29:
Hm, the dependency libpng12 has been removed from Ubuntu's repository for some time now. You also mentioned there is no need to recompile it, assuming there's still a way to install it? I've searched online but haven't had much luck with it.

avafinger, 2021-06-29:
If you are using this tag: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.10
there is no mail gpu in the image, it is a server-like image.

If you want to use 20.04 LTS with mali, try the previous image, it is 19.10 but you can upgrade to 20.04 LTS with this instruction: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/54

Maybe I will get back to the board and rebuild mali support with the latest kernel, or if you are in a hurry, kernel 5.13 has upstream support for M2 zero with lima. (no mail blobs).