avafinger/bananapi-zero-ubuntu-base-minimal #49: runtime error: not running on a Rpi

djk300, 2020-09-05:
I  am continuing to get the Runtime error: Not running on a RPi. I am using a Raspberry pi 4B with Ubuntu Mate 20.04. I was using the same simple code on my old 3B+ and it worked fine. I have tried adding the username to the gpio group and it did not fix the issue. I have tried gpiozero and get the same error. Not sure what to do now as I read a lot of other people with the same issue but have not yet found a solution that works for me. Thanks.

code is below:
import RPi.GPIO as GPIO
from hx711 import HX711

#set the pin numbers for the weight sensor on raspberry pi
hx = HX711(5, 6)

hx.set_reading_format("MSB", "MSB")

#set calibration factor aka reference unit
hx.set_reference_unit(105)
hx.reset()
hx.tare()
abc=input("scale is tared, press enter to start ->")
#print("Tare done! Add weight now...")

#begin weight count and export to google sheet
#cellcounter = 2
endloop = 10 #number of times to run the code
for k in range(endloop):
total = 0
endrange = 10 #number of readings to average
for i in range(endrange):
val = hx.get_weight(5)
total = val + total
#set time delay between readings
hx.power_down()
hx.power_up()
time.sleep(0.1)

here is the error:

runfile('/home/4di/.config/spyder-py3/weightgpio.py', wdir='/home/4di/.config/spyder-py3')
/usr/lib/python3/dist-packages/gpiozero/pins/pi.py:129: SPISoftwareFallback: failed to initialize hardware SPI, falling back to software (error was: failed to import spidev)
warnings.warn(
Traceback (most recent call last):

File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 69, in call
self = cls._instances[key]

KeyError: (8,)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 71, in pin
pin = self.pins[n]

KeyError: 8

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

File "", line 1, in
runfile('/home/4di/.config/spyder-py3/weightgpio.py', wdir='/home/4di/.config/spyder-py3')

File "/usr/lib/python3/dist-packages/spyder_kernels/customize/spydercustomize.py", line 827, in runfile
execfile(filename, namespace)

File "/usr/lib/python3/dist-packages/spyder_kernels/customize/spydercustomize.py", line 110, in execfile
exec(compile(f.read(), filename, 'exec'), namespace)

File "/home/4di/.config/spyder-py3/weightgpio.py", line 4, in
spi = HX711()

File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 95, in call
self = super(GPIOMeta, cls).call(*args, **kwargs)

File "/home/4di/.local/lib/python3.8/site-packages/hx711_gpiozero/init.py", line 19, in init
super().init(bits, max_voltage, **spi_args)

File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 128, in init
super(AnalogInputDevice, self).init(shared=True, **spi_args)

File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 34, in init
self._spi = self.pin_factory.spi(**spi_args)

File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 142, in spi
return self.spi_classes[('software', shared)](self, **spi_args)

File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 72, in call
self = super(GPIOMeta, cls).call(*args, **kwargs)

File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 152, in init
super(LocalPiSoftwareSPI, self).init(select_pin, active_high=False)

File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 47, in init
super(OutputDevice, self).init(pin, pin_factory=pin_factory)

File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 69, in init
super(SourceMixin, self).init(*args, **kwargs)

File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 375, in init
pin = self.pin_factory.pin(pin)

File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 73, in pin
pin = self.pin_class(self, n)

File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 101, in init
GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])

avafinger, 2020-09-05:
_failed to initialize hardware SPI,_

You have to enable spi (0 or 1) or spidev0 or spidev1 for your need. It is "disabled" by default in DTB.

Steps:

1.  install dtc

        sudo apt-get install device-tree-compiler

2. Decompile the currently **DTB** file in use, find it with:

        cd /boot
        ls -la bpi-m2-zero.dtb
        
3. If the file is (for example, you did not specify): bpi-m2-zero-v4.dtb_5.7.10, decompile it

        sudo dtc -O dts -o m2z.dts bpi-m2-zero-v4.dtb

4 . Edit the fileuse jed or any other text editor) , change the **spi** you need (and the one that can work on M2Z)

		spi@1c68000 {
			compatible = "allwinner,sun8i-h3-spi";
			reg = <0x1c68000 0x1000>;
			interrupts = <0x0 0x41 0x4>;
			clocks = <0x3 0x1e 0x3 0x52>;
			clock-names = "ahb", "mod";
			dmas = <0x15 0x17 0x15 0x17>;
			dma-names = "rx", "tx";
			pinctrl-names = "default";
			pinctrl-0 = <0x16 0x17>;
			resets = <0x3 0xf>;
			status = "disabled"; <==== change it do "okay"
			#address-cells = <0x1>;
			#size-cells = <0x0>;
			cs-gpios = <0xc 0x2 0x3 0x0 0xc 0x0 0x6 0x0>;

			spi@0 {
				compatible = "nanopi,spidev"; <==== change it to "ge,achc"
				reg = <0x0>;
				status = "okay";
				spi-max-frequency = <0x989680>;
			};

		};

		spi@1c69000 {
			compatible = "allwinner,sun8i-h3-spi";
			reg = <0x1c69000 0x1000>;
			interrupts = <0x0 0x42 0x4>;
			clocks = <0x3 0x1f 0x3 0x53>;
			clock-names = "ahb", "mod";
			dmas = <0x15 0x18 0x15 0x18>;
			dma-names = "rx", "tx";
			pinctrl-names = "default";
			pinctrl-0 = <0x18>;
			resets = <0x3 0x10>;
			status = "disabled"; <==== change it do "okay"
			#address-cells = <0x1>;
			#size-cells = <0x0>;

			spi@0 {
				compatible = "nanopi,spidev"; <==== change it to "ge,achc"
				reg = <0x0>;
				spi-max-frequency = <0x989680>;
			};
		};
       

5. after you enable the correct spi (spidev) you recompile it and make it default

        sudo dtc -I dts -O dtb -o m2z.dtb m2z.dts
        sudo ln -sf m2z.dtb bpi-m2-zero-v4.dtb
        sudo reboot

djk300, 2020-09-05:
thank you, but during step 2 i get the following:
/boot$ ls -la bpi-m2-zero.dtb
ls: cannot access 'bpi-m2-zero.dtb': No such file or directory

these are what is in the boot dir:
config-5.4.0-1015-raspi  initrd.img-5.4.0-1015-raspi
config-5.4.0-1016-raspi  initrd.img-5.4.0-1016-raspi
dtb			 initrd.img.old
dtb-5.4.0-1015-raspi	 System.map-5.4.0-1015-raspi
dtb-5.4.0-1016-raspi	 System.map-5.4.0-1016-raspi
dtbs			 vmlinuz
firmware		 vmlinuz-5.4.0-1015-raspi
grub			 vmlinuz-5.4.0-1016-raspi
initrd.img		 vmlinuz.old

for step 3, which file needs to be specified?

avafinger, 2020-09-06:
It should be:

**ls -la bpi-m2-zero-v4.dtb**

But, but, but wait a minute, i realized you are running **raspi** on **m2z**, is that correct?
Unfortunately this question must be directed to the **raspi** creators. :)

Sorry, i can't help you with raspi.