avafinger/bananapi-zero-ubuntu-base-minimal #52: How to update - install kernel - install uboot

nordurljosahvida, 2020-10-09:
Hi, thanks for the work

I'm on: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.8, that's the latest img i've found.

How do i update kernel [all of them say `not a debian archive format` when dpkg installing?
How do I update uboot on a booted system?

Thanks

avafinger, 2020-10-09:
Have a look at https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases

nordurljosahvida, 2020-10-10:
Thank you very much, i did find instructions for uboot in [v3.4](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.4), not sure how i was using wget incorrectly and downloads were not the actual dpkg file.

A few quick questions:
- i've installed uboot from 3.4
- i've installed kernel from 3.7

so am I now on both latest? i have not passed by any in-between releases, so haven't followed instructions on 3.6 to increase thermal throttling to 80deg, but i understand correctly now all future releases will have it set at that?

is there a way to display boot operations rather than the simple blinking underscore?

thanks

nordurljosahvida, 2020-11-08:
Hi again @avafinger , can I ask you for confirmation about questions above if you have a minute, and if it's advisable to dist-upgrade to 20.04?

avafinger, 2020-11-08:
I will try to look into it.
But why you want to display info while booting? Display info takes time and the boot is slower.

nordurljosahvida, 2020-11-08:
Thank you so much. The display info while booting is solved, I had to go through some really insane stuff to get the pi to stop using dhcp when assigning it a static address, and now it works. So the only remaining questions are if I'm on the latest everything having installed uboot from 3.4 and kernel from 3.7, and if usb camera support is planned ;]

Thanks again!

closed 2021-01-03 by nordurljosahvida