avafinger/bananapi-zero-ubuntu-base-minimal #54: Ubuntu 19.10 Eoan is EOL. Upgrade to 20.04 LTS ok?

nordurljosahvida, 2020-12-25:
And will upgrading interfere with the kernel or other custom flashed files from your release page in any way?

Also, while we're at it, a quick check, what are the latest files for every category [kernel, etc] I should flash to ensure I'm up to date?

Thanks as always!

avafinger, 2020-12-25:
> And will upgrading interfere with the kernel or other custom flashed files from your release page in any way?

If you mean upgrading from 19.10 to 20.04, no problem at all. Upgrading to 20.04 (by command line) may re-create some (or new) systemd services that are now disabled and not needed, just disabled those unneeded services again and you are good.

> Also, while we're at it, a quick check, what are the latest files for every category [kernel, etc] I should flash to ensure I'm up to date?

Get a new SD card and flash this one: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.8 and updated to the latest kernel:
https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases

You can also check for the lastest u-boot.

i will build a new Image with Kernel 5.10 soon, but it can take some days...

nordurljosahvida, 2021-01-01:
ok great thanks! is there a list of services you've disabled?

i have a highly refined image i've been curating for months now, is there a way to make sure it is running the latest kernel and uboot on its already installed system?

Thanks

nordurljosahvida, 2021-01-01:
Also, completely separate topic but I want to make sure this doesn't become my personal support thread, let me know if this could be of interest for the repo please:

I have netplan set up with network manager:

```
network:
  version: 2
  renderer: NetworkManager
  wifis:
    wlan0:
      dhcp4: no
      dhcp6: no
      addresses: [10.42.11.1/16]
      gateway4: 10.42.0.1
      nameservers:
        addresses: [10.42.0.1]
      access-points:
        "ssid":
          password: "wpa2pass"
```

and wifi power management off:

```
cat etc/NetworkManager/conf.d/default-wifi-powersave-on.conf 
[connection]
wifi.powersave = 2
```

I still keep getting this error that shuts down my wifi every few days:

```
Jan  1 18:15:06 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521306.3585] manager: rfkill: Wi-Fi hardware radio set disabled
Jan  1 18:15:06 Tv-Ampere-PRSA-MK3S-01 wpa_supplicant[342]: rfkill: WLAN soft blocked
Jan  1 18:15:06 Tv-Ampere-PRSA-MK3S-01 wpa_supplicant[342]: rfkill: WLAN soft blocked
Jan  1 18:15:06 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521306.5032] manager: rfkill: Wi-Fi now disabled by radio killswitch
Jan  1 18:15:10 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521310.8709] manager: rfkill: Wi-Fi hardware radio set enabled
Jan  1 18:15:10 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521310.8973] manager: rfkill: Wi-Fi now enabled by radio killswitch
Jan  1 18:15:15 Tv-Ampere-PRSA-MK3S-01 systemd[1]: systemd-rfkill.service: Succeeded.
Jan  1 18:18:06 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521486.1333] manager: rfkill: Wi-Fi hardware radio set disabled
Jan  1 18:18:06 Tv-Ampere-PRSA-MK3S-01 wpa_supplicant[342]: rfkill: WLAN soft blocked
Jan  1 18:18:06 Tv-Ampere-PRSA-MK3S-01 wpa_supplicant[342]: rfkill: WLAN soft blocked
Jan  1 18:18:06 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521486.2564] manager: rfkill: Wi-Fi now disabled by radio killswitch
Jan  1 18:18:10 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521490.6041] manager: rfkill: Wi-Fi hardware radio set enabled
Jan  1 18:18:10 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521490.6304] manager: rfkill: Wi-Fi now enabled by radio killswitch
Jan  1 18:18:15 Tv-Ampere-PRSA-MK3S-01 systemd[1]: systemd-rfkill.service: Succeeded.
Jan  1 18:21:06 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521666.6582] manager: rfkill: Wi-Fi hardware radio set disabled
Jan  1 18:21:06 Tv-Ampere-PRSA-MK3S-01 wpa_supplicant[342]: rfkill: WLAN soft blocked
Jan  1 18:21:06 Tv-Ampere-PRSA-MK3S-01 wpa_supplicant[342]: rfkill: WLAN soft blocked
Jan  1 18:21:06 Tv-Ampere-PRSA-MK3S-01 NetworkManager[339]: <info>  [1609521666.7706] manager: rfkill: Wi-Fi now disabled by radio killswitch
```

Thanks again

nordurljosahvida, 2021-01-01:
Last thing, related to my first reply of the last two, the repos for eoan on `ubuntu-ports` are no longer available, therefore `do-release-upgrade` no longer works. Is there any way around this? I'm searching for old mirrors now but haven't found any yet.

avafinger, 2021-01-03:
I will look into it.

nordurljosahvida, 2021-01-03:
Thank you so much, I've looked into [this thread's script](https://askubuntu.com/questions/428698/are-there-alternative-repositories-to-ports-ubuntu-com-for-arm) but have so far found no mirrors with eoan armhf ports.

avafinger, 2021-01-06:
You can try this:

```
sudo sed -i 's/eoan/focal/g' /etc/apt/sources.list
sudo apt-get update
sudo apt-get dist-upgrade
```

Make a backup of the sd card, just in case....

nordurljosahvida, 2021-01-08:
That worked thanks! Would I still then install the same kernel and uboot as 19?

avafinger, 2021-01-08:
Yes.

nordurljosahvida, 2021-01-08:
great thanks

nordurljosahvida, 2021-01-08:
last question to close the thread: is that wifi issue relevant for a new issue?

> Also, completely separate topic but I want to make sure this doesn't become my personal support thread, let me know if this could be of interest for the repo please:
>
> I have netplan set up with [...]

avafinger, 2021-01-08:
I still use the deprecated/old style /etc/network/interfaces and there you have something like this:

```
# Disable power saving on compatible chipsets (prevents SSH/connection dropouts over WiFi)
# wireless-mode Managed
wireless-power off
```

And in some circumstances, **wireless-power off** does not work. I have to use:
` iwconfig wlan0 power off` after boot.

I don't have the problem you reported, maybe because I turn off the board more often.
Maybe **wifi.powersave = 2** is not permanently set  with NetworkManager.

avafinger, 2021-01-08:
What i mean is a NetworkManager issue and not an HW issue... ;)

nordurljosahvida, 2021-01-08:
ok thanks, I already run that after boot, and was really trying to embrace the new standard, even though there is no documentation about it anywhere.