avafinger/bananapi-zero-ubuntu-base-minimal #55: UART baudrate accuracy on ttyS3

PLMR-FR, 2021-03-12:
Hi,

I am struggling with the baudrate accuracy. I need to communicate with another device at a baudrate of 921600 bauds. But it is impossible.

Normaly at 921600 bauds, the duration of a bit should be arround 1.085 us. But in my case it is around 1.44 us.

[image]

Here is the output of stty < /dev/ttyS3

speed 921600 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo

The max speed that I can reach without too many error is 115200 bauds on the BPI zero M2. Are there some issues with the UART baudrate accuracy? Or maybe have I a crappy board with the bad crystal ?

Thanks for you help

avafinger, 2021-03-12:
I have never used UART with more than 115200 bauds.
Can you test with /dev/ttyS2 ? (Extended UART 2). I can't remember If UART2 is activated, if not you need to edit the DT.
Perhaps the temperature of the Board is too High?

PLMR-FR, 2021-03-12:
I found the problem. 

Actually it is written in the H2+ datasheet : 
[image]

And the serial clock frequency is at 24 MHz.  The divisor is on 16 bit.

If we take the value of the divisor from 1 to 10 we have the following baud rates : 

Divisor | BaudRate
-- | --
1 | 1500000
2 | 750000
3 | 500000
4 | 375000
5 | 300000
6 | 250000
7 | 214285,714
8 | 187500
9 | 166666,667
10 | 150000

So it is not possible to have something close to 921600 bauds.

I decrease the baudrate of my device at 500000 bauds and now the BPI M2 zero send and receive without problem.

closed 2021-03-12 by PLMR-FR