avafinger/bananapi-zero-ubuntu-base-minimal #56: Problem with OV5640 DVP camera

thiagobucca, 2021-03-15:
Hi, i'm having some issues on getting my OV5640 DVP camera to work with this image, whenever i start mjpeg-streamer, i get this:

`MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: BA81
    ... will try to handle this by checking against supported formats. 
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed`

My topology is:

`Media controller API version 5.7.7

Media device information
------------------------
driver          sun6i-csi
model           Allwinner Video Capture Device
serial          
bus info        
hw revision     0x0
driver version  5.7.7

Device topology
- entity 1: sun6i-csi (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video1
	pad0: Sink
		<- "ov5640 3-003c":0 [ENABLED,IMMUTABLE]

- entity 5: ov5640 3-003c (1 pad, 1 link)
            type V4L2 subdev subtype Sensor flags 0
            device node name /dev/v4l-subdev0
	pad0: Source
		-> "sun6i-csi":0 [ENABLED,IMMUTABLE]
`

did u used a different mjpeg-streamer version or any custom config to make this work like you show in this repo readme?

thanks in advance

avafinger, 2021-03-16:
Can you give the complete output?
Looks like it is asking for MJPEG and the sensor is only JPEG.

You should have in your code:

```
        case V4L2_PIX_FMT_MJPEG:
            // Fall-through intentional
        case V4L2_PIX_FMT_JPEG:
            fmtString = "JPEG";
            break;

```

thiagobucca, 2021-03-16:
Yes i checked it and i have this line under input_uvc.c. 

Here's the complete output of what i've tried so far:

`pi@bpizero:~/mjpg-streamer$ ll /dev/video*
crw-rw---- 1 root video 81, 0 Apr  1  2020 /dev/video0
crw-rw---- 1 root video 81, 1 Apr  1  2020 /dev/video1
pi@bpizero:~/mjpg-streamer$ `

`pi@bpizero:~/mjpg-streamer$ ll /dev/media*
crw-rw-rw- 1 root video 243, 0 Apr  1  2020 /dev/media0
crw-rw-rw- 1 root video 243, 1 Apr  1  2020 /dev/media1`

`pi@bpizero:~/mjpg-streamer$ media-ctl --device /dev/media1 --print-topology
Media controller API version 5.7.6

Media device information
------------------------
driver          sun6i-csi
model           Allwinner Video Capture Device
serial          
bus info        
hw revision     0x0
driver version  5.7.6

Device topology
- entity 1: sun6i-csi (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video1
	pad0: Sink
		<- "ov5640 3-003c":0 [ENABLED,IMMUTABLE]

- entity 5: ov5640 3-003c (1 pad, 1 link)
            type V4L2 subdev subtype Sensor flags 0
            device node name /dev/v4l-subdev0
	pad0: Source
		-> "sun6i-csi":0 [ENABLED,IMMUTABLE]

`

`pi@bpizero:~/mjpg-streamer$ media-ctl -v --device /dev/media1 --known-mbus-fmts
RGB444_1X12                     0x1016
RGB444_2X8_PADHI_BE             0x1001
RGB444_2X8_PADHI_LE             0x1002
RGB555_2X8_PADHI_BE             0x1003
RGB555_2X8_PADHI_LE             0x1004
RGB565_1X16                     0x1017
BGR565_2X8_BE                   0x1005
BGR565_2X8_LE                   0x1006
RGB565_2X8_BE                   0x1007
RGB565_2X8_LE                   0x1008
RGB666_1X18                     0x1009
RBG888_1X24                     0x100e
RGB666_1X24_CPADHI              0x1015
RGB666_1X7X3_SPWG               0x1010
BGR888_1X24                     0x1013
BGR888_3X8                      0x101b
GBR888_1X24                     0x1014
RGB888_1X24                     0x100a
RGB888_2X12_BE                  0x100b
RGB888_2X12_LE                  0x100c
RGB888_1X7X4_SPWG               0x1011
RGB888_1X7X4_JEIDA              0x1012
ARGB8888_1X32                   0x100d
RGB888_1X32_PADHI               0x100f
RGB101010_1X30                  0x1018
RGB121212_1X36                  0x1019
RGB161616_1X48                  0x101a
Y8_1X8                          0x2001
UV8_1X8                         0x2015
[... 77 lines cut ...]

and then:

`pi@bpizero:~/mjpg-streamer$ sudo media-ctl --device /dev/media1 --set-v4l2 '"ov5640 3-003c":0[fmt:JPEG_1X8/1280x720]'`

and finally:

`pi@bpizero:~/mjpg-streamer/mjpg-streamer-experimental$ sudo ./mjpg_streamer -i "./input_uvc.so -d /dev/video1 -r 1280x720 -q 90 -n" -o "./output_http.so -p 8080 -w ./www"
MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: BA81
    ... will try to handle this by checking against supported formats. 
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed`

Kernel information: 

`pi@bpizero:~/mjpg-streamer/mjpg-streamer-experimental$ uname -a
Linux bpizero 5.7.6 #1 SMP Sun Jun 28 00:11:55 -03 2020 armv7l armv7l armv7l GNU/Linux`

is there anything else that i should look?

avafinger, 2021-03-16:
The sensor is fine.

Try changing line: https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c#L141

to

`int width = 640, height = 480, fps = -1, format = V4L2_PIX_FMT_JPEG, i;`

thiagobucca, 2021-03-16:
nice, now is working flawlessly

thank you!

closed 2021-03-16 by thiagobucca