avafinger/bananapi-zero-ubuntu-base-minimal #57: Enable Power-off pin

PLMR-FR, 2021-03-16:
Hi,

Is there anyway to enable a power off pin ?

My BPI M2 Zero is on a carrierboard and it would be nice to be able to configure a pin which go to low level when the BPI is shuted down. So the power supply can be switched off also.

Is there any way to activate this feature ?

Thanks 

PLMR

avafinger, 2021-03-16:
> Is there anyway to enable a power off pin ?

AFAIK BPI M2 zero (and possibly all other H2+/H3 like OPI,NANOPI,etc...) never turn off the power. I think the AR100 embedded controller is responsible for this. I think the newer u-boot has some support for AR100, not really sure.

But if you want a pin to go low when the kernel is in power off, yes it is possible but you have to add this to the kernel.
You can use this kernel here: https://github.com/avafinger/linux-5.6.y
or a newer one, support for H2+ and/or H3 is almost complete.

PLMR-FR, 2021-03-16:
Hi, thanks for your reply.

My carrier board will take care of switching off the power for the BPI M2 Zero, So it's not a problem if the M2 never turn off the power.

I am sorry I am not experienced with kernel modification for now I have the 5.3.4 kernel. Do I have to decompile the .dtb file, modify it and recompile it ?
My goal is to have the pin 37 (PA17) which goes low when BPI is shutted down. How can I add this feature in the kernel ? Is there a guide somewhere ?

Last thing, Is it a must to use a 5.6.y kernel for this ? Or can I modify my 5.3.4 kernel ?

Thanks for your help

avafinger, 2021-03-16:
> I am sorry I am not experienced with kernel modification for now I have the 5.3.4 kernel. Do I have to decompile the .dtb file, modify it and recompile it ?
> My goal is to have the pin 37 (PA17) [...]

No guide, only kernel docs or some kernel book.
There are many ways to do it, I don't know which way would work best. Maybe the best place to turn the pin low would be in u-boot if the kernel returns control to u-boot after Power down. I don't know anyone who has done it.

In kernel you should look for:

```
/**
 *	kernel_power_off - power_off the system
 *
 *	Shutdown everything and perform a clean system power_off.
 */
void kernel_power_off(void)
```
You also may post this question on linux-sunxi@googlegroups.com (there are kernel developers there).

PLMR-FR, 2021-03-18:
Hi,

Ok I think I got the thing. But Before to power off the system I need to shut down. And I just discoverd that power button doesn't shutdown the board with the kernel 5.3.4.

In the device tree there is this part linked to the push button PWR:

```
	r_gpio_keys {
		compatible = "gpio-keys";
		pinctrl-names = "default";
		pinctrl-0 = <0x3c>;

		k1 {
			label = "shutdown";
			linux,code = <0x74>;
			gpios = <0xe 0x0 0x3 0x1>;			
		};
	};
```
It is written compatible ="gpio-keys"
This is corresponding to the driver gpio_keys.c
But I think this driver has not been compiled in the kernel

Here is the output of : `ls /lib/modules/5.3.4/kernel/input/keyboard`
-->`sun4i-lradc-keys.ko`

Does it mean that I need to recompile the kernel by integrating the gpio-keys driver ?

Thanks

avafinger, 2021-03-18:
> Does it mean that I need to recompile the kernel by integrating the gpio-keys driver ?

Nope. It's built in:
CONFIG_KEYBOARD_GPIO=y

Search:
`dmesg|grep input`

I just can't remember if the **gpios** is the correct one for the BPI-M2Z. You need to lookup up the schematic.

PLMR-FR, 2021-03-18:
Hi,

I checked the schematic and it looks correct : 
[image]

Power key is wired on PL3

and the device tree looks ok also : 

```
	pinctrl@1f02c00 {
		compatible = "allwinner,sun8i-h3-r-pinctrl";
		reg = <0x1f02c00 0x400>;
		interrupts = <0x0 0x2d 0x4>;
		clocks = <0x2f 0x3 0xf 0x10>;
		clock-names = "apb", "hosc", "losc";
		gpio-controller;
		#gpio-cells = <0x3>;
		interrupt-controller;
		#interrupt-cells = <0x3>;
		phandle = <0xe>;
		
		...
		
		key_pins@0 {
			pins = "PL3";
			function = "gpio_in";
			phandle = <0x3c>;
		};
		
		...
		
	};
	
	...
	
	r_gpio_keys {
		compatible = "gpio-keys";
		pinctrl-names = "default";
		pinctrl-0 = <0x3c>;

		k1 {
			label = "shutdown";
			linux,code = <0x74>;
			gpios = <0xe 0x0 0x3 0x1>;			
		};
	};
```
What Can I miss ?

Thanks,

PLMR

avafinger, 2021-03-18:
`dmesg|grep input`

PLMR-FR, 2021-03-19:
Hi,

Here is the output of `dmesg | grep input `

`[    6.713937] input: sunxi-ir as /devices/platform/soc/1f02000.ir/rc/rc0/input0`

Any thoughts ?

PLMR

PLMR-FR, 2021-03-19:
Hi again,

I checked in your source code of linux-5.6-y kernel.

In the file \arch\arm\configs\m2z_defconfig :
Line 2226: # CONFIG_KEYBOARD_GPIO is not set

So I think that's why it doesn't work. It should be the same for 5.3.4.

I will try to compile a new kernel with this flag activated. What do you use for cross compilation ? Is it possible to compile it on BPI M2Z ?

avafinger, 2021-03-19:
Interesting, it should have been set.

> I will try to compile a new kernel with this flag activated. What do you use for cross compilation ?

arm-linux-gnueabihf

> Is it possible to compile it on BPI M2Z ?

Yes. You can follow these instructions here: https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal#mainline-kernel-510
Don't forget to install the necessary dependencies to build a kernel.
zImage is the kernel and study the /boot partitions.

PLMR-FR, 2021-03-20:
Hi,

Where can I find the source code of your kernel 5.3.4 ? Because when I download the source code of your release 5.3.4 it's not the source code of the kernel

[image]

And your repository of linux 5.3.y is empty.

Thanks for your help

PLMR-FR, 2021-03-21:
Hi again,

I compiled a kernel from you repository 5.6.y, but unfortunately the board doesn't start after installing the kernel. I am using your image of ubuntu 16.04

Here is what I did for building : 
```
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- m2z_defconfig
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- oldconfig
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=output zImage 
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=output dtbs
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=output modules
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=output modules_install
export KV=$(strings ./arch/arm/boot/Image |grep "Linux version"|awk '{print $3}')
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_HDR_PATH=output/usr/src/linux-headers-${KV} headers_install
```

Then I copied via SSH the content of output directory in my home directory of BPI M2Z, the Zimage and the dtb file.

 And the for installing
```
sudo cp -vfr ./output/* /
sync
sudo cp -fv ./zImage_v5.6.0 /boot
sync
sudo cp -fv ./bpi-m2-zero.dtb_v5.6.0 /boot
sync
```
And make it current kernel

```
cd /boot
sudo ln -sf zImage_v5.6.0 zImage
sudo ln -sf bpi-m2-zero.dtb_v5.6.0 bpi-m2-zero.dtb
cd ~
sync
```
And unfortunately if I do `sudo reboot` then the board doesn't start anymore

Did I make something wrong ?

Sorry for all these questions ...

avafinger, 2021-03-21:
Please, restore the previous links in order to boot the older kernel.
And what is the output of:
export KV=$(strings ./arch/arm/boot/Image |grep "Linux version"|awk '{print $3}')
**echo $KV**

avafinger, 2021-03-21:
it should be something like:

sudo cp -fv ./arch/arm/boot/zImage /boot/zImage_${KV}
sudo cp -vf  ./arch/arm/boot/dts/bpi-m2-zero-v4.dtb /boot/bpi-m2-zero-v4.dtb_${KV}
sudo cp -vfr ./output/usr/* /usr/
sudo cp -vfr ./output/lib/* /usr/lib/
sync
cd /boot	
sudo ln -sf zImage_${KV} zImage
sudo ln -sf bpi-m2-zero-v4.dtb_${KV} bpi-m2-zero.dtb

check if you have:
/lib/modules/
or
/usr/lib/modules/

PLMR-FR, 2021-03-21:
Hi;

```
echo $KV
5.6.0-rc7+
```
Thanks for your support, it's a first time for me with a kernel building ...

PLMR-FR, 2021-03-21:
Hi,

So now the board is able to boot with the new kernel but I lost at least HDMI and Wifi. But I can see the heartbeat and now the stop button is working.

What could cause the HDMI and Wifi issue ?

avafinger, 2021-03-21:
ls /usr/lib
ls /lib

PLMR-FR, 2021-03-21:
ls /usr/lib
```
apt                  initramfs-tools      pkgconfig
arm-linux-gnueabihf  insserv              pkg-config.multiarch
at-spi2-core         klibc                pm-utils
binfmt.d             ldscripts            policykit-1
colord               libacl.a             python2.7
compat-ld            libacl.so            python3
dbus-1.0             libibverbs.so.1      python3.5
dconf                libibverbs.so.1.0.0  rsyslog
dpkg                 libwiringPiDev.so    sasl2
eject                libwiringPi.so       sftp-server
emacsen-common       locale               ssl
gcc                  man-db               sudo
gettext              mime                 systemd
girepository-1.0     modules              tar
git-core             modules-load.d       tc
glib-networking      ocf                  tmpfiles.d
glusterfs            openssh              valgrind
gnupg                os-release           X11
gold-ld              p7zip                xvmcconfig-standard
groff                perl5

```
ls /usr/lib/modules
`5.6.0-rc7+`

ls /lib
```
arm-linux-gnueabihf  init                                  modprobe.d  terminfo
cpp                  klibc-wBxlC8vEZykm9g94Kf1Dr2tj4ts.so  modules     udev
firmware             ld-linux-armhf.so.3                   resolvconf
ifupdown             lsb                                   systemd
```
ls /lib/modules
`4.17.4-m2z  4.18.4-m2z  4.4.0-204-generic  5.3.4`

I don't know why byt in /lib/modules it seems that 5.6.0-rc7+ is missing

avafinger, 2021-03-21:
fix with: 
**sudo cp -vfr ./output/lib/* //lib/**

PLMR-FR, 2021-03-21:
Hi,

Wifi is back but not HDMI. I don't know if it's because I mess up with the kernel installation. I will try again by doing right the first time.

Thanks for your help.

PLMR-FR, 2021-03-25:
Hi,

Finally it works, it was a mistake in the .config file.
I missed something in menuconfi for building kernel.
Just check in the .config file that both following flags are set :
```
CONFIG_POWER_RESET=y
CONFIG_POWER_RESET_GPIO=y
```

Thank you very much avafinger for you support !

closed 2021-03-25 by PLMR-FR