avafinger/bananapi-zero-ubuntu-base-minimal #58: Netplan issue on Ubuntu 20.04 LTS (focal)

rlesac, 2021-04-12:
Fresh install just edited /etc/netplan/01-netcfg.yaml to change to my details.
Any idea?
```
@bpi-m2z:~$ sudo netplan apply

A dependency job for netplan-wpa-wlan0.service failed. See 'journalctl -xe' for details.
Traceback (most recent call last):
  File "/usr/sbin/netplan", line 23, in <module>
    netplan.main()
  File "/usr/share/netplan/netplan/cli/core.py", line 50, in main
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 257, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/apply.py", line 55, in run
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 257, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/apply.py", line 232, in command_apply
    utils.systemctl_networkd('start', sync=True, extra_services=netplan_wpa + netplan_ovs)
  File "/usr/share/netplan/netplan/cli/utils.py", line 125, in systemctl_networkd
    subprocess.check_call(command)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['systemctl', 'start', 'systemd-networkd.service', 'netplan-wpa-wlan0.service', 'netplan-ovs-cleanup.service']' returned non-zero exit status 1.

```

avafinger, 2021-04-12:
Did you try this one?: https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.10

How did you change the 01-netcfg.yaml? Don't use windows program for that.

rlesac, 2021-04-12:
I used the v3.10 release. 
Used vim for editing, just changed the network details to mine and that is it.

avafinger, 2021-04-13:
Please, check if vim used TABs.
Check if you have changed the indentation.

And try:
```
sudo netplan generate
sudo netplan apply
```

rlesac, 2021-04-13:
The yaml seems valid. Tested with an online validator.
Also here's the output of  "netplan --debug generate"

```
@bpi-m2z:~$ sudo netplan --debug generate
DEBUG:command generate: running ['/lib/netplan/generate']
** (generate:14877): DEBUG: 13:15:30.328: Processing input file /etc/netplan/01-netcfg.yaml..
** (generate:14877): DEBUG: 13:15:30.336: starting new processing pass
** (generate:14877): DEBUG: 13:15:30.336: wlan0: adding wifi AP 'Just Testing'
** (generate:14877): DEBUG: 13:15:30.336: We have some netdefs, pass them through a final round of validation
** (generate:14877): DEBUG: 13:15:30.336: wlan0: setting default backend to 1
** (generate:14877): DEBUG: 13:15:30.337: Configuration is valid
** (generate:14877): DEBUG: 13:15:30.337: eth0: setting default backend to 1
** (generate:14877): DEBUG: 13:15:30.337: Configuration is valid
** (generate:14877): DEBUG: 13:15:30.338: Generating output files..
** (generate:14877): DEBUG: 13:15:30.338: openvswitch: definition eth0 is not for us (backend 1)
** (generate:14877): DEBUG: 13:15:30.338: NetworkManager: definition eth0 is not for us (backend 1)
** (generate:14877): DEBUG: 13:15:30.338: Creating wpa_supplicant config
** (generate:14877): DEBUG: 13:15:30.339: wlan0: Creating wpa_supplicant configuration file run/netplan/wpa-wlan0.conf
** (generate:14877): DEBUG: 13:15:30.339: Creating wpa_supplicant unit /run/systemd/system/netplan-wpa-wlan0.service
(generate:14877): GLib-DEBUG: 13:15:30.339: posix_spawn avoided (workdir specified) (fd close requested)
** (generate:14877): DEBUG: 13:15:30.400: Creating wpa_supplicant service enablement link /run/systemd/system/systemd-networkd.service.wants/netplan-wpa-wlan0.service
** (generate:14877): DEBUG: 13:15:30.401: openvswitch: definition wlan0 is not for us (backend 1)
** (generate:14877): DEBUG: 13:15:30.401: NetworkManager: definition wlan0 is not for us (backend 1)
(generate:14877): GLib-DEBUG: 13:15:30.403: posix_spawn avoided (fd close requested)
(generate:14877): GLib-DEBUG: 13:15:30.411: posix_spawn avoided (fd close requested)

```

rlesac, 2021-04-13:
```
i@bpi-m2z:~$ ls /sys/class/net
eth0  lo
```
I'm missing wlan here

avafinger, 2021-04-13:
what's the output of:

```
buntu@bpi-m2z:~$ dmesg|grep 'brc\|wl\|802'
[    2.475267] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    2.486390] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    5.802730] systemd[1]: Created slice system-systemd\x2dfsck.slice.
[    8.129744] cfg80211: failed to load regulatory.db
[    8.454290] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    8.630684] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    9.213590] Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.hcd' Patch
[    9.570462] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-11), device may have limited channels available
[    9.575469] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug 29 2016 20:48:16 version 7.45.41.26 (r640327) FWID 01-4527cfab

```

avafinger, 2021-04-13:
Maybe there is a new hw release with A2 chip?

rlesac, 2021-04-13:
> Maybe there is a new hw release with A2 chip?

How do I check?
This is my output:
```
bpi-m2z:~$ dmesg|grep 'brc\|wl\|802'
[    2.432926] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    2.443580] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    7.736974] cfg80211: failed to load regulatory.db
```

avafinger, 2021-04-13:
Try to find any version printed on the silkscreen

avafinger, 2021-04-13:
Also, check what is printed on the wifi chip.

rlesac, 2021-04-14:
[image]
Seems normal to me.

Wlan was working with your earlier build, but it was not reliable.

avafinger, 2021-04-14:
Board version 1.0 is the same as mine. Your Wifi chip version is newer.
Can you try with another SD card?