avafinger/bananapi-zero-ubuntu-base-minimal #60: Can't install Haproxy with APT

rlesac, 2021-08-25:
Using Ubuntu 20.04 LTS - Kernel 5.7.10 / 5.11.10 - BPI-M2Z, fresh install.

`user@bpi-m2z:~$ sudo apt install haproxy
[sudo] password for user:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Suggested packages:
  vim-haproxy haproxy-doc
The following NEW packages will be installed:
  haproxy
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/1469 kB of archives.
After this operation, 2645 kB of additional disk space will be used.
Selecting previously unselected package haproxy.
(Reading database ... 35683 files and directories currently installed.)
Preparing to unpack .../haproxy_2.0.13-2ubuntu0.1_armhf.deb ...
Unpacking haproxy (2.0.13-2ubuntu0.1) ...
Setting up haproxy (2.0.13-2ubuntu0.1) ...
Detected unsafe path transition / → /run during canonicalization of /run.
Job for haproxy.service failed because the control process exited with error code.
See "systemctl status haproxy.service" and "journalctl -xe" for details.
invoke-rc.d: initscript haproxy, action "start" failed.
● haproxy.service - HAProxy Load Balancer
     Loaded: loaded (/lib/systemd/system/haproxy.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2021-08-25 11:12:07 UTC; 69ms ago
       Docs: man:haproxy(1)
             file:/usr/share/doc/haproxy/configuration.txt.gz
    Process: 589 ExecStartPre=/usr/sbin/haproxy -f $CONFIG -c -q $EXTRAOPTS (code=exited, status=0/SUCCESS)
    Process: 590 ExecStart=/usr/sbin/haproxy -Ws -f $CONFIG -p $PIDFILE $EXTRAOPTS (code=exited, status=1/FAILURE)
   Main PID: 590 (code=exited, status=1/FAILURE)
dpkg: error processing package haproxy (--configure):
 installed haproxy package post-installation script subprocess returned error exit status 1
Processing triggers for rsyslog (8.2001.0-1ubuntu1.1) ...
Processing triggers for systemd (245.4-4ubuntu3.6) ...
Errors were encountered while processing:
 haproxy
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

The error is with admin.sock creation:

`@bpi-m2z:~$ haproxy -f /etc/haproxy/haproxy.cfg -db
[ALERT] 236/111311 (666) : Starting frontend GLOBAL: cannot bind UNIX socket [/run/haproxy/admin.sock]
`

avafinger, 2021-08-25:
No experience with haproxy, but i can see the install process failed, more likely due  /run/haproxy not created.
Looks like permission issue to me.

Read this: https://stackoverflow.com/questions/30101075/haproxy-doesnt-start-can-not-bind-unix-socket-run-haproxy-admin-sock

rlesac, 2021-08-25:
I wasted a bunch of hours researching and trying to find a solution already. If I create the file manually, Haproxy starts up fine, but doesn't persist after a reboot. I'm guessing the root cause is in the image, as Haproxy is very widely used.

avafinger, 2021-08-25:
You should check for permissions in /run and for your user 'user'.

Try to run the installer as root.

`sudo su -`

rlesac, 2021-08-25:
The issue was / being owned by the user "ubuntu".  I think you should fix that in the image.