avafinger/bananapi-zero-ubuntu-base-minimal #62: cpu no longer @1200 MHz

gavan1, 2021-08-29:
using  https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.10

CPU mac Mhz: 1008

anyway to bring this back up as you had before?
The CPU  is rated for 1200 MHz.

avafinger, 2021-08-29:
> anyway to bring this back up as you had before?

Can you remember which kernel was it?

avafinger, 2021-08-29:
Can you try this one and report the results?
**Kernel 5.11.10-server**

[bpi-m2-zero-v4.dtb.tar.gz](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/7071857/bpi-m2-zero-v4.dtb.tar.gz)

avafinger, 2021-08-29:
In case you don't know how to install it, make sure you have a backup of the file **bpi-m2-zero-v4.dtb** in **/boot** prior to decompressing (untar) the file bpi-m2-zero-v4.dtb.tar.gz in **/boot** partition with **sudo** or **root**.

gavan1, 2021-08-29:
currently kernel 5.7.10

avafinger, 2021-08-29:
Hmmm,  It should not interfere with the heartbeat, watch the CPU Temp and stability issues. Use a really good heat sink!

gavan1, 2021-09-07:
Board works....Not sure what happened there.

Does Kernel 5.11.10-server have Mali support built in?
How to test?

avafinger, 2021-09-07:
The "server" has no mali support.

If I remember correctly https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.3 support mali gbm.

glmark2-2014.03 or kmscube

closed 2021-09-23 by gavan1