avafinger/bananapi-zero-ubuntu-base-minimal #64: Mining with sgminer

gavan1, 2021-09-19:
https://www.novaspirit.com/2017/12/21/gpu-mining-on-tinkerboard/

essentially the guide should work on all Mali SBC's.
Works fine on my tinkerboard.

I thought using the [image](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.3) with Mali gbm support would work.

Builds fine but get not OpenCL not found.

`------------------------------------------------------------------------
sgminer 5.1.0
------------------------------------------------------------------------

Configuration Options Summary:

  Use git version......: yes
  libcurl(GBT+getwork).: Enabled: -lcurl
  curses.TUI...........: FOUND: -lncurses
  OpenCL...............: NOT FOUND. GPU mining support DISABLED
configure: error: OpenCL not found
`

avafinger, 2021-09-19:
https://community.arm.com/developer/tools-software/graphics/f/discussions/9464/mali-400-mp2-opencl

gavan1, 2021-09-19:
confirmed Mali 400 no OPEN CL support!

closed 2021-09-19 by gavan1