avafinger/bananapi-zero-ubuntu-base-minimal #65: Unable to use micro USB Port (Keyboard/Mouse) up with Focal

matthewoots, 2021-10-14:
Hi @avafinger
Sorry I am new to this, I've used other images for Bananapi zero and some of them work, I managed to flash https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v3.10 and managed to enter the login screen, but I am unable to use the keyboard and mouse over the micro USB port.

My setup is below, the power goes through the first and the second is the connection to the keyboard (or vice versa supplies power), I have even tried splitting the power for the USB device to take power off another computer but to no avail.
[image]

I checked the power and it is 4.95V so that rules it out.
Is there any thing in the rootfs folder that I need to change to get it to work? Thank you!

avafinger, 2021-10-14:
No need to do anything. 
Try a micro USB hub like this one (my current setup)

[image]
.

closed 2021-10-15 by matthewoots

avafinger, 2021-10-18:
I think you have 4 UART, one is for the Debug, I think uart0. uart1 is for Wifi/BT. So you have uart3 enabled and ready to be used. I think in one of the releases I had enabled uart2, but I don't know it was used for something else. 
For your reference: https://github.com/avafinger/linux-5.6.y/blob/6dc50035fef1f65b7f5b3b818f69e1e7a3ef0616/arch/arm/boot/dts/bpi-m2-zero-v4.dts#L60

Yes, I am interested in such a  project. I will follow your progress. If you can post a link for each hw you used would be great.
When you finish the project I ask for the CAD files, thank you.

Regarding OTG, see here if it is the right otg mode:
https://github.com/avafinger/linux-5.6.y/blob/6dc50035fef1f65b7f5b3b818f69e1e7a3ef0616/arch/arm/boot/dts/bpi-m2-zero-v4.dts#L624

You can decompile the dtb and change or enable what you want.

Hope you get all you need.

avafinger, 2021-10-19:
uart0 is reserved for console debugging (the 3 pins GND, TX and RX), if you want to use uart0 I think you need change bpi-m2-zero_boot.scr environment, from 
`setenv bootargs "console=ttyS0,115200 earlyprintk rootfstype=ext4 root=/dev/mmcblk0p2 rw rootwait fsck.repair=${fsck.repair} panic=10 ${extra}"
`to
`setenv bootargs "console=ttyS3,115200 earlyprintk rootfstype=ext4 root=/dev/mmcblk0p2 rw rootwait fsck.repair=${fsck.repair} panic=10 ${extra}"
`

# Recompile with:
# cd /media/alex/boot/
# sudo mkimage -C none -A arm -T script -d bpi-m2-zero_boot.cmd bpi-m2-zero_boot.scr

Regarding USB, try:
  dr_mode = "otg"; 
to
  dr_mode = "host";

matthewoots, 2021-10-19:
@avafinger thank you so very much man!! "host" fixed it, finally can see the realsense!
**BEFORE**
```
ubuntu@bpi-m2z:~/catkin_ws$ ls -lR /dev/input
/dev/input:
total 0
drwxr-xr-x 2 root root      80 Sep  7 18:37 by-path
crw-rw---- 1 root input 13, 64 Sep  7 18:37 event0
crw-rw---- 1 root input 13, 65 Sep  7 18:37 event1
crw-rw---- 1 root input 13, 63 Sep  7 18:37 mice

/dev/input/by-path:
total 0
lrwxrwxrwx 1 root root 9 Sep  7 18:37 platform-1f02000.ir-event -> ../event1
lrwxrwxrwx 1 root root 9 Sep  7 18:37 platform-r_gpio_keys-event -> ../event0
```
**AFTER**
```
ubuntu@bpi-m2z:~$ ls -lR /dev/input
/dev/input:
total 0
drwxr-xr-x 2 root root        60 Sep  7 18:37 by-id
drwxr-xr-x 2 root root       100 Sep  7 18:37 by-path
crw-rw---- 1 root input   13, 64 Sep  7 18:37 event0
crw-rw-rw- 1 root plugdev 13, 65 Sep  7 18:37 event1
crw-rw---- 1 root input   13, 66 Sep  7 18:37 event2
crw-rw---- 1 root input   13, 63 Sep  7 18:37 mice

/dev/input/by-id:
total 0
lrwxrwxrwx 1 root root 9 Sep  7 18:37 usb-Intel_R__RealSense_TM__Depth_Camera_430i_Intel_R__RealSense_TM__Depth_Camera_430i-event-if00 -> ../event1

/dev/input/by-path:
total 0
lrwxrwxrwx 1 root root 9 Sep  7 18:37 platform-1c1a000.usb-usb-0:1:1.0-event -> ../event1
lrwxrwxrwx 1 root root 9 Sep  7 18:37 platform-1f02000.ir-event -> ../event2
lrwxrwxrwx 1 root root 9 Sep  7 18:37 platform-r_gpio_keys-event -> ../event0
```
Will then use the debug for UART let me try it out first. Thank you agin!