avafinger/bananapi-zero-ubuntu-base-minimal #67: Bluetooth connect to GN

Bahir, 2022-04-19:
$ bt-network -c 00:15:83:4B:1F:1B GN
Segmentation fault

$tail /var/log/syslog

[469774.598379] Bluetooth: hci0: Received unexpected HCI Event 00000000
[469775.428119] Bluetooth: hci0: Frame reassembly failed (-84)
[469775.428184] Bluetooth: unknown link type 4
[469776.567365] Bluetooth: hci0: command 0x0405 tx timeout

avafinger, 2022-04-19:
uname -ra
attach zipped bootlog please.

Bahir, 2022-04-19:
`uname -ra`
Linux bpi-m2z 5.11.10-server #1 SMP Sat Mar 27 09:46:38 -03 2021 armv7l armv7l armv7l GNU/Linux
[bootlog.zip](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/files/8516906/bootlog.zip)

avafinger, 2022-04-19:
Are you trying to connect to BT headset?

Bahir, 2022-04-19:
No, to Bluetooth shared network (on PC)

avafinger, 2022-04-19:
Check it out this notes and see if it helps:
https://wiki.polaire.nl/doku.php?id=network_over_bluetooth
If you explain in details what you are trying to accomplish i will try to reproduce it here, when i have time.