avafinger/bananapi-zero-ubuntu-base-minimal #68: Problems connecting to wifi

MrOneTwo, 2022-08-10:
I'm having troubles connecting the WiFi.
When I run `sudo wpa_cli -i wlan0` I see:

```
<3>CTRL-EVENT-SCAN-STARTED 
> 
<3>CTRL-EVENT-SCAN-RESULTS 
> 
<4>Failed to initiate sched scan
```

I have used [flash_sdcard_m2z_v2.sh](https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/blob/master/flash_sdcard_m2z_v2.sh) to flash my SD card. I initially just edited the `interfaces` file but now I'm using `wpa_cli` through UART to fix the fact that it fails to connect.

MrOneTwo, 2022-08-10:
Oh and very rarely I see:

```
<3>WPS-AP-AVAILABLE 
> 
<3>CTRL-EVENT-SSID-REENABLED id=0 ssid=“Tele2_fca84e”
> 
<3>Trying to associate with SSID ‘Tele2_fca84e’
> 
<3>CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
> 
<3>CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid=“Tele2_fca84e” auth_failures=4 duration=77 reason=CONN_FAILED
> 
<3>CTRL-EVENT-SCAN-STARTED 
> 
<3>CTRL-EVENT-SCAN-RESULTS 
> 
<3>WPS-AP-AVAILABLE 
> 
<4>Failed to initiate sched scan
> 
```

avafinger, 2022-08-10:
Maybe the new board has a different wifi chip or similar chip. Try to update firmware if that is the case.