avafinger/bananapi-zero-ubuntu-base-minimal #70: AddScreen/ScreenInit failed for driver 0

lblabr, 2022-08-20:
i followed your instruction to install ubuntu, i used 

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/tree/v3.10

and installed lxde.

lxde does not start up it exiits with the following errors: 

`

        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sat Aug 20 16:22:28 2022
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) AddScreen/ScreenInit failed for driver 0
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE)
`
any idea, how to solve?

thanks a lot!

avafinger, 2022-08-22:
Please, type "uname -ra"

avafinger, 2022-08-29:
https://www.x.org/wiki/FAQErrorMessages/

Make sure you have the "server" kernel without mali. The mali version released with the image is for GBM only.